CadQuery/CQ-editor #1: Segmentation Fault on First Run

jmwright, 2018-06-08:
I followed the instructions to set the conda environment up, verified that it was the active one, and tried to run CQ-editor. I immediately get a seg fault.
```bash
Segmentation fault (core dumped)
```
There's no way to know where it comes from based on that message, but I've gotten a lot of seg faults with FreeCAD over the years.

I'm running Ubuntu 17.10 64-bit

adam-urbanczyk, 2018-06-08:
Well, not exactly the user experience I had in mind.

Could you dump your conda env:
```
conda env list
```
Could you also run it with gdb and share the backtrace. I think it has to roughly go like this:
```
gdb -args python run.py
> run
... after core dump
> bt
```

jmwright, 2018-06-08:
```bash
$ conda env list
# conda environments:
#
cq-occ-testing           /home/jwright/.conda/envs/cq-occ-testing
freecad_cq3              /home/jwright/.conda/envs/freecad_cq3
                         /home/jwright/miniconda
                         /home/jwright/miniconda/envs/freecad_cq3
base                     /home/jwright/miniconda3
cqgui                 *  /home/jwright/miniconda3/envs/cqgui
env-name                 /home/jwright/miniconda3/envs/env-name
freecad-py3              /home/jwright/miniconda3/envs/freecad-py3
```

gdb output:
```bash
(gdb) run
Starting program: /home/jwright/miniconda3/envs/cqgui/bin/python run.py
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
PyType_IsSubtype (a=0x626f6174656d5f74, b=0x7ffff23e0680 <sipVoidPtr_Type>)
    at Objects/typeobject.c:1357
1357	Objects/typeobject.c: No such file or directory
```

Is there maybe a *-dev package I need to install?

jmwright, 2018-06-08:
> could you also run conda list on the cqgui env
```bash
$ conda list
# packages in environment at /home/jwright/miniconda3/envs/cqgui:
#
# Name                    Version                   Build  Channel
alabaster                 0.7.10                   py36_1    conda-forge
asn1crypto                0.24.0                   py36_0    conda-forge
astroid                   1.6.5                    py36_0    conda-forge
babel                     2.6.0                    py36_0    conda-forge
backcall                  0.1.0                      py_0    conda-forge
bleach                    2.1.3                      py_0    conda-forge
ca-certificates           2018.4.16                     0    conda-forge
cadquery-occ              0                        py36_0    cadquery
certifi                   2018.4.16                py36_0    conda-forge
cffi                      1.11.5                   py36_0    conda-forge
chardet                   3.0.4                    py36_0    conda-forge
cloudpickle               0.5.3                      py_0    conda-forge
cryptography              2.2.1                    py36_0    conda-forge
dbus                      1.13.0               h3a4f0e9_0    conda-forge
decorator                 4.3.0                      py_0    conda-forge
docutils                  0.14                     py36_0    conda-forge
entrypoints               0.2.3                    py36_1    conda-forge
expat                     2.2.5                         0    conda-forge
fontconfig                2.12.1                        4    conda-forge
freeimageplus             3.17.0                        0    dlr-sc
freetype                  2.7                           1    conda-forge
gettext                   0.19.8.1                      0    conda-forge
gl2ps                     1.3.8                         1    dlr-sc
glib                      2.55.0                        0    conda-forge
gmp                       6.1.2                         0    conda-forge
gst-plugins-base          1.8.0                         0    conda-forge
gstreamer                 1.8.0                         1    conda-forge
html5lib                  1.0.1                      py_0    conda-forge
icu                       58.2                          0    conda-forge
idna                      2.6                      py36_1    conda-forge
imagesize                 1.0.0                    py36_0    conda-forge
ipykernel                 4.8.2                    py36_0    conda-forge
ipython                   6.4.0                    py36_0    conda-forge
ipython_genutils          0.2.0                    py36_0    conda-forge
isort                     4.3.4                    py36_0    conda-forge
jedi                      0.12.0                   py36_0    conda-forge
jinja2                    2.10                     py36_0    conda-forge
jpeg                      9b                            2    conda-forge
jsonschema                2.6.0                    py36_1    conda-forge
jupyter_client            5.2.3                    py36_0    conda-forge
jupyter_core              4.4.0                      py_0    conda-forge
lazy-object-proxy         1.3.1                    py36_0    conda-forge
libffi                    3.2.1                         3    conda-forge
libgcc                    7.2.0                h69d50b8_2  
libgcc-ng                 7.2.0                hdf63c60_3  
libgfortran-ng            7.2.0                hdf63c60_3  
libiconv                  1.15                          0    conda-forge
libopenblas               0.2.20               h9ac9557_7  
libpng                    1.6.34                        0    conda-forge
libsodium                 1.0.16                        0    conda-forge
libstdcxx-ng              7.2.0                hdf63c60_3  
libxcb                    1.13                          0    conda-forge
libxml2                   2.9.3                         1    dlr-sc
Logbook                   1.4.0                     <pip>
markupsafe                1.0                      py36_0    conda-forge
mccabe                    0.6.1                    py36_0    conda-forge
mistune                   0.8.3                    py36_1    conda-forge
nbconvert                 5.3.1                      py_1    conda-forge
nbformat                  4.4.0                    py36_0    conda-forge
ncurses                   5.9                          10    conda-forge
numpy                     1.14.3           py36h28100ab_2  
numpy-base                1.14.3           py36h0ea5e3f_1  
numpydoc                  0.8.0                    py36_0    conda-forge
oce                       0.18.1                        1    oce
[... 227 lines cut ...]
    qualname=qualname@entry=0x0) at Python/ceval.c:4153
---Type <return> to continue, or q <return> to quit---
#70 0x00007ffff79fd64d in PyEval_EvalCodeEx (_co=_co@entry=0x7ffff7ef7ae0, 
    globals=globals@entry=0x7ffff7f3b288, locals=locals@entry=0x7ffff7f3b288, args=args@entry=0x0, 
    argcount=argcount@entry=0, kws=kws@entry=0x0, kwcount=kwcount@entry=0, defs=defs@entry=0x0, 
    defcount=defcount@entry=0, kwdefs=kwdefs@entry=0x0, closure=closure@entry=0x0)
    at Python/ceval.c:4174
#71 0x00007ffff79fd69b in PyEval_EvalCode (co=co@entry=0x7ffff7ef7ae0, 
    globals=globals@entry=0x7ffff7f3b288, locals=locals@entry=0x7ffff7f3b288) at Python/ceval.c:730
[... cut ...]
    closeit=closeit@entry=1, flags=flags@entry=0x7fffffffd7d0) at Python/pythonrun.c:420
#75 0x00007ffff7a39693 in PyRun_AnyFileExFlags (fp=fp@entry=0x69a030, filename=<optimized out>, 
    closeit=closeit@entry=1, flags=flags@entry=0x7fffffffd7d0) at Python/pythonrun.c:81
#76 0x00007ffff7a556cd in run_file (p_cf=0x7fffffffd7d0, filename=0x603550 L"run.py", fp=0x69a030)
    at Modules/main.c:340
#77 Py_Main (argc=argc@entry=2, argv=argv@entry=0x602260) at Modules/main.c:810
#78 0x0000000000400bbc in main (argc=2, argv=<optimized out>) at ./Programs/python.c:69
```

adam-urbanczyk, 2018-06-09:
Thanks @jmwright , so the crash is either PyQT or SIP related. I do not really get how this occurs - I have exactly the same packages in my env. The only idea I can is updating to a different version of one of those packages

Could you try updating the sip package to a different version
```conda install -c conda-forge sip=4.18.1```
or using a different pyqt build:
```conda install -c conda-forge 5.6.0=py36_2```
but definitely not at once.

I still do not get the big picture, you nominally have the same packages as me but still getting a segfault. Could that be because some of the packages are cached locally and in reality different to what is on the server? Maybe if you have time, you could try running `conda clean --all` and repeating the install.

Anyhow, thanks for trying it out!

adam-urbanczyk, 2018-06-11:
Any luck @jmwright ? I did setup CI, so the env spec seems to be OK in principle. I'll try to modify the conda environment to use the defaults channel for pyqt; maybe this will solve the problem.

jmwright, 2018-06-11:
I hadn't gotten back to it yet. I just tried the following line and it fixed the segfault.
> conda install -c conda-forge sip=4.18.1

closed 2018-06-11 by jmwright

adam-urbanczyk, 2018-06-13:
OK, just in case I added this SIP version to the env file. CI seems to be happy about it too.

Spoutnik97, 2023-10-20:
@adam-urbanczyk it seems that I have the same issue "Segmentation fault" with the last version of Cadquery

here is the gdb log : 
```
Program received signal SIGSEGV, Segmentation fault.
0x0000000000004740 in ?? ()
(gdb) bt
#0  0x0000000000004740 in ?? ()
#1  0x00007f3167cc445e in ?? () from /lib64/ld-linux-x86-64.so.2
#2  0x00007f3167cc454c in ?? () from /lib64/ld-linux-x86-64.so.2
#3  0x00007f3167b10d55 in _dl_catch_exception () from /usr/glibc-compat/lib/libc.so.6
#4  0x00007f3167cc8bec in ?? () from /lib64/ld-linux-x86-64.so.2
#5  0x00007f3167b10cf8 in _dl_catch_exception () from /usr/glibc-compat/lib/libc.so.6
#6  0x00007f3167cc83fe in ?? () from /lib64/ld-linux-x86-64.so.2
#7  0x00007f3167a4069c in ?? () from /usr/glibc-compat/lib/libc.so.6
#8  0x00007f3167b10cf8 in _dl_catch_exception () from /usr/glibc-compat/lib/libc.so.6
#9  0x00007f3167b10dc3 in _dl_catch_error () from /usr/glibc-compat/lib/libc.so.6
#10 0x00007f3167a401be in ?? () from /usr/glibc-compat/lib/libc.so.6
#11 0x00007f3167a4072a in dlopen () from /usr/glibc-compat/lib/libc.so.6
#12 0x0000556ffdf987ec in _PyImport_FindSharedFuncptr (prefix=prefix@entry=0x556ffdfed40b "PyInit",
    shortname=shortname@entry=0x7f31673eda90 "OCP",
    pathname=pathname@entry=0x7f316737cf50 "/opt/miniforge3/lib/python3.10/site-packages/OCP.cpython-310-x86_64-linux-gnu.so", fp=fp@entry=0x0) at /usr/local/src/conda/python-3.10.12/Python/dynload_shlib.c:100
#13 0x0000556ffdf9817e in _PyImport_LoadDynamicModuleWithSpec (fp=0x0, spec=0x7f31673edba0)
    at /usr/local/src/conda/python-3.10.12/Python/importdl.c:137
#14 _imp_create_dynamic_impl (module=<optimized out>, file=<optimized out>, spec=0x7f31673edba0)
    at /usr/local/src/conda/python-3.10.12/Python/import.c:2050
#15 _imp_create_dynamic (module=<optimized out>, args=<optimized out>, nargs=<optimized out>)
    at /usr/local/src/conda/python-3.10.12/Python/clinic/import.c.h:330
#16 0x0000556ffdecdb63 in cfunction_vectorcall_FASTCALL (func=0x7f316793d8a0, args=0x7f31673edaf8,
    nargsf=<optimized out>, kwnames=<optimized out>)
    at /usr/local/src/conda/python-3.10.12/Objects/methodobject.c:430
#17 0x0000556ffdec334b in do_call_core (kwdict=0x7f316740c080, callargs=0x7f31673edae0,
    func=0x7f316793d8a0, trace_info=0x7fff5754d2b0, tstate=<optimized out>)
    at /usr/local/src/conda/python-3.10.12/Python/ceval.c:5917
#18 _PyEval_EvalFrameDefault (tstate=<optimized out>, f=0x7f3167400550, throwflag=<optimized out>)
    at /usr/local/src/conda/python-3.10.12/Python/ceval.c:4277
#19 0x0000556ffdecd99c in _PyEval_EvalFrame (throwflag=0, f=0x7f3167400550, tstate=0x556fff573880)
    at /usr/local/src/conda/python-3.10.12/Include/internal/pycore_ceval.h:46
#20 _PyEval_Vector (kwnames=<optimized out>, argcount=<optimized out>, args=<optimized out>, locals=0x0,
    con=0x7f31678f1520, tstate=0x556fff573880) at /usr/local/src/conda/python-3.10.12/Python/ceval.c:5067
#21 _PyFunction_Vectorcall (func=0x7f31678f1510, stack=<optimized out>, nargsf=<optimized out>,
    kwnames=<optimized out>) at /usr/local/src/conda/python-3.10.12/Objects/call.c:342
#22 0x0000556ffdec2142 in _PyObject_VectorcallTstate (kwnames=0x0, nargsf=<optimized out>,
    args=0x7f3167955188, callable=0x7f31678f1510, tstate=0x556fff573880)
    at /usr/local/src/conda/python-3.10.12/Include/cpython/abstract.h:114
#23 PyObject_Vectorcall (kwnames=0x0, nargsf=<optimized out>, args=0x7f3167955188,
    callable=0x7f31678f1510) at /usr/local/src/conda/python-3.10.12/Include/cpython/abstract.h:123
    ```
    
    
    Could you help me to understand this message please ?