CadQuery/CQ-editor #2: Missing libraries for GTK

gntech, 2018-06-10:
I am running this on miniconda on Ubuntu Mate 18.04 and I get the following error together with that the styling of buttons in the application looks odd. Not at all like they do in the screenshot in the readme.

``` bash
~/c/CQ-editor (master|✔) $ python run.py                                                                               (cqgui) 
INFO:OCC.Display.backend:backend loaded: qt-pyqt5
QGtkStyle could not resolve GTK. Make sure you have installed the proper libraries.
```

adam-urbanczyk, 2018-06-11:
Thanks for trying! Could you post a screenshot and maybe run the app with an strace to see which library it tries to load?

I found the following workaround to a similar issues, maybe worth checking in a short term:

https://stackoverflow.com/questions/50555973/qgtkstyle-could-not-resolve-gtk/50583925#50583925

```
export QT_STYLE_OVERRIDE=gtk2
```

I also see a similar issue here: https://github.com/spyder-ide/spyder/issues/7269. Will check if their advice works.

jmwright, 2018-06-19:
@gntech Maybe try the accepted answer [here](https://stackoverflow.com/questions/50555973/qgtkstyle-could-not-resolve-gtk)?

gntech, 2018-06-19:
@jmwright @adam-urbanczyk 
Yes, setting the environment variable `QT_STYLE_OVERRIDE=gtk2` works, thanks!

closed 2018-06-19 by gntech