CadQuery/CQ-editor #6: Segmentation fault on run

manymachines, 2018-08-29:
This is my first try running the editor.  I followed the debug steps from issue #1.   This is a clean install.  In my case it looks like SIP is the expected version 4.18.1.

```
$conda env list
# conda environments:
#
base                  *  /home/lorenzn/tools/miniconda3
cqgui                    /home/lorenzn/tools/miniconda3/envs/cqgui
```

```
$ conda list
# packages in environment at /home/lorenzn/tools/miniconda3/envs/cqgui:
#
# Name                    Version                   Build  Channel
alabaster                 0.7.11                     py_3    conda-forge
asn1crypto                0.24.0                   py36_3    conda-forge
astroid                   2.0.2                    py36_0    conda-forge
babel                     2.6.0                      py_1    conda-forge
backcall                  0.1.0                      py_0    conda-forge
bleach                    2.1.4                      py_1    conda-forge
bzip2                     1.0.6                h470a237_2    conda-forge
ca-certificates           2018.8.24            ha4d7672_0    conda-forge
cadquery-occ              0.1.5             py3.6180e50c0    cadquery
certifi                   2018.8.24                py36_1    conda-forge
cffi                      1.11.5           py36h5e8e0c9_1    conda-forge
chardet                   3.0.4                    py36_3    conda-forge
cloudpickle               0.5.5                      py_0    conda-forge
cryptography              2.3.1            py36hdffb7b8_0    conda-forge
cryptography-vectors      2.3.1                    py36_0    conda-forge
dbus                      1.13.0               h3a4f0e9_0    conda-forge
decorator                 4.3.0                      py_0    conda-forge
docutils                  0.14                     py36_1    conda-forge
entrypoints               0.2.3                    py36_2    conda-forge
expat                     2.2.5                hfc679d8_1    conda-forge
fontconfig                2.12.1                        4    conda-forge
freeimageplus             3.17.0                        0    dlr-sc
freetype                  2.7                           1    conda-forge
gettext                   0.19.8.1                      0    conda-forge
gl2ps                     1.3.8                         1    dlr-sc
glib                      2.55.0               h464dc38_2    conda-forge
gmp                       6.1.2                hfc679d8_0    conda-forge
gst-plugins-base          1.12.5               hde13a9d_0    conda-forge
gstreamer                 1.12.5               h61a6719_0    conda-forge
html5lib                  1.0.1                      py_0    conda-forge
icu                       58.2                 hfc679d8_0    conda-forge
idna                      2.7                      py36_2    conda-forge
imagesize                 1.0.0                      py_1    conda-forge
ipykernel                 4.8.2                    py36_0    conda-forge
ipython                   6.5.0                    py36_0    conda-forge
ipython_genutils          0.2.0                      py_1    conda-forge
isort                     4.3.4                    py36_0    conda-forge
jedi                      0.12.1                   py36_0    conda-forge
jinja2                    2.10                       py_1    conda-forge
jpeg                      9c                   h470a237_0    conda-forge
jsonschema                2.6.0                    py36_2    conda-forge
jupyter_client            5.2.3                      py_1    conda-forge
jupyter_core              4.4.0                      py_0    conda-forge
lazy-object-proxy         1.3.1            py36h470a237_0    conda-forge
libffi                    3.2.1                hfc679d8_4    conda-forge
libgcc-ng                 7.2.0                hdf63c60_3    conda-forge
libgfortran-ng            7.2.0                hdf63c60_3    conda-forge
libiconv                  1.15                 h470a237_2    conda-forge
libopenblas               0.2.20               h9ac9557_7  
libpng                    1.6.35               ha92aebf_0    conda-forge
libsodium                 1.0.16                        0    conda-forge
libstdcxx-ng              7.2.0                hdf63c60_3    conda-forge
libxcb                    1.13                 h470a237_2    conda-forge
libxml2                   2.9.3                         1    dlr-sc
Logbook                   1.4.0                     <pip>
[... 172 lines cut ...]
#34 0x00007ffff79fe09e in _PyEval_EvalCodeWithName (_co=_co@entry=0x7ffff7f049c0, 
    globals=globals@entry=0x7ffff7f451b0, locals=locals@entry=0x7ffff7f451b0, args=args@entry=0x0, 
    argcount=argcount@entry=0, kwnames=kwnames@entry=0x0, kwargs=kwargs@entry=0x0, kwcount=kwcount@entry=0, 
    kwstep=kwstep@entry=2, defs=defs@entry=0x0, defcount=defcount@entry=0, kwdefs=kwdefs@entry=0x0, 
    closure=closure@entry=0x0, name=name@entry=0x0, qualname=qualname@entry=0x0) at Python/ceval.c:4159
#35 0x00007ffff79fe6cd in PyEval_EvalCodeEx (_co=_co@entry=0x7ffff7f049c0, globals=globals@entry=0x7ffff7f451b0, 
    locals=locals@entry=0x7ffff7f451b0, args=args@entry=0x0, argcount=argcount@entry=0, kws=kws@entry=0x0, 
    kwcount=kwcount@entry=0, defs=defs@entry=0x0, defcount=defcount@entry=0, kwdefs=kwdefs@entry=0x0, 
    closure=closure@entry=0x0) at Python/ceval.c:4180
#36 0x00007ffff79fe71b in PyEval_EvalCode (co=co@entry=0x7ffff7f049c0, globals=globals@entry=0x7ffff7f451b0, 
    locals=locals@entry=0x7ffff7f451b0) at Python/ceval.c:731
#37 0x00007ffff7a3a0a2 in run_mod (arena=0x7ffff7f61078, flags=0x7fffffffd120, locals=0x7ffff7f451b0, 
    globals=0x7ffff7f451b0, filename=0x7ffff7e83260, mod=0x640628) at Python/pythonrun.c:1025
#38 PyRun_FileExFlags (fp=fp@entry=0x69b370, filename_str=filename_str@entry=0x7ffff7f60078 "run.py", 
    start=start@entry=257, globals=globals@entry=0x7ffff7f451b0, locals=locals@entry=0x7ffff7f451b0, 
    closeit=closeit@entry=1, flags=flags@entry=0x7fffffffd120) at Python/pythonrun.c:978
#39 0x00007ffff7a3a207 in PyRun_SimpleFileExFlags (fp=fp@entry=0x69b370, filename=<optimized out>, 
    closeit=closeit@entry=1, flags=flags@entry=0x7fffffffd120) at Python/pythonrun.c:420
#40 0x00007ffff7a3a6a3 in PyRun_AnyFileExFlags (fp=fp@entry=0x69b370, filename=<optimized out>, 
    closeit=closeit@entry=1, flags=flags@entry=0x7fffffffd120) at Python/pythonrun.c:81
#41 0x00007ffff7a566fd in run_file (p_cf=0x7fffffffd120, filename=0x6036e0 L"run.py", fp=0x69b370)
    at Modules/main.c:340
#42 Py_Main (argc=argc@entry=2, argv=argv@entry=0x602260) at Modules/main.c:810
#43 0x0000000000400bbc in main (argc=2, argv=<optimized out>) at ./Programs/python.c:69
```

adam-urbanczyk, 2018-08-29:
Thanks for trying and reporting in such detail! I conclude from the backtrace that this is a different issue. The segfault happens from within QT on a `QLibrary::load()`. The question is which library is being loaded there. To answer this could you help me out and run
```
strace -o strace.out python run.py
```
and share the output file (`strace.out`)? Also running `qtdiag` and sharing the output might be helpful too.

manymachines, 2018-08-29:
It looks like an interesting project.  I actually haven't tried the FreeCAD based version of CadQuery yet - thought I would try this first.

Attached is the strace.out and qtdiag output.

[strace.out.TXT](https://github.com/CadQuery/CQ-editor/files/2333936/strace.out.TXT)
[qtdiag.out.TXT](https://github.com/CadQuery/CQ-editor/files/2333939/qtdiag.out.TXT)

adam-urbanczyk, 2018-08-31:
Thanks @manymachines . Looking at strace results I assume that the last library loaded is 
```
/lib64/libpcre2-8.so.0
```
It being "Perl-compatible regular expression tools v2" does not tell me much. What is more I do not see it being loaded on my system. Not sure how to interpret this. For now I have one more idea, could you try with a different QT style?
```
export QT_STYLE_OVERRIDE=fusion
python run.py
```
It is a long shot so ymmv.

manymachines, 2018-08-31:
That workaround was successful.  The GUI now opens!

adam-urbanczyk, 2018-09-02:
Glad to hear that! Would you help me out further in reproducing this issue locally? I'd like to know what Linux distribution and destop environment are you using. Furthermore, I'd be really great if you could try what values of `QT_STYLE_OVERRIDE` cause a segfault. Based on the `qtdiag` output it can have the following values

* `gtk2`
* `gnome`
* `generic`
* `GTK+`
* `fusion`
* `windows`

manymachines, 2018-09-03:
Let me know if you could use any more information.

Fedora 28
GNOME 3.28.2

no crash:
- gtk2
- gnome
- generic
- fusion
- windows

crash:
- GTK+
- (or QT_STYLE_OVERRIDE unset)

westurner, 2019-07-09:
Fedora 29
Gnome 3.30.3
Conda CPython 3.6.7

Invalid (`QApplication: invalid style override passed, ignoring it.`) & crash:
```
QT_STYLE_OVERRIDE=gtk-2 cq-editor
QT_STYLE_OVERRIDE=gnome cq-editor
QT_STYLE_OVERRIDE=generic cq-editor
QT_STYLE_OVERRIDE=GTK+ cq-editor
```

Window opens but then crashes when I maximize:
```
QT_STYLE_OVERRIDE=fusion cq-editor
QT_STYLE_OVERRIDE=windows cq-editor
```

*****

```
$ conda list
# packages in environment at /home/wturner/-wrk/-ce37/woodwork:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                        main  
alabaster                 0.7.12                     py_0    conda-forge
asn1crypto                0.24.0                py36_1003    conda-forge
astroid                   2.2.5                    py36_0    conda-forge
attrs                     19.1.0                     py_0    conda-forge
babel                     2.7.0                      py_0    conda-forge
backcall                  0.1.0                      py_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.os              0.1.1                 py36_1001    conda-forge
bleach                    3.1.0                      py_0    conda-forge
bzip2                     1.0.6             h14c3975_1002    conda-forge
ca-certificates           2019.6.16            hecc5488_0    conda-forge
cadquery                  0.5-SNAPSHOT             pypi_0    pypi
certifi                   2019.6.16                py36_0    conda-forge
cffi                      1.12.3           py36h8022711_0    conda-forge
chardet                   3.0.4                 py36_1003    conda-forge
cloudpickle               1.2.1                      py_0    conda-forge
cq-editor                 master                    py3.6    cadquery
cryptography              2.7              py36h72c5cf5_0    conda-forge
dataclasses               0.6                      pypi_0    pypi
dbus                      1.13.6               he372182_0    conda-forge
decorator                 4.4.0                      py_0    conda-forge
defusedxml                0.5.0                      py_1    conda-forge
docutils                  0.14                  py36_1001    conda-forge
entrypoints               0.3                   py36_1000    conda-forge
expat                     2.2.5             he1b5a44_1003    conda-forge
fontconfig                2.13.1            he4413a7_1000    conda-forge
freeimageplus             3.17.0                        0    cadquery
freetype                  2.10.0               he983fc9_0    conda-forge
gettext                   0.19.8.1          hc5be6a0_1002    conda-forge
glib                      2.58.3            h6f030ca_1001    conda-forge
gst-plugins-base          1.14.5               h0935bb2_0    conda-forge
gstreamer                 1.14.5               h36ae1b5_0    conda-forge
icu                       58.2              hf484d3e_1000    conda-forge
idna                      2.8                   py36_1000    conda-forge
imagesize                 1.1.0                      py_0    conda-forge
importlib_metadata        0.18                     py36_0    conda-forge
ipydatawidgets            4.0.0                    pypi_0    pypi
ipykernel                 5.1.1            py36h24bf2e0_0    conda-forge
ipython                   7.6.1            py36h5ca1d4c_0    conda-forge
ipython_genutils          0.2.0                      py_1    conda-forge
ipywidgets                7.4.2                    pypi_0    pypi
isort                     4.3.21                   py36_0    conda-forge
jedi                      0.13.3                   py36_0    conda-forge
jeepney                   0.4                        py_0    conda-forge
[... 90 lines cut ...]
spyder                    3.3.4                    py36_1    conda-forge
spyder-kernels            0.5.0                    py36_0    conda-forge
sqlite                    3.28.0               hcee41ef_1    conda-forge
tbb                       2019.7               hc9558a2_0    conda-forge
terminado                 0.8.2                    py36_0    conda-forge
testpath                  0.4.2                   py_1001    conda-forge
tk                        8.6.9             hed695b0_1002    conda-forge
tornado                   6.0.3            py36h516909a_0    conda-forge
traitlets                 4.3.2                 py36_1000    conda-forge
traittypes                0.2.1                    pypi_0    pypi
typed-ast                 1.4.0            py36h516909a_0    conda-forge
urllib3                   1.24.3                   py36_0    conda-forge
wcwidth                   0.1.7                      py_1    conda-forge
webencodings              0.5.1                      py_1    conda-forge
wheel                     0.33.4                   py36_0    conda-forge
widgetsnbextension        3.4.2                    pypi_0    pypi
wrapt                     1.11.2           py36h516909a_0    conda-forge
wurlitzer                 1.0.3                    py36_0    conda-forge
xorg-libxau               1.0.9                h14c3975_0    conda-forge
xorg-libxdmcp             1.1.3                h516909a_0    conda-forge
xz                        5.2.4             h14c3975_1001    conda-forge
zeromq                    4.3.1             hf484d3e_1000    conda-forge
zipp                      0.5.1                      py_0    conda-forge
zlib                      1.2.11            h14c3975_1004    conda-forge
```

adam-urbanczyk, 2019-07-09:
Thanks for your detailed report @westurner . So to my understanding it 'works' with the style set to `fusion`, right? The segfault on maximizing might be actually related to this issue: #50. The solution (Ubuntu specific) is to install the following packages:
```
sudo apt install libglu1-mesa libgl1-mesa-dri mesa-common-dev libglu1-mesa-dev
```
Could you try an analogous solution on Fedora?

westurner, 2019-07-10:
Ah, thanks! I installed these packages on Fedora 29 and haven't seen a segfault as a result of resizing (as #50 "Sefault when resizing" more specifically describes)
```bash
dnf install -y mesa-libGLU mesa-libGL mesa-libGLU-devel
```

jmwright, 2019-11-18:
The instructions to correct this problem are included in the readme. @adam-urbanczyk can we close this?

adam-urbanczyk, 2019-11-18:
Indeed @jmwright , let's close this.

closed 2019-11-18 by adam-urbanczyk