CadQuery/CQ-editor #9: Abort on startup (BAD LEXICON descriptor)

johnbeard, 2018-09-05:
I get a crash when I start up on the current master (4e13229c50214d614d6ee3c14135394d4a4f62b6) on Arch:

```
python run.py 
INFO:OCC.Display.backend:backend loaded: qt-pyqt5
 ###### 3D rendering pipe initialisation #####
Display3d class initialization starting ...
Aspect_DisplayConnection created.
Graphic_Driver created.
V3d_Viewer created.
AIS_InteractiveContext created.
V3d_View created
Xw_Window created.
Display3d class successfully initialized.
 ########################################
Layer manager created
Layer dimensions: 100, 30
TKOpenGl | Type: Error | ID: 0 | Severity: High | Message:
  Failed to compile shader object. Compilation log:
0:264(2): error: `occFragColor' undeclared
0:264(2): error: value of type vec4 cannot be assigned to variable of type error

TKOpenGl | Type: Error | ID: 0 | Severity: High | Message:
  Error! FBO blitting has failed
DEBUG:asyncio:Using selector: EpollSelector
DEBUG:ipykernel.inprocess.ipkernel:
*** MESSAGE TYPE:kernel_info_request***
DEBUG:ipykernel.inprocess.ipkernel:   Content: {}
   --->
   
DEBUG:ipykernel.inprocess.ipkernel:kernel_info_request: {'header': {'version': '5.3', 'date': datetime.datetime(2018, 9, 5, 15, 14, 35, 655936, tzinfo=tzutc()), 'session': '17e1a50c-2ef784919e81f837a8330272', 'username': 'john', 'msg_type': 'kernel_info_request', 'msg_id': '3b1d660f-f7ab6b08bc6b55f4c1b21595'}, 'msg_id': '3b1d660f-f7ab6b08bc6b55f4c1b21595', 'msg_type': 'kernel_info_request', 'parent_header': {}, 'metadata': {}, 'content': {}, 'buffers': []}
DEBUG:ipykernel.inprocess.ipkernel:{'header': {'version': '5.3', 'date': datetime.datetime(2018, 9, 5, 15, 14, 35, 661540, tzinfo=datetime.timezone.utc), 'session': '17e1a50c-2ef784919e81f837a8330272', 'username': 'john', 'msg_type': 'kernel_info_reply', 'msg_id': '56430d19-c1135ee1952d0cf2c0c92052'}, 'msg_id': '56430d19-c1135ee1952d0cf2c0c92052', 'msg_type': 'kernel_info_reply', 'parent_header': {'version': '5.3', 'date': datetime.datetime(2018, 9, 5, 15, 14, 35, 655936, tzinfo=tzutc()), 'session': '17e1a50c-2ef784919e81f837a8330272', 'username': 'john', 'msg_type': 'kernel_info_request', 'msg_id': '3b1d660f-f7ab6b08bc6b55f4c1b21595'}, 'content': {'status': 'ok', 'protocol_version': '5.1', 'implementation': 'ipython', 'implementation_version': '6.5.0', 'language_info': {'name': 'python', 'version': '3.6.6', 'mimetype': 'text/x-python', 'codemirror_mode': {'name': 'ipython', 'version': 3}, 'pygments_lexer': 'ipython3', 'nbconvert_exporter': 'python', 'file_extension': '.py'}, 'banner': '', 'help_links': [{'text': 'Python Reference', 'url': 'https://docs.python.org/3.6'}, {'text': 'IPython Reference', 'url': 'https://ipython.org/documentation.html'}, {'text': 'NumPy Reference', 'url': 'https://docs.scipy.org/doc/numpy/reference/'}, {'text': 'SciPy Reference', 'url': 'https://docs.scipy.org/doc/scipy/reference/'}, {'text': 'Matplotlib Reference', 'url': 'https://matplotlib.org/contents.html'}, {'text': 'SymPy Reference', 'url': 'http://docs.sympy.org/latest/index.html'}, {'text': 'pandas Reference', 'url': 'https://pandas.pydata.org/pandas-docs/stable/'}]}, 'metadata': {}, 'tracker': <zmq.sugar.tracker.MessageTracker object at 0x7efd3f490c18>}
DEBUG:ipykernel.inprocess.ipkernel:
*** MESSAGE TYPE:kernel_info_request***
DEBUG:ipykernel.inprocess.ipkernel:   Content: {}
   --->
   
DEBUG:ipykernel.inprocess.ipkernel:kernel_info_request: {'header': {'version': '5.3', 'date': datetime.datetime(2018, 9, 5, 15, 14, 35, 664904, tzinfo=tzutc()), 'session': '17e1a50c-2ef784919e81f837a8330272', 'username': 'john', 'msg_type': 'kernel_info_request', 'msg_id': 'd8aded3b-5a6b66b1eef6c7e6e54b1995'}, 'msg_id': 'd8aded3b-5a6b66b1eef6c7e6e54b1995', 'msg_type': 'kernel_info_request', 'parent_header': {}, 'metadata': {}, 'content': {}, 'buffers': []}
DEBUG:ipykernel.inprocess.ipkernel:{'header': {'version': '5.3', 'date': datetime.datetime(2018, 9, 5, 15, 14, 35, 667502, tzinfo=datetime.timezone.utc), 'session': '17e1a50c-2ef784919e81f837a8330272', 'username': 'john', 'msg_type': 'kernel_info_reply', 'msg_id': '675f2227-f9bb28ae6315b6524ef285a6'}, 'msg_id': '675f2227-f9bb28ae6315b6524ef285a6', 'msg_type': 'kernel_info_reply', 'parent_header': {'version': '5.3', 'date': datetime.datetime(2018, 9, 5, 15, 14, 35, 664904, tzinfo=tzutc()), 'session': '17e1a50c-2ef784919e81f837a8330272', 'username': 'john', 'msg_type': 'kernel_info_request', 'msg_id': 'd8aded3b-5a6b66b1eef6c7e6e54b1995'}, 'content': {'status': 'ok', 'protocol_version': '5.1', 'implementation': 'ipython', 'implementation_version': '6.5.0', 'language_info': {'name': 'python', 'version': '3.6.6', 'mimetype': 'text/x-python', 'codemirror_mode': {'name': 'ipython', 'version': 3}, 'pygments_lexer': 'ipython3', 'nbconvert_exporter': 'python', 'file_extension': '.py'}, 'banner': '', 'help_links': [{'text': 'Python Reference', 'url': 'https://docs.python.org/3.6'}, {'text': 'IPython Reference', 'url': 'https://ipython.org/documentation.html'}, {'text': 'NumPy Reference', 'url': 'https://docs.scipy.org/doc/numpy/reference/'}, {'text': 'SciPy Reference', 'url': 'https://docs.scipy.org/doc/scipy/reference/'}, {'text': 'Matplotlib Reference', 'url': 'https://matplotlib.org/contents.html'}, {'text': 'SymPy Reference', 'url': 'http://docs.sympy.org/latest/index.html'}, {'text': 'pandas Reference', 'url': 'https://pandas.pydata.org/pandas-docs/stable/'}]}, 'metadata': {}, 'tracker': <zmq.sugar.tracker.MessageTracker object at 0x7efd3f490c18>}
DEBUG:ipykernel.inprocess.ipkernel:
*** MESSAGE TYPE:history_request***
DEBUG:ipykernel.inprocess.ipkernel:   Content: {'raw': True, 'output': False, 'hist_access_type': 'tail', 'n': 1000}
   --->
   
DEBUG:ipykernel.inprocess.ipkernel:history_request: {'header': {'version': '5.3', 'date': datetime.datetime(2018, 9, 5, 15, 14, 35, 668707, tzinfo=tzutc()), 'session': '17e1a50c-2ef784919e81f837a8330272', 'username': 'john', 'msg_type': 'history_request', 'msg_id': 'f60beaae-cb8207b52d14bb6745a4febf'}, 'msg_id': 'f60beaae-cb8207b52d14bb6745a4febf', 'msg_type': 'history_request', 'parent_header': {}, 'metadata': {}, 'content': {'raw': True, 'output': False, 'hist_access_type': 'tail', 'n': 1000}, 'buffers': []}
DEBUG:ipykernel.inprocess.ipkernel:{'header': {'version': '5.3', 'date': datetime.datetime(2018, 9, 5, 15, 14, 35, 672181, tzinfo=datetime.timezone.utc), 'session': '17e1a50c-2ef784919e81f837a8330272', 'username': 'john', 'msg_type': 'history_reply', 'msg_id': '0562323a-08fc4f03869dda3fd38231b1'}, 'msg_id': '0562323a-08fc4f03869dda3fd38231b1', 'msg_type': 'history_reply', 'parent_header': {'version': '5.3', 'date': datetime.datetime(2018, 9, 5, 15, 14, 35, 668707, tzinfo=tzutc()), 'session': '17e1a50c-2ef784919e81f837a8330272', 'username': 'john', 'msg_type': 'history_request', 'msg_id': 'f60beaae-cb8207b52d14bb6745a4febf'}, 'content': {'status': 'ok', 'history': []}, 'metadata': {}, 'tracker': <zmq.sugar.tracker.MessageTracker object at 0x7efd3f490c18>}
Traceback (most recent call last):
  File "/home/john/src/CQ-editor/src/widgets/viewer.py", line 175, in display_many
    context.Display(ais.GetHandle())
RuntimeError: Units_NoSuchType
BAD LEXICON descriptor
Aborted (core dumped)
```

GDB backtrace:

```
Thread 1 "python" received signal SIGABRT, Aborted.
0x00007ffff76e5d7f in raise () from /usr/lib/libc.so.6
(gdb) bt
#0  0x00007ffff76e5d7f in raise () from /usr/lib/libc.so.6
#1  0x00007ffff76d0672 in abort () from /usr/lib/libc.so.6
#2  0x00007ffff662bc3e in QMessageLogger::fatal(char const*, ...) const () from /home/john/.conda/envs/cqgui/lib/python3.6/site-packages/PyQt5/../../../libQt5Core.so.5
#3  0x00007ffff6cb69e0 in pyqt5_err_print() () from /home/john/.conda/envs/cqgui/lib/python3.6/site-packages/PyQt5/QtCore.so
#4  0x00007ffff6cb51c1 in qt_metacall_worker(_sipSimpleWrapper*, _typeobject*, _sipTypeDef*, QMetaObject::Call, int, void**) () from /home/john/.conda/envs/cqgui/lib/python3.6/site-packages/PyQt5/QtCore.so
#5  0x00007ffff6cb54d2 in qpycore_qobject_qt_metacall(_sipSimpleWrapper*, _sipTypeDef*, QMetaObject::Call, int, void**) () from /home/john/.conda/envs/cqgui/lib/python3.6/site-packages/PyQt5/QtCore.so
#6  0x00007ffff6833d59 in QMetaObject::activate(QObject*, int, int, void**) () from /home/john/.conda/envs/cqgui/lib/python3.6/site-packages/PyQt5/../../../libQt5Core.so.5
#7  0x00007ffff6cb18b9 in pyqtBoundSignal_emit () from /home/john/.conda/envs/cqgui/lib/python3.6/site-packages/PyQt5/QtCore.so
[... cut ...]
wheel                     0.31.1                   py36_1    conda-forge
wrapt                     1.10.11                  py36_0    conda-forge
xorg-libxau               1.0.8                h470a237_6    conda-forge
xorg-libxdmcp             1.1.2                h470a237_7    conda-forge
xz                        5.2.4                h470a237_1    conda-forge
zeromq                    4.2.5                hfc679d8_5    conda-forge
zlib                      1.2.11               h470a237_3    conda-forge
```

Let me know if I can provide more details!

adam-urbanczyk, 2018-09-11:
Thanks for trying and giving such a detailed bug report. Looking at this part
```
Layer manager created
Layer dimensions: 100, 30
TKOpenGl | Type: Error | ID: 0 | Severity: High | Message:
  Failed to compile shader object. Compilation log:
0:264(2): error: `occFragColor' undeclared
0:264(2): error: value of type vec4 cannot be assigned to variable of type error
```
I conclude that something goes wrong with initialization of the 3d display pipeline of PythonOCC and then I do not handle it well later in my code.

Could you give some details on your graphics card and the driver used? Also running 
```
qtdiag --gl-extensions
```
 might be useful.

johnbeard, 2018-09-20:
GTX card is:

```
01:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Redwood XT [Radeon HD 5670/5690/5730] (prog-if 00 [VGA controller])
	Subsystem: XFX Pine Group Inc. Redwood XT [Radeon HD 5670/5690/5730]
	Flags: bus master, fast devsel, latency 0, IRQ 27, NUMA node 0
	Memory at d0000000 (64-bit, prefetchable) [size=256M]
	Memory at fdec0000 (64-bit, non-prefetchable) [size=128K]
	I/O ports at ee00 [size=256]
	[virtual] Expansion ROM at 000c0000 [disabled] [size=128K]
	Capabilities: <access denied>
	Kernel driver in use: radeon
	Kernel modules: radeon
```

Some moaning about Vulkan in qtdiag - could be relevant?

```
$qtdiag --gl-extensions
Failed to create Vulkan instance: -9
Failed to create platform Vulkan instance
Qt 5.11.1 (x86_64-little_endian-lp64 shared (dynamic) release build; by GCC 8.2.0) on "xcb" 
OS: Antergos Linux [linux version 4.18.5-arch1-1-ARCH]

Architecture: x86_64; features: SSE2 SSE3 SSSE3 SSE4.1 SSE4.2 AVX

Environment:

Features: QT_NO_EXCEPTIONS

Library info:
  PrefixPath: /usr
  DocumentationPath: /usr/share/doc/qt
  HeadersPath: /usr/include/qt
  LibrariesPath: /usr/lib
  LibraryExecutablesPath: /usr/lib/qt/libexec
  BinariesPath: /usr/bin
  PluginsPath: /usr/lib/qt/plugins
  ImportsPath: /usr/lib/qt/imports
  Qml2ImportsPath: /usr/lib/qt/qml
  ArchDataPath: /usr/lib/qt
  DataPath: /usr/share/qt
  TranslationsPath: /usr/share/qt/translations
  ExamplesPath: /usr/share/doc/qt/examples
  TestsPath: /usr/tests
  SettingsPath: /etc/xdg

Standard paths [*...* denote writable entry]:
  DesktopLocation: "Desktop" */home/john/Desktop*
  DocumentsLocation: "Documents" */home/john/Documents*
  FontsLocation: "Fonts" */home/john/.local/share/fonts* /home/john/.fonts
  ApplicationsLocation: "Applications" */home/john/.local/share/applications* /usr/local/share/applications /usr/share/applications
  MusicLocation: "Music" */home/john/Music*
  MoviesLocation: "Movies" */home/john/Videos*
  PicturesLocation: "Pictures" */home/john/Pictures*
  TempLocation: "Temporary Directory" */tmp*
  HomeLocation: "Home" */home/john*
  AppLocalDataLocation: "Application Data" */home/john/.local/share/QtProject/qtdiag* /usr/local/share/QtProject/qtdiag /usr/share/QtProject/qtdiag
  CacheLocation: "Cache" */home/john/.cache/QtProject/qtdiag*
  GenericDataLocation: "Shared Data" */home/john/.local/share* /usr/local/share /usr/share
  RuntimeLocation: "Runtime" */run/user/1000*
  ConfigLocation: "Configuration" */home/john/.config* /etc/xdg
  DownloadLocation: "Download" */home/john/Downloads*
  GenericCacheLocation: "Shared Cache" */home/john/.cache*
  GenericConfigLocation: "Shared Configuration" */home/john/.config* /etc/xdg
  AppDataLocation: "Application Data" */home/john/.local/share/QtProject/qtdiag* /usr/local/share/QtProject/qtdiag /usr/share/QtProject/qtdiag
  AppConfigLocation: "Application Configuration" */home/john/.config/QtProject/qtdiag* /etc/xdg/QtProject/qtdiag

File selectors (increasing order of precedence):
  en_GB unix linux antergos
[... 321 lines cut ...]
  GL_MESA_window_pos
  GL_NVX_gpu_memory_info
  GL_NV_blend_square
  GL_NV_conditional_render
  GL_NV_depth_clamp
  GL_NV_fog_distance
  GL_NV_light_max_exponent
  GL_NV_packed_depth_stencil
  GL_NV_primitive_restart
  GL_NV_texgen_reflection
  GL_NV_texture_barrier
  GL_NV_texture_env_combine4
  GL_NV_texture_rectangle
  GL_NV_vdpau_interop
  GL_OES_EGL_image
  GL_OES_read_format
  GL_S3_s3tc
  GL_SGIS_generate_mipmap
  GL_SGIS_texture_border_clamp
  GL_SGIS_texture_edge_clamp
  GL_SGIS_texture_lod
  GL_SUN_multi_draw_arrays

Unable to create a Vulkan instance, error code is-9
```

jpmlt, 2018-10-03:
Hi Adam. I have same issue, running in Manjaro Linux (Arch based)
Attached my log files.

[log-condalist.log](https://github.com/CadQuery/CQ-editor/files/2440448/log-condalist.log)
[log-qtdiag.log](https://github.com/CadQuery/CQ-editor/files/2440449/log-qtdiag.log)
[log-runpy.log](https://github.com/CadQuery/CQ-editor/files/2440450/log-runpy.log)

justbuchanan, 2018-10-05:
I also ran into this issue on Arch Linux. I played around with it for a while and was able to narrow down a minimal change to get things to work. You can try it out at https://github.com/justbuchanan/CQ-editor/commit/cb316e0fc8740e2d7a4dc67e25f5dfececa7c73f.

I'm not familiar enough with the code to know why this fixes the problem and it's likely just a work around for some side-effects and not addressing the root issue. @adam-urbanczyk any thoughts on why this might fix the problem?

Something else I've noticed that may or not be related is that a lot of "DEBUG:ipykernel.inprocess.ipkernel: ..." messages are printed out at startup that contain content from other ipython notebooks I've run recently, which are independent of cadquery/cq-editor. @jpmlt's log-runpy.log seems to contain similar content.

adam-urbanczyk, 2018-10-05:
Wow thanks. I'll check out your fix during the weekend.

The logging noise is coming from the PythonOCC widget setting logging level to too low value AFAIK. I'll add an issue to track it.

jpmlt, 2018-10-05:
Hi Adam
I tried Justin workaround and it works also for me. So I will try you editor, thanks for your great work. @justbuchanan you're right, the file log-runpy.log also shows messages from previous run in jupyter notebook. 
Jean-Paul

adam-urbanczyk, 2018-10-05:
Great, please share your experience. I will check if the workaround is ok for CI too.

adam-urbanczyk, 2018-10-07:
@johnbeard does the workaround proposed by @justbuchanan  work for you too?

I investigated briefly and moving this line up
```python
self.components['object_tree'].addLines()
```
actually prevents adding the helper X,Y,Z lines to the view. Somehow this addition seems to trigger the error (on your PCs, I cannot unfortunately reproduce it).

@justbuchanan could you share what is your graphics card? I'm trying to find some common denominator here...

justbuchanan, 2018-10-07:
I looked into this some more and found similar issues (https://github.com/tpaviot/pythonocc-core/issues/279) reported in the pythonocc-core repo. It seems to be an issue with OCC finding data files that contain definitions of units for different locales.

I got this working on the master branch (without the hack I posted above) just by setting the `CASROOT` environment variable. With my setup this is:

```{.sh}
export CASROOT=/home/justin/.conda/envs/cqgui/share/oce-0.18/
```

Everything works great, including drawing of the helper axis lines!

adam-urbanczyk, 2018-10-07:
@justbuchanan  this is very interesting. I do not have `CASROOT` set in my env. If you have time, could you post strace output with and without `CASROOT` set?

If this is indeed the solution for everyone, I will change `run.py` accordingly.

justbuchanan, 2018-10-07:
I'm looking at this again and the default value I have for `CASROOT` is `/usr`. I'm not actually sure where this is exported from since I don't have it in my bashrc and don't see any relevant files pulled in that might define it. If I do `unset CASROOT`, it seems to work just as well as setting it to the path I mentioned above.

Here are strace logs with `CASROOT=/home/justin/.conda/envs/cqgui/share/oce-0.18/` and with it unset:
[strace-run-casroot.log](https://github.com/CadQuery/CQ-editor/files/2454451/run-casroot.log)
[strace-run-no-casroot.log](https://github.com/CadQuery/CQ-editor/files/2454452/run-no-casroot.log)

Also output from `qtdiag`:
[qtdiag.log](https://github.com/CadQuery/CQ-editor/files/2454453/qtdiag.log)

justbuchanan, 2018-10-07:
The relevant lines from `strace-run-no-casroot.log` (where CASROOT=/usr) seem to be:

~~~{.sh}
openat(AT_FDCWD, "/usr/src/UnitsAPI/Lexi_Expr.dat", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/src/UnitsAPI/Units.dat", O_RDONLY) = -1 ENOENT (No such file or directory)
~~~

justbuchanan, 2018-10-07:
It looks like the `export CASROOT=/usr` is coming from `/etc/profile.d/opencascade.sh` from the `opencascade` package, which is a dependency of freecad.

adam-urbanczyk, 2018-10-08:
Great, getting to the bottom of it. Does it work if you unset the environment variable?

adam-urbanczyk, 2018-10-09:
OK, I just committed a workaround (run.py unsets `CASROOT`) : 106036259c0a4ac519acc2d62710f8d826623ffd

@johnbeard @jpmlt  could you confirm that this works for you too?

@justbuchanan does this sound like Arch Linux bug to you? Any idea where to submit it?

justbuchanan, 2018-10-10:
That workaround looks good to me :+1:

adam-urbanczyk, 2018-10-11:
I looked quickly at the sources of 7.3 and the following fragment is quite revealing
```
// =======================================================================
// function : ShadersFolder
// purpose  :
// =======================================================================
const TCollection_AsciiString& Graphic3d_ShaderProgram::ShadersFolder()
{
  static Standard_Boolean        THE_IS_DEFINED = Standard_False;
  static TCollection_AsciiString THE_SHADERS_FOLDER;
  if (!THE_IS_DEFINED)
  {
    THE_IS_DEFINED = Standard_True;
    OSD_Environment aDirEnv ("CSF_ShadersDirectory");
    THE_SHADERS_FOLDER = aDirEnv.Value();
    if (THE_SHADERS_FOLDER.IsEmpty())
    {
      OSD_Environment aCasRootEnv ("CASROOT");
      THE_SHADERS_FOLDER = aCasRootEnv.Value();
      if (!THE_SHADERS_FOLDER.IsEmpty())
      {
        THE_SHADERS_FOLDER += "/src/Shaders";
      }

(...)
```
So to my understanding `$CASROOT/src/shaders` should contain the shader codes of OCCT. I think that in that case `CASROOT=\usr` is not the correct value  (should be smth like `\usr\OCCT\...`; I do not have ARCH linux locally). Anyhow that does not mean that `CASROOT` pointing to OCCT7.3 would not result in an issue. On the other hand I am not sure if `CASROOT` should be set at all. From the code above it seems that it is some kind of fallback.

@johnbeard  @jpmlt  would be great if you could confirm that this workaround is OK for you too.

justbuchanan, 2018-10-12:
Below is what `/etc/profile.d/opencascade.sh` file looks like. I agree that `CASROOT=/usr` is weird and it seems like it should be something like `/usr/OCCT`, however it looks like opencascade 7.3 just uses different paths. For example, `CSF_ShadersDirectory` expands to `/usr/share/opencascade/resources/Shaders`, which has the expected files in it.

~~~{.sh}
# /etc/profile.d/opencascade.sh from opencascade 7.3.0-3

export CASROOT=/usr

export CSF_LANGUAGE=us
export MMGT_CLEAR=1
export CSF_EXCEPTION_PROMPT=1

export CSF_OCCTResourcePath=$CASROOT/share/opencascade/resources
export CSF_SHMessage=$CSF_OCCTResourcePath/SHMessage
export CSF_MDTVTexturesDirectory=$CSF_OCCTResourcePath/Textures
export CSF_ShadersDirectory=$CSF_OCCTResourcePath/Shaders
export CSF_XSMessage=$CSF_OCCTResourcePath/XSMessage
export CSF_TObjMessage=$CSF_OCCTResourcePath/TObj
export CSF_StandardDefaults=$CSF_OCCTResourcePath/StdResource
export CSF_PluginDefaults=$CSF_OCCTResourcePath/StdResource
export CSF_XCAFDefaults=$CSF_OCCTResourcePath/StdResource
export CSF_TObjDefaults=$CSF_OCCTResourcePath/StdResource
export CSF_StandardLiteDefaults=$CSF_OCCTResourcePath/StdResource
export CSF_IGESDefaults=$CSF_OCCTResourcePath/XSTEPResource
export CSF_STEPDefaults=$CSF_OCCTResourcePath/XSTEPResource
export CSF_XmlOcafResource=$CSF_OCCTResourcePath/XmlOcafResource
export CSF_MIGRATION_TYPES=$CSF_OCCTResourcePath/StdResource/MigrationSheet.txt
export CSF_DrawPluginDefaults=$CSF_OCCTResourcePath/DrawResources
export DRAWHOME=$CSF_OCCTResourcePath/DrawResources
export DRAWDEFAULT=$CSF_OCCTResourcePath/DrawResources/DrawDefault
~~~

adam-urbanczyk, 2018-10-14:
Great, good to hear! I'll think about permanently adding the `QT_STYLE_OVERRIDE=fusion` to run.py but for now I am closing the issue. Thanks for trying out CQ-editor and for your extensive feedback!

closed 2018-10-14 by adam-urbanczyk

roipoussiere, 2019-08-06:
I'm facing to a similar issue. When I executing the `./runtests_locally.sh` script, almost all tests passes, only one fails:

```
_______________________________________________________________________ ERROR at setup of test_render ________________________________________________________________________
SETUP ERROR: Qt exceptions in virtual methods:
________________________________________________________________________________
Traceback (most recent call last):
  File "/home/nath/divers/CQ-editor/cq_editor/widgets/viewer.py", line 162, in display_many
    context.Display(ais)
RuntimeError: Units_NoSuchType
BAD LEXICON descriptor
```

(complete stacktrace [here](https://gist.github.com/roipoussiere/fc31bb9f6d8fd5cc87f72a72fcd943ae))

Same error when I run cq-editor directly.

I also use Manjaro Arch, with the last version of CQ-editor, and the `CASROOT` environment variable is set to `/usr`.

If I unset it, cq-editor doesn't start: the process is still alive, taking 100% of the cpu, but nothing append.

I tried to export `CASROOT` to `/usr/share/opencascade/`, `/usr/include/opencascade` and `/usr/local/include/oce` without more success. I don't know which path I'm supposed to use here and how to find it.

adam-urbanczyk, 2019-08-07:
I vaguely remember seeing this before. Could you try compiling oce exactly like here:
https://raw.githubusercontent.com/CadQuery/conda-packages/master/oce-0.18.3/build.sh