CadQuery/CQ-editor #11: Add option to autoreload file when it's changed by an external editor

justbuchanan, 2018-10-06:
Hi,
I'm a longtime user of OpenSCAD that's recently discovered cadquery and I'm pretty impressed! I've been interested in making use of cq-editor and although the builtin spyder editor is nice, I'd prefer to use my normal text editor setup. I'd like to have a workflow similar to OpenSCAD where I can have my text editor open next to cq-editor and have my changes reflected in the display whenever I save the file.

Is this something you'd be interested in incorporating into the project? If so, I'd be happy to put some work into adding this feature.

In OpenSCAD, there's a checkbox for "Automatic reload and preview" that turns this on and off. I'm thinking the option in cq-editor could look roughly the same. Are there any caveats that you can think of?

jmwright, 2018-10-06:
The FreeCAD workbench has this setting and quite a few people seem to use it.

https://github.com/jmwright/cadquery-freecad-module/blob/master/Settings.py#L2

The only drawback I've ever found is when working with scripts that take a long time to execute. It makes things a little cumbersome if you save often like I do.

adam-urbanczyk, 2018-10-07:
@justbuchanan yes, if you have something I can incorporate it. Adding tests will be highly appreciated.

I'd propose to add (to the editor component) a QFileSystemWatcher on the currently open file and an option in the preferences that will enable it. Let me know what is your plan.

adam-urbanczyk, 2018-10-07:
@jmwright good point. I added https://github.com/CadQuery/CQ-editor/issues/14 to solve this (potential) problem in the future

justbuchanan, 2018-10-07:
Sounds good. I have this partially working using a QFileSystemWatcher within the editor component as you describe. I'll clean up the code and add some tests, then send you a pull request.

+1 for only connecting components via signals/slots, that seems like a clean design.

adam-urbanczyk, 2018-10-07:
Sounds great!

justbuchanan, 2018-10-19:
Completed in #15.

closed 2018-10-19 by justbuchanan