CadQuery/CQ-editor #17: CQparts support

adam-urbanczyk, 2018-10-22:


justbuchanan, 2019-01-18:
I was able to get this working with a helper function that traverses the cqparts Assembly/Part tree and displays each one. I'm not sure what the best way to implement this with cqgi would be, but here's what I used for reference:

```python
import cadquery as cq
import cqparts
import car # cqparts car example

 def show_cqparts_obj(cqobj):
    cqobj.build(recursive=True)
    def _show_recursive(cqobj, prefix=''):
        if isinstance(cqobj, cqparts.Part):
            # cq-editor detects cadquery objects in the global namespace
            globals()[prefix] = cqobj.world_obj
        elif isinstance(cqobj, cqparts.Assembly):
            if prefix: prefix = prefix + "_"
            for name, subobj in cqobj.components.items():
                _show_recursive(subobj, prefix + name)
    _show_recursive(cqobj)

 show_cqparts_obj(car.Car())
```

jmwright, 2019-01-18:
In the FreeCAD workbench, I would just import and use the cqparts.display method. cqparts has the concept of an execution environment, and there's one for FreeCAD, which is why that worked.

https://github.com/cqparts/cqparts/blob/master/src/cqparts/display/freecad.py

So there are probably two ways to go.
1. Modify `show_object` to properly support cqparts.
2. Add a CQ-editor environment to cqparts.

The first one has the advantage that it makes the scripts more portable. However, I seem to remember there being some previous discussion on why that was more complicated than it sounds. @fragmuffin can you shed any light on this?

jmwright, 2019-01-18:
Potentially relevant discussions:

https://github.com/dcowden/cadquery/issues/273

https://github.com/dcowden/cadquery/issues/279

adam-urbanczyk, 2019-01-18:
@justbuchanan looks good, but let's think how to get rid of `globals`

adam-urbanczyk, 2019-01-21:
@fragmuffin  @justbuchanan if you could add a function to cqparts that flattens Part/Assembly to a flat list of:
``` 
[(CQ object, properties dict),...]
```
that would be it I think. Then it would be really easy to display it in CQ-editor.

zignig, 2019-01-21:
I have an extractor in cqparts_bucket that flattents a cqparts assembly.

https://github.com/zignig/cqparts_bucket/blob/master/svg_test.py#L17

adam-urbanczyk, 2019-01-21:
OK good, I can add it to `cq_utils` submodule in this project and extend `show_object` accordingly. Does everyone agree that it is the way to go?

jmwright, 2019-01-22:
No objections.

justbuchanan, 2019-01-23:
Sounds good to me.

A couple random thoughts:
1. It would be nice to give each part a human-readable name. I was doing this before by just concatenating a part's "component" name with those of its parent components/assemblies.
2. For now, a flat list of parts in the "CQ models" pane should be fine. Eventually though, it would be nice to consider a tree view that reflects the structure of the assembly.

jmwright, 2019-05-12:
Adding a couple more links to this issue.

https://github.com/CadQuery/cqparts

https://groups.google.com/forum/m/#!topic/cadquery/cgdmkO5oq8E

Where does CQParts support for CadQuery 2.0 stand?

fragmuffin, 2019-05-26:
@jmwright 

> Where does CQParts support for CadQuery 2.0 stand?

The most suitable description would be "long overdue".
But perhaps a more useful description is "on its way".

I've started a PR cqparts/cqparts#155, will be working on it this week.

jmwright, 2019-05-26:
@fragmuffin Awesome, thanks! There will be plenty of excited users when cqparts is compatible with CQ 2.0 and Python-OCC.

fragmuffin, 2019-05-27:
Yes! I agree.
It's actually passing all the core tests now, in the `feature/freecad-occ` branch.
I've just got to put the hours in to release it.... but I'll definitely be focusing on that this week.

jmwright, 2025-01-28:
cq-parts has not been active for years now. Closing this issue.

closed 2025-01-28 by jmwright