CadQuery/CQ-editor #19: Staying Up To Date

jmwright, 2018-12-29:
There were changes made recently in CadQuery/cadquery that I would like to have access to. What's the best/proper way to pull the latest changes from the lib into the Conda environment for this UI?

jmwright, 2018-12-29:
I don't know if it's the best way, but this method works.
```
pip install --upgrade git+https://github.com/CadQuery/cadquery.git
```

adam-urbanczyk, 2018-12-29:
It works fine with pip, but I'd like to handle this via conda.

jmwright, 2018-12-30:
It looks like the .travis.yml file does an Anaconda deploy. Is there a command I can run to do the update of the cadquery lib using conda?

justbuchanan, 2019-01-18:
@jmwright see https://github.com/CadQuery/cadquery/issues/59. Travis CI does do an Anaconda deploy but only for tagged commits/releases. The linux builds of cadquery right now are fairly out of date.

adam-urbanczyk, 2019-06-04:
There is now preliminary support for checking for new CQ versions availability.

closed 2019-06-04 by adam-urbanczyk