CadQuery/CQ-editor #23: Automatic reload and preview Option Scrolls Code Editor Back to the Top on Each Save

jmwright, 2019-01-17:
When the `Automatic reload and preview` menu item is checked, clicking save will scroll the contents of the code editor back to the top, no matter how far down the user has scrolled. This causes the user to lose the spot where they were editing the script. I think that this feature really only needs to reload the 3D view, and can leave the code editor window as-is. I believe that when using an external editor would be the only time the code editor would need to reload.

justbuchanan, 2019-01-18:
That's a good point, it is annoying that it loses your place each time it refreshes. I hadn't run into it because I was mostly using this feature with the builtin editor pane closed :).

Right now, there's a single "file changed" signal that triggers a file reload and rerender. It would be better to split this into "file changed internally" and "file changed externally" so that internal changes don't reload the editor. I can look into getting this fixed.

adam-urbanczyk, 2019-04-23:
Solved by #36

closed 2019-04-23 by adam-urbanczyk