CadQuery/CQ-editor #24: Allow import of modules in the same directory as the current script

rowanG077, 2019-02-12:
I have some helper scripts that I use for multiple parts and they are in the same directory as my actual part python file. In FreeCad I can just import that helper script without issue however in the CQ-editor the helper script cannot be import as it throws a `ModuleNotFoundError`. I would really like this feature in the CQ-editor

.

jmwright, 2019-02-13:
@rowanG077 Does it work if you append the current directory to `PYTHONPATH` before running CQ-editor?
On Linux/MacOS:
```bash
PYTHONPATH=$PYTHONPATH:/your/script/dir python run.py
```

adam-urbanczyk, 2019-02-13:
@rowanG077 I indeed do not add explicitly location of the current script t python path. It does sound like a useful feature.

PR #28 (2019-03-06): CQ reloading and pythonpath modification
WIP

This PR will resolve #24 and #27
closed 2019-03-06 by adam-urbanczyk