CadQuery/CQ-editor #31: Workplane can not be stored for later use

rowanG077, 2019-03-20:
I finally got CQ-editor to work after completely reinstalling anaconda. I have been testing out parts i have allready made and almost none of them work. One of the main culprits is that I store workplanes for later use, in FreeCAD with cadquery this allowed me to modify the part using that stored workplane. However cadquery with CQ-editor creates a new part and doesn't modify the original. Which means this technique doesn't work.

See this example:

```
import cadquery as cq
result = cq.Workplane("XY" )\
    .box(10, 10, 1)

top_workplane = result.faces(">Z").workplane()

top_workplane.transformed(
        rotate = (0, 0, 0)
    )\
    .transformed(
        offset = (0, 3, 0),
    )\
    .rect(1, 1)\
    .cutThruAll()
    
top_workplane.transformed(
        rotate = (0, 0, 120)
    )\
    .transformed(
        offset = (0, 3, 0),
    )\
    .rect(1, 1)\
    .cutThruAll()
    
top_workplane.transformed(
        rotate = (0, 0, 240)
    )\
    .transformed(
        offset = (0, 3, 0),
    )\
    .rect(1, 1)\
    .cutThruAll()

show_object(result)
```

In this part I store the top workplane and use that workplane to go to three different locations and cut a rectangle out of the box. In FreeCAD this works and I get the expected part. in CQ-editor the result part is not modified and thus the created holes are not shown.

see the attached image of FreeCAD and CQ-editor for the difference.

[image]

[image]

adam-urbanczyk, 2019-03-20:
@rowanG077 glad to hear that CQ-editor works! Regarding the issue - this is actually expected behavior for CQ2. Every operation creates new workplane object which holds reference to its parent. You need to store the result explicitly:
```python
top_workplane = top_workplane.transformed(rotate = (0, 0, 0))\
    .transformed(offset = (0, 3, 0))\
    .rect(1, 1)\
    .cutThruAll()
```

The rationale for this change was ease of debugging. Every operation is now stored on the stack and you can inspect all the steps using the `Inspect CQ object` function.

rowanG077, 2019-03-20:
I see. So is there no other way to somehow store a known point from which to do multiple things? In the If I store the new workplane and then use that then of course I get a completely different part since I rotate and offset the workplane each time.

Basically if I want to old behavior I would have to undo all transforms each time I perform this. This vastly increases the code size and complexity.

Would the "Tagged entities" feature help with this. It's mention on the documentation page: https://dcowden.github.io/cadquery/roadmap.html.

rowanG077, 2019-03-21:
Maybe something like explicitly storing a workplane and then restoring it into a part could work. Something like this:

```
top_workplane = result.faces(">Z").workplane().store()

result = top_workplane.transformed(rotate = (0, 0, 0))\
    .transformed(offset = (0, 3, 0))\
    .rect(1, 1)\
    .cutThruAll()

result = result.loadWorkplane(top_workplane)
    .sphere(1)
```

I don't have any clue about the internals of cadquery or pythonocc to say whether this is possible at all but something like this seems to be the most elegant while preserving that every operation is now stored on the stack.

adam-urbanczyk, 2019-03-21:
With the current API I was able to come up with the following solution:
```python

from cadquery import *

class Rect(object):
    
    a = 0.8
    ix = 0
    w = cq.Wire.makePolygon(map(Vector,[(-a,a,0),(a,a,0),(a,-a,0),(-a,-a,0),(-a,a,0)]))
    
    @classmethod
    def makeRect(cls,pt):
        
        rv = cls.w.rotate((0,0,0),(0,0,1),cls.ix*120).translate(pt)
        cls.ix+=1
        
        return rv

r = cq.Workplane("XY" )\
    .box(10, 10, 1).faces(">Z")\
    .polarArray(3.5,0,360,3).eachpoint(Rect.makeRect).cutThruAll()

show_object(r)
```
[image]

I must say I do not fully like it. We do have a `polarArray`, but no clean way to rotate the object at each point. @jmwright @dcowden any thoughts on how to improve this?

jmwright, 2019-03-22:
@rowanG077 @adam-urbanczyk I'm not sure that I have much that's useful to add. There has been discussion before about using copy.copy and copy.deepcopy as @rowanG077 is using store() in the code above (I think), but I'm not sure of the implications. 

I haven't spent enough time with the CQ 2.0 API to really come up with a cleaner or more concise solution than @adam-urbanczyk posted. His solution does have a nice code re-use aspect to it. The RectMaker class can be modified and reused for any number of use cases. There's more boilerplate with Adam's code, but less repetition than with the original example. I'm not sure how helpful this all is for @rowanG077 though.

rowanG077, 2019-03-22:
@jmwright @adam-urbanczyk 

Thank you for providing a solution. In this case this works fine of course. But in general I use storing workplanes for multiple reasons. For instance in one of the parts I do something like this example and then keep on using that workplane to add other things to the other. Sometimes it's very impractical to obtain the exact some workplane again because the part has been modified.

Something I have been thinking about is the implement a transform stack. Then I can obtain a workplane, transform it and then undo those transforms to obtain the original workplane. This would require monkey patching a lot of methods though which also sucks. (center, transform...)

jmwright, 2019-03-22:
> Sometimes it's very impractical to obtain the exact some workplane again because the part has been modified.

By default, CadQuery uses the center of mass to determine the center of a workplane. This leads to some unintuitive behavior when you remove material from a model in an asymmetrical way, because that center moves. Using the geometric bounding box helps combat that. Also, using the direction Nth selectors, along with the logical selector operators can help you get back to an original location of a workplane.

adam-urbanczyk, 2019-03-22:
@rowanG077 here is yet another solution that is closest to your design intent. I must say that personally I'd prefer to go the `polarArray` way and maybe extend the API of CQ to allow for rotation of each wire.

I will close this issue. If you want to continue or have similar issues regarding CQ2.0 please use the cadquery repo https://github.com/CadQuery/cadquery.

```python
import cadquery as cq
result = cq.Workplane("XY" )\
    .box(10, 10, 1)

top_workplane = result.faces(">Z").workplane()
starting_plane = top_workplane.plane 

top_workplane = top_workplane.transformed(
        rotate = (0, 0, 0)
    )\
    .transformed(
        offset = (0, 3, 0),
    )\
    .rect(1, 1)\
    .cutThruAll()

top_workplane.plane = starting_plane
top_workplane = top_workplane.transformed(
        rotate = (0, 0, 120)
    )\
    .transformed(
        offset = (0, 3, 0),
    )\
    .rect(1, 1)\
    .cutThruAll()

top_workplane.plane = starting_plane
top_workplane = top_workplane.transformed(
        rotate = (0, 0, 240)
    )\
    .transformed(
        offset = (0, 3, 0),
    )\
    .rect(1, 1)\
    .cutThruAll()

show_object(top_workplane)
```

[image]

closed 2019-03-22 by adam-urbanczyk

rowanG077, 2019-03-22:
@adam-urbanczyk Yeah in this case using `polarArray` is better. But it's in some cases to have a constant starting point. In some parts I do over 10 things on a single workplane and use transforms each time to get to the correct locations. Rewritting would have been a nightmare.

Thank you for your quick support!