CadQuery/CQ-editor #34: Highlight object in view when selected in CQ models panel

rowanG077, 2019-03-26:
Currently when there are multiple objects there are just some numbers in the CQ models panels without a way to find out what numbers corresponds to what model. If there is a simple highlight of the model in the view when the model is selected either in the view or in the panel would really help with this.

adam-urbanczyk, 2019-04-07:
@rowanG077 thanks for reporting this and #33. Always good to improve usability!

In the meantime I added an option in #36 to name displayed objects like so:
```python
result = cq.Workplane("XY" ).box(3, 3, 0.5).edges("|Z").fillet(0.125)
show_object(result,'box')
```
[image]

adam-urbanczyk, 2019-04-23:
Solved by #36

closed 2019-04-23 by adam-urbanczyk