CadQuery/CQ-editor #42: Keystrokes Ignored in Editor

jmwright, 2019-05-12:
I can open the script and highlight a line with the mouse, but no keystrokes seem to make it into the editor, not even arrow key presses. I haven't had this issue before with CQ-editor.

Ubuntu 18.04 64-bit, latest master of CQ-editor, CadQuery master installed via pip.

jmwright, 2019-05-12:
I don't know, maybe something is broken on my system. I checked out a much earlier commit, rebuilt the environment, and still have the same problem.

adam-urbanczyk, 2019-05-13:
@jmwright not sure if I understand the issue. Looks like the OCC widget grabs focus to quickly. I did modify some code in our build of pythonOCC so this could be the root cause. Could you confirm that I understand you correctly? Does this still happen after clicking on the editor?

jmwright, 2019-05-13:
@adam-urbanczyk Correct, it happens even after clicking on/in the editor. It's like the editor has focus, and accepts mouse input, but ignores keyboard input.

adam-urbanczyk, 2019-05-13:
Would this line be an explanation?

https://github.com/CadQuery/pythonocc-core/blob/41fbec4ca3da8e5050bc84d064f8f6bcf743ac55/src/Display/qtDisplay.py#L50

jmwright, 2019-05-13:
I can comment that out in Anaconda's copy of the file here in a little while and try it.

Do you not have this same issue? You run Linux, correct?

adam-urbanczyk, 2019-05-13:
Nope, I cannot reproduce it.

jmwright, 2019-05-13:
Is there a setting that could cause this behavior? I changed the auto-reload setting, thinking maybe that was it. It didn't have any effect though.

jmwright, 2019-05-13:
I commented that line out of the environment's qtDisplay.py, and it didn't help. I need to put some debugging statements in to log out when keys are pressed to make sure the editor sees them.

I'm on the master branch of CQ-editor. Not sure if you're running on a branch or not.

adam-urbanczyk, 2019-05-14:
Will check that. In the meantime could you post which qt and pyqt version you are using ( via conda list )?

adam-urbanczyk, 2019-05-14:
I pulled latest master and still cannot reproduce it.

adam-urbanczyk, 2019-05-15:
@jmwright  any new insights? I am really interested in your `conda list`. Also you could try running `qtdiag`.

jmwright, 2019-05-15:
@adam-urbanczyk Here's what `conda list` shows:
```
$ conda list
# packages in environment at /home/jwright/miniconda3/envs/cqgui:
#
# Name                    Version                   Build  Channel
alabaster                 0.7.12                     py_0    conda-forge
asn1crypto                0.24.0                py36_1003    conda-forge
astroid                   2.2.5                    py36_0    conda-forge
attrs                     19.1.0                     py_0    conda-forge
babel                     2.6.0                      py_1    conda-forge
backcall                  0.1.0                      py_0    conda-forge
backports.os              0.1.1                 py36_1000    conda-forge
ca-certificates           2019.3.9             hecc5488_0    conda-forge
cadquery                  0.5-SNAPSHOT             pypi_0    pypi
certifi                   2019.3.9                 py36_0    conda-forge
cffi                      1.12.3           py36h8022711_0    conda-forge
chardet                   3.0.4                 py36_1003    conda-forge
cloudpickle               1.0.0                      py_0    conda-forge
cryptography              2.6.1            py36h72c5cf5_0    conda-forge
dbus                      1.13.6               he372182_0    conda-forge
decorator                 4.4.0                      py_0    conda-forge
defusedxml                0.5.0                      py_1    conda-forge
docutils                  0.14                  py36_1001    conda-forge
entrypoints               0.3                   py36_1000    conda-forge
expat                     2.2.5             hf484d3e_1002    conda-forge
fontconfig                2.13.1            he4413a7_1000    conda-forge
freeimageplus             3.17.0                        0    CadQuery
freetype                  2.10.0               he983fc9_0    conda-forge
gettext                   0.19.8.1          hc5be6a0_1002    conda-forge
glib                      2.58.3            hf63aee3_1001    conda-forge
gst-plugins-base          1.14.4            hdf3bae2_1001    conda-forge
gstreamer                 1.14.4            h66beb1c_1001    conda-forge
icu                       58.2              hf484d3e_1000    conda-forge
idna                      2.8                   py36_1000    conda-forge
imagesize                 1.1.0                      py_0    conda-forge
importlib_metadata        0.11                     py36_0    conda-forge
ipykernel                 5.1.0           py36h24bf2e0_1002    conda-forge
ipython                   7.5.0            py36h24bf2e0_0    conda-forge
ipython_genutils          0.2.0                      py_1    conda-forge
isort                     4.3.18                   py36_0    conda-forge
jedi                      0.13.3                   py36_0    conda-forge
jinja2                    2.10.1                     py_0    conda-forge
jpeg                      9c                h14c3975_1001    conda-forge
jsonschema                3.0.1                    py36_0    conda-forge
jupyter_client            5.2.4                      py_3    conda-forge
jupyter_core              4.4.0                      py_0    conda-forge
lazy-object-proxy         1.4.1            py36h516909a_0    conda-forge
libblas                   3.8.0                9_openblas    conda-forge
libcblas                  3.8.0                9_openblas    conda-forge
libffi                    3.2.1             he1b5a44_1006    conda-forge
libgcc-ng                 8.2.0                hdf63c60_1  
libgfortran-ng            7.3.0                hdf63c60_0  
libiconv                  1.15              h516909a_1005    conda-forge
liblapack                 3.8.0                9_openblas    conda-forge
libpng                    1.6.37               hed695b0_0    conda-forge
libsodium                 1.0.16            h14c3975_1001    conda-forge
libstdcxx-ng              8.2.0                hdf63c60_1  
libuuid                   2.32.1            h14c3975_1000    conda-forge
libxcb                    1.13              h14c3975_1002    conda-forge
libxml2                   2.9.9                h13577e0_0    conda-forge
logbook                   1.4.3            py36h516909a_0    conda-forge
markupsafe                1.1.1            py36h14c3975_0    conda-forge
mccabe                    0.6.1                      py_1    conda-forge
mistune                   0.8.4           py36h14c3975_1000    conda-forge
nbconvert                 5.5.0                      py_0    conda-forge
nbformat                  4.4.0                      py_1    conda-forge
ncurses                   6.1               hf484d3e_1002    conda-forge
numpy                     1.16.3           py36he5ce36f_0    conda-forge
numpydoc                  0.9.1                      py_0    conda-forge
[... 45 lines cut ...]
sphinxcontrib-devhelp     1.0.1                      py_0    conda-forge
sphinxcontrib-htmlhelp    1.0.2                      py_0    conda-forge
sphinxcontrib-jsmath      1.0.1                      py_0    conda-forge
sphinxcontrib-qthelp      1.0.2                      py_0    conda-forge
sphinxcontrib-serializinghtml 1.1.1                      py_0    conda-forge
spyder                    3.2.8                    py36_0    conda-forge
sqlite                    3.26.0            h67949de_1001    conda-forge
tbb                       2019.6               hc9558a2_0    conda-forge
[... cut ...]
wrapt                     1.11.1           py36h14c3975_0    conda-forge
xorg-libxau               1.0.9                h14c3975_0    conda-forge
xorg-libxdmcp             1.1.3                h516909a_0    conda-forge
xz                        5.2.4             h14c3975_1001    conda-forge
zeromq                    4.3.1             hf484d3e_1000    conda-forge
zipp                      0.4.0                      py_0    conda-forge
zlib                      1.2.11            h14c3975_1004    conda-forge
```

jmwright, 2019-05-15:
Oh, and here's output from qtdiag (maybe the XKB errors could be related?):
```
$ qtdiag
xkbcommon: ERROR: failed to add default include path 
Qt: Failed to create XKB context!
Use QT_XKB_CONFIG_ROOT environmental variable to provide an additional search path, add ':' as separator to provide several search paths and/or make sure that XKB configuration data directory contains recent enough contents, to update please see http://cgit.freedesktop.org/xkeyboard-config/ .
Qt 5.9.7 (x86_64-little_endian-lp64 shared (dynamic) release build; by GCC 7.3.0) on "xcb" 
OS: Ubuntu 18.04.2 LTS [linux version 4.15.0-48-generic]

Architecture: x86_64; features: SSE2 SSE3 SSSE3 SSE4.1 SSE4.2 AVX AVX2 RTM HLE

Environment:
  QT4_IM_MODULE="xim"
  QT_ACCESSIBILITY="1"
  QT_IM_MODULE="ibus"

Features: QT_NO_EXCEPTIONS

Library info:
  PrefixPath: /home/jwright/miniconda3/envs/cqgui
  DocumentationPath: /home/jwright/miniconda3/envs/cqgui/doc
  HeadersPath: /home/jwright/miniconda3/envs/cqgui/include/qt
  LibrariesPath: /home/jwright/miniconda3/envs/cqgui/lib
  LibraryExecutablesPath: /home/jwright/miniconda3/envs/cqgui/libexec
  BinariesPath: /home/jwright/miniconda3/envs/cqgui/bin
  PluginsPath: /home/jwright/miniconda3/envs/cqgui/plugins
  ImportsPath: /home/jwright/miniconda3/envs/cqgui/imports
  Qml2ImportsPath: /home/jwright/miniconda3/envs/cqgui/qml
  ArchDataPath: /home/jwright/miniconda3/envs/cqgui
  DataPath: /home/jwright/miniconda3/envs/cqgui
  TranslationsPath: /home/jwright/miniconda3/envs/cqgui/translations
  ExamplesPath: /home/jwright/miniconda3/envs/cqgui/examples
  TestsPath: /home/jwright/miniconda3/envs/cqgui/tests
  SettingsPath: /home/jwright/miniconda3/envs/cqgui

Standard paths [*...* denote writable entry]:
  DesktopLocation: "Desktop" */home/jwright/Desktop*
  DocumentsLocation: "Documents" */home/jwright/Documents*
  FontsLocation: "Fonts" */home/jwright/.local/share/fonts* /home/jwright/.fonts
  ApplicationsLocation: "Applications" */home/jwright/.local/share/applications* /usr/share/ubuntu/applications /home/jwright/.local/share/flatpak/exports/share/applications /var/lib/flatpak/exports/share/applications /usr/local/share/applications /usr/share/applications /var/lib/snapd/desktop/applications
  MusicLocation: "Music" */home/jwright/Music*
  MoviesLocation: "Movies" */home/jwright/Videos*
  PicturesLocation: "Pictures" */home/jwright/Pictures*
  TempLocation: "Temporary Directory" */tmp*
  HomeLocation: "Home" */home/jwright*
  AppLocalDataLocation: "Application Data" */home/jwright/.local/share/QtProject/qtdiag* /usr/share/ubuntu/QtProject/qtdiag /home/jwright/.local/share/flatpak/exports/share/QtProject/qtdiag /var/lib/flatpak/exports/share/QtProject/qtdiag /usr/local/share/QtProject/qtdiag /usr/share/QtProject/qtdiag /var/lib/snapd/desktop/QtProject/qtdiag
  CacheLocation: "Cache" */home/jwright/.cache/QtProject/qtdiag*
  GenericDataLocation: "Shared Data" */home/jwright/.local/share* /usr/share/ubuntu /home/jwright/.local/share/flatpak/exports/share /var/lib/flatpak/exports/share /usr/local/share /usr/share /var/lib/snapd/desktop
  RuntimeLocation: "Runtime" */run/user/1000*
  ConfigLocation: "Configuration" */home/jwright/.config* /etc/xdg/xdg-ubuntu /etc/xdg
  DownloadLocation: "Download" */home/jwright/Downloads*
  GenericCacheLocation: "Shared Cache" */home/jwright/.cache*
  GenericConfigLocation: "Shared Configuration" */home/jwright/.config* /etc/xdg/xdg-ubuntu /etc/xdg
  AppDataLocation: "Application Data" */home/jwright/.local/share/QtProject/qtdiag* /usr/share/ubuntu/QtProject/qtdiag /home/jwright/.local/share/flatpak/exports/share/QtProject/qtdiag /var/lib/flatpak/exports/share/QtProject/qtdiag /usr/local/share/QtProject/qtdiag /usr/share/QtProject/qtdiag /var/lib/snapd/desktop/QtProject/qtdiag
  AppConfigLocation: "Application Configuration" */home/jwright/.config/QtProject/qtdiag* /etc/xdg/xdg-ubuntu/QtProject/qtdiag /etc/xdg/QtProject/qtdiag

File selectors (increasing order of precedence):
  en_US unix linux ubuntu

Network:
  Using "OpenSSL 1.1.1b  26 Feb 2019", version: 0x1010102f

Platform capabilities: ThreadedPixmaps OpenGL WindowMasks MultipleWindows ForeignWindows NonFullScreenWindows NativeWidgets WindowManagement SyncState RasterGLSurface SwitchableWidgetComposition

Style hints:
  mouseDoubleClickInterval: 400
  mousePressAndHoldInterval: 800
  startDragDistance: 10
  startDragTime: 500
  startDragVelocity: 0
[... 44 lines cut ...]
  HighlightedText: #ffffffff
  Link: #ff0000ff
  LinkVisited: #ffff00ff
  AlternateBase: #fff7f5f3
  NoRole: #ff000000
  ToolTipBase: #ffffffdc
  ToolTipText: #ff000000

Screens: 1, High DPI scaling: inactive
# 0 "eDP-1" Depth: 24 Primary: yes
  Manufacturer: 
  Model: 
  Serial number: 
  Geometry: 1920x1080+0+0 Available: 1853x1053+67+27
  Physical size: 345x194 mm  Refresh: 120 Hz Power state: 0
  Physical DPI: 141.357,141.402 Logical DPI: 96,96 Subpixel_None
  DevicePixelRatio: 1 Pixel density: 1
  Primary orientation: 2 Orientation: 2 Native orientation: 0 OrientationUpdateMask: 0

LibGL Vendor: Intel Open Source Technology Center
Renderer: Mesa DRI Intel(R) HD Graphics 620 (Kaby Lake GT2) 
Version: 3.0 Mesa 18.2.8
Shading language: 1.30
Format: Version: 3.0 Profile: 0 Swap behavior: 0 Buffer size (RGB): 8,8,8
```

adam-urbanczyk, 2019-05-22:
Indeed, after googling it looks like this is a known issue (https://stackoverflow.com/questions/26974644/no-keyboard-input-in-qt-creator-after-update-to-qt5). I do not understand the root cause yet. Could you check which xkb related packages are installed on your system?
```
apt list --installed | grep xkb
```

jmwright, 2019-05-22:
```
$ apt list --installed | grep xkb

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libxcb-xkb1/bionic-updates,now 1.13-2~ubuntu18.04 amd64 [installed]
libxkbcommon-x11-0/bionic-updates,bionic-security,now 0.8.0-1ubuntu0.1 amd64 [installed]
libxkbcommon0/bionic-updates,bionic-security,now 0.8.0-1ubuntu0.1 amd64 [installed]
libxkbfile1/bionic,now 1:1.0.9-2 amd64 [installed]
x11-xkb-utils/bionic-updates,now 7.7+3ubuntu0.18.04.1 amd64 [installed]
xkb-data/bionic,bionic,now 2.23.1-1ubuntu1 all [installed]
```

I can confirm that setting the environment variable as suggested in the accepted answer on StackOverflow fixes the problem. That gives me a work-around for now.

adam-urbanczyk, 2019-05-22:
I think I understand the root cause: it seems that qt build from conda-forge is "sub-optimally" configured (the default xkb search path of qt is defined at build-time). Solution is to use qt from the defaults channel. I will adjust the env file to reflect it and maybe open an issue in conda-forge.

BTW: this issue allowed my to find what is going on:
https://sft.its.cern.ch/jira/browse/SPI-1042

adam-urbanczyk, 2019-05-22:
@jmwright  could try with the latest env file? For me it solves the issue, but it'd be nice to have a confirmation.

jmwright, 2019-05-24:
@adam-urbanczyk I can confirm that it is fixed. Thanks!

PR #43 (2019-05-22): Update env file to use defaults as well
This is should resolve #42
commit b0f54ba8 (2019-05-22): Update env file to use defaults as well

This is should resolve #42
closed 2019-05-24 by jmwright