CadQuery/CQ-editor #45: CQ-editor does not start in Windows anymore

tolot27, 2019-05-29:
Since 5df9a51 CQ-editor does not start in Windows anymore. There are a lot of warnings after starting the editor using `python run.py` but they do not differ between the previous commit 7ca1b92 except the last additional line `Namespace(filename=None)`:

```
C:\ProgramData\Miniconda3\envs\cqgui\lib\site-packages\OCC\AIS.py:3: DeprecationWarning: OCC.AIS is deprecated since pythonocc-0.18.2. Use OCC.Core.AIS
  warnings.warn("OCC.AIS is deprecated since pythonocc-0.18.2. Use OCC.Core.AIS", DeprecationWarning)
C:\ProgramData\Miniconda3\envs\cqgui\lib\site-packages\OCC\Quantity.py:3: DeprecationWarning: OCC.Quantity is deprecated since pythonocc-0.18.2. Use OCC.Core.Quantity
  warnings.warn("OCC.Quantity is deprecated since pythonocc-0.18.2. Use OCC.Core.Quantity", DeprecationWarning)
C:\ProgramData\Miniconda3\envs\cqgui\lib\site-packages\OCC\Aspect.py:3: DeprecationWarning: OCC.Aspect is deprecated since pythonocc-0.18.2. Use OCC.Core.Aspect
  warnings.warn("OCC.Aspect is deprecated since pythonocc-0.18.2. Use OCC.Core.Aspect", DeprecationWarning)
C:\ProgramData\Miniconda3\envs\cqgui\lib\site-packages\OCC\Geom.py:3: DeprecationWarning: OCC.Geom is deprecated since pythonocc-0.18.2. Use OCC.Core.Geom
  warnings.warn("OCC.Geom is deprecated since pythonocc-0.18.2. Use OCC.Core.Geom", DeprecationWarning)
C:\ProgramData\Miniconda3\envs\cqgui\lib\site-packages\OCC\gp.py:3: DeprecationWarning: OCC.gp is deprecated since pythonocc-0.18.2. Use OCC.Core.gp
  warnings.warn("OCC.gp is deprecated since pythonocc-0.18.2. Use OCC.Core.gp", DeprecationWarning)
C:\ProgramData\Miniconda3\envs\cqgui\lib\site-packages\OCC\BRepBuilderAPI.py:3: DeprecationWarning: OCC.BRepBuilderAPI is deprecated since pythonocc-0.18.2. Use OCC.Core.BRepBuilderAPI
  warnings.warn("OCC.BRepBuilderAPI is deprecated since pythonocc-0.18.2. Use OCC.Core.BRepBuilderAPI", DeprecationWarning)
Layer manager created
Layer dimensions: 100, 30
C:\ProgramData\Miniconda3\envs\cqgui\lib\site-packages\qtconsole\jupyter_widget.py:561: DeprecationWarning: ConsoleWidget._style_sheet_changed is deprecated in traitlets 4.1: use @observe and @unobserve instead.
  def _style_sheet_changed(self):
C:\ProgramData\Miniconda3\envs\cqgui\lib\site-packages\jupyter_client\session.py:371: DeprecationWarning: Session._key_changed is deprecated in traitlets 4.1: use @observe and @unobserve instead.
  def _key_changed(self):
C:\ProgramData\Miniconda3\envs\cqgui\lib\site-packages\ipykernel\datapub.py:5: DeprecationWarning: ipykernel.datapub is deprecated. It has moved to ipyparallel.datapub
  warnings.warn("ipykernel.datapub is deprecated. It has moved to ipyparallel.datapub", DeprecationWarning)
C:\ProgramData\Miniconda3\envs\cqgui\lib\site-packages\ipykernel\serialize.py:7: DeprecationWarning: ipykernel.serialize is deprecated. It has moved to ipyparallel.serialize
  warnings.warn("ipykernel.serialize is deprecated. It has moved to ipyparallel.serialize", DeprecationWarning)
C:\ProgramData\Miniconda3\envs\cqgui\lib\site-packages\ipykernel\pickleutil.py:8: DeprecationWarning: ipykernel.pickleutil is deprecated. It has moved to ipyparallel.
  warnings.warn("ipykernel.pickleutil is deprecated. It has moved to ipyparallel.", DeprecationWarning)
C:\ProgramData\Miniconda3\envs\cqgui\lib\site-packages\ipykernel\codeutil.py:17: DeprecationWarning: ipykernel.codeutil is deprecated since IPykernel 4.3.1. It has moved to ipyparallel.serialize
  warnings.warn("ipykernel.codeutil is deprecated since IPykernel 4.3.1. It has moved to ipyparallel.serialize", DeprecationWarning)
C:\ProgramData\Miniconda3\envs\cqgui\lib\site-packages\ipykernel\iostream.py:66: DeprecationWarning: zmq.eventloop.ioloop is deprecated in pyzmq 17. pyzmq now works with default tornado and asyncio eventloops.
  self.io_loop = IOLoop(make_current=False)
C:\ProgramData\Miniconda3\envs\cqgui\lib\site-packages\jupyter_client\session.py:803: DeprecationWarning: Accessing zmq Socket attribute recv_multipart on BackgroundSocket
  msg_list = socket.recv_multipart(mode, copy=copy)
Namespace(filename=None)
```

Installation was done as described in the README.md.

tolot27, 2019-05-29:
I could narrow it down to https://github.com/CadQuery/CQ-editor/blob/07e6eb0b5077393187b68b0a6f24a682e4c65373/src/widgets/editor.py#L122. If I comment out this line, the editor starts.

tolot27, 2019-05-29:
The problem occured becase the original source file has moved to another location. 

BTW: Where are the component states stored?

adam-urbanczyk, 2019-05-30:
Thanks for investigating this. State is saved using QSettings object which on Windows uses the registry I think. Btw this issue is not Windows specific and is solved in #44. 

Regarding excessive logging this should be solved in the latest build of the pythonocc conda package on the cadquery channel

adam-urbanczyk, 2019-05-31:
Solved in #44

closed 2019-05-31 by adam-urbanczyk