CadQuery/CQ-editor #50: Segfault when resizing

rowanG077, 2019-06-18:
CQ-gui seg faults for me when resizing the window.

tonious, 2019-06-18:
I also observed this on Ubuntu 18.04.

adam-urbanczyk, 2019-06-18:
Thanks for reporting this issue. Could you provide the output from:

conda info 
conda list

It would be nice to have a  backtrace from a debugger too.

rowanG077, 2019-06-19:
Output from `conda info`:

```
     active environment : cqgui
    active env location : /home/rowangoemans/anaconda3/envs/cqgui
            shell level : 2
       user config file : /home/rowangoemans/.condarc
 populated config files : 
          conda version : 4.6.11
    conda-build version : 3.17.8
         python version : 3.7.3.final.0
       base environment : /home/rowangoemans/anaconda3  (writable)
           channel URLs : https://repo.anaconda.com/pkgs/main/linux-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/free/linux-64
                          https://repo.anaconda.com/pkgs/free/noarch
                          https://repo.anaconda.com/pkgs/r/linux-64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /home/rowangoemans/anaconda3/pkgs
                          /home/rowangoemans/.conda/pkgs
       envs directories : /home/rowangoemans/anaconda3/envs
                          /home/rowangoemans/.conda/envs
               platform : linux-64
             user-agent : conda/4.6.11 requests/2.21.0 CPython/3.7.3 Linux/4.18.0-21-generic ubuntu/18.04.2 glibc/2.27
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False
```

Output from `conda list`:

```
WARNING: The conda.compat module is deprecated and will be removed in a future release.
# packages in environment at /home/rowangoemans/anaconda3/envs/cqgui:
#
# Name                    Version                   Build  Channel
alabaster                 0.7.12                   py36_0  
asn1crypto                0.24.0                   py36_0  
astroid                   2.2.5                    py36_0  
attrs                     19.1.0                   py36_1  
babel                     2.7.0                      py_0  
backcall                  0.1.0                    py36_0  
backports                 1.0                        py_2  
backports.os              0.1.1                    py36_0  
blas                      1.0                         mkl  
bleach                    3.1.0                    py36_0  
ca-certificates           2019.5.15                     0  
cadquery                  0.5-SNAPSHOT             pypi_0    pypi
certifi                   2019.3.9                 py36_0  
cffi                      1.12.3           py36h2e261b9_0  
chardet                   3.0.4                    py36_1  
cloudpickle               1.1.1                      py_0  
cryptography              2.7              py36h1ba5d50_0  
dbus                      1.13.6               h746ee38_0  
decorator                 4.4.0                    py36_1  
defusedxml                0.6.0                      py_0  
docutils                  0.14                     py36_0  
entrypoints               0.3                      py36_0  
expat                     2.2.6                he6710b0_0  
fontconfig                2.13.0               h9420a91_0  
freeimageplus             3.17.0                        0    CadQuery
freetype                  2.9.1                h8a8886c_1  
glib                      2.56.2               hd408876_0  
gmp                       6.1.2                h6c8ec71_1  
gst-plugins-base          1.14.0               hbbd80ab_1  
gstreamer                 1.14.0               hb453b48_1  
icu                       58.2                 h9c2bf20_1  
idna                      2.8                      py36_0  
imagesize                 1.1.0                    py36_0  
importlib_metadata        0.17                     py36_1  
[... 226 lines cut ...]
#57 0x00007ffff524aeb7 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /home/rowangoemans/anaconda3/envs/cqgui/lib/python3.6/site-packages/PyQt5/../../../libQt5Core.so.5
#58 0x00007ffff524eaab in QCoreApplication::exec() () from /home/rowangoemans/anaconda3/envs/cqgui/lib/python3.6/site-packages/PyQt5/../../../libQt5Core.so.5
#59 0x00007ffff5f1c970 in meth_QApplication_exec_ () from /home/rowangoemans/anaconda3/envs/cqgui/lib/python3.6/site-packages/PyQt5/QtWidgets.so
#60 0x0000555555665681 in _PyCFunction_FastCallDict () at /tmp/build/80754af9/python_1546130271559/work/Objects/methodobject.c:234
#61 0x00005555556ec42c in call_function () at /tmp/build/80754af9/python_1546130271559/work/Python/ceval.c:4851
#62 0x000055555571138a in _PyEval_EvalFrameDefault () at /tmp/build/80754af9/python_1546130271559/work/Python/ceval.c:3335
#63 0x00005555556e653b in _PyFunction_FastCall (globals=<optimized out>, nargs=0, args=<optimized out>, co=<optimized out>) at /tmp/build/80754af9/python_1546130271559/work/Python/ceval.c:4933
#64 fast_function () at /tmp/build/80754af9/python_1546130271559/work/Python/ceval.c:4968
#65 0x00005555556ec505 in call_function () at /tmp/build/80754af9/python_1546130271559/work/Python/ceval.c:4872
#66 0x000055555571138a in _PyEval_EvalFrameDefault () at /tmp/build/80754af9/python_1546130271559/work/Python/ceval.c:3335
#67 0x00005555556e7289 in _PyEval_EvalCodeWithName (qualname=0x0, name=<optimized out>, closure=0x0, kwdefs=0x0, defcount=0, defs=0x0, kwstep=2, kwcount=<optimized out>, kwargs=0x0, kwnames=0x0, argcount=0, args=0x0, locals=0x7ffff7f50168, globals=0x7ffff7f50168, _co=0x7ffff7f0c930)
    at /tmp/build/80754af9/python_1546130271559/work/Python/ceval.c:4166
#68 PyEval_EvalCodeEx () at /tmp/build/80754af9/python_1546130271559/work/Python/ceval.c:4187
[... cut ...]
#74 0x000055555576e4b3 in run_file (p_cf=0x7fffffffdb5c, filename=0x5555558aa7a0 L"run.py", fp=0x5555559568f0) at /tmp/build/80754af9/python_1546130271559/work/Modules/main.c:340
#75 Py_Main () at /tmp/build/80754af9/python_1546130271559/work/Modules/main.c:811
#76 0x000055555563702e in main () at /tmp/build/80754af9/python_1546130271559/work/Programs/python.c:69
#77 0x00007ffff77e6b97 in __libc_start_main (main=0x555555636f40 <main>, argc=2, argv=0x7fffffffdd68, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffdd58) at ../csu/libc-start.c:310
#78 0x0000555555717e0e in _start () at ../sysdeps/x86_64/elf/start.S:103
(gdb) 
```

tonious, 2019-06-19:
On my end:

Output from `conda info`

~~~~~~
     active environment : cqgui
    active env location : /home/tthompson/anaconda3/envs/cqgui
            shell level : 1
       user config file : /home/tthompson/.condarc
 populated config files :
          conda version : 4.6.14
    conda-build version : 3.18.2
         python version : 3.6.8.final.0
       base environment : /home/tthompson/anaconda3  (writable)
           channel URLs : https://repo.anaconda.com/pkgs/main/linux-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/free/linux-64
                          https://repo.anaconda.com/pkgs/free/noarch
                          https://repo.anaconda.com/pkgs/r/linux-64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /home/tthompson/anaconda3/pkgs
                          /home/tthompson/.conda/pkgs
       envs directories : /home/tthompson/anaconda3/envs
                          /home/tthompson/.conda/envs
               platform : linux-64
             user-agent : conda/4.6.14 requests/2.22.0 CPython/3.6.8 Linux/4.15.0-51-generic ubuntu/18.04.2 glibc/2.27
                UID:GID : 1153436474:1153434113
             netrc file : None
           offline mode : False
~~~~~~

Output from `conda list`:

~~~~~~
# packages in environment at /home/tthompson/anaconda3/envs/cqgui:
#
# Name                    Version                   Build  Channel
alabaster                 0.7.12                   py36_0  
asn1crypto                0.24.0                   py36_0  
astroid                   2.2.5                    py36_0  
attrs                     19.1.0                   py36_1  
babel                     2.7.0                      py_0  
backcall                  0.1.0                    py36_0  
backports                 1.0                        py_2  
backports.os              0.1.1                    py36_0  
blas                      1.0                         mkl  
bleach                    3.1.0                    py36_0  
ca-certificates           2019.5.15                     0  
cadquery                  0.5-SNAPSHOT             pypi_0    pypi
certifi                   2019.3.9                 py36_0  
cffi                      1.12.3           py36h2e261b9_0  
chardet                   3.0.4                    py36_1  
cloudpickle               1.1.1                      py_0  
cqparts                   0.2.2.dev1               pypi_0    pypi
cryptography              2.7              py36h1ba5d50_0  
dbus                      1.13.6               h746ee38_0  
decorator                 4.4.0                    py36_1  
defusedxml                0.6.0                      py_0  
docutils                  0.14                     py36_0  
entrypoints               0.3                      py36_0  
expat                     2.2.6                he6710b0_0  
fontconfig                2.13.0               h9420a91_0  
freeimageplus             3.17.0                        0    CadQuery
freetype                  2.9.1                h8a8886c_1  
glib                      2.56.2               hd408876_0  
gmp                       6.1.2                h6c8ec71_1  
gst-plugins-base          1.14.0               hbbd80ab_1  
gstreamer                 1.14.0               hb453b48_1  
icu                       58.2                 h9c2bf20_1  
idna                      2.8                      py36_0  
[... 231 lines cut ...]
#57 0x00007ffff524beb7 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /home/tthompson/anaconda3/envs/cqgui/lib/python3.6/site-packages/PyQt5/../../../libQt5Core.so.5
#58 0x00007ffff524faab in QCoreApplication::exec() () from /home/tthompson/anaconda3/envs/cqgui/lib/python3.6/site-packages/PyQt5/../../../libQt5Core.so.5
#59 0x00007ffff5f1d970 in meth_QApplication_exec_ () from /home/tthompson/anaconda3/envs/cqgui/lib/python3.6/site-packages/PyQt5/QtWidgets.so
#60 0x0000555555665681 in _PyCFunction_FastCallDict () at /tmp/build/80754af9/python_1546130271559/work/Objects/methodobject.c:234
#61 0x00005555556ec42c in call_function () at /tmp/build/80754af9/python_1546130271559/work/Python/ceval.c:4851
#62 0x000055555571138a in _PyEval_EvalFrameDefault () at /tmp/build/80754af9/python_1546130271559/work/Python/ceval.c:3335
#63 0x00005555556e653b in _PyFunction_FastCall (globals=<optimized out>, nargs=0, args=<optimized out>, co=<optimized out>) at /tmp/build/80754af9/python_1546130271559/work/Python/ceval.c:4933
#64 fast_function () at /tmp/build/80754af9/python_1546130271559/work/Python/ceval.c:4968
#65 0x00005555556ec505 in call_function () at /tmp/build/80754af9/python_1546130271559/work/Python/ceval.c:4872
#66 0x000055555571138a in _PyEval_EvalFrameDefault () at /tmp/build/80754af9/python_1546130271559/work/Python/ceval.c:3335
#67 0x00005555556e7289 in _PyEval_EvalCodeWithName (qualname=0x0, name=<optimized out>, closure=0x0, kwdefs=0x0, defcount=0, defs=0x0, kwstep=2, kwcount=<optimized out>, kwargs=0x0, kwnames=0x0, argcount=0, args=0x0,
    locals=0x7ffff7f483f0, globals=0x7ffff7f483f0, _co=0x7ffff7f05ae0) at /tmp/build/80754af9/python_1546130271559/work/Python/ceval.c:4166
#68 PyEval_EvalCodeEx () at /tmp/build/80754af9/python_1546130271559/work/Python/ceval.c:4187
[... cut ...]
#74 0x000055555576e4b3 in run_file (p_cf=0x7fffffffd69c, filename=0x5555558a96d0 L"run.py", fp=0x55555593a0e0) at /tmp/build/80754af9/python_1546130271559/work/Modules/main.c:340
#75 Py_Main () at /tmp/build/80754af9/python_1546130271559/work/Modules/main.c:811
#76 0x000055555563702e in main () at /tmp/build/80754af9/python_1546130271559/work/Programs/python.c:69
#77 0x00007ffff77e6b97 in __libc_start_main (main=0x555555636f40 <main>, argc=2, argv=0x7fffffffd8a8, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffd898) at ../csu/libc-start.c:310
#78 0x0000555555717e0e in _start () at ../sysdeps/x86_64/elf/start.S:103
(gdb)
~~~~~~

adam-urbanczyk, 2019-06-20:
Thanks, I can reproduce this on appveyor now.

adam-urbanczyk, 2019-06-20:
It seems installing additional libraries with apt solves the issue. Cf #51 last commit. Could you try if this solves the issue on your machines too?

rowanG077, 2019-06-20:
It no longer freezes with the additional libraries for me.

adam-urbanczyk, 2019-06-21:
Great, thanks for checking. I'll edit the README to indicate that additional libs might be needed.

adam-urbanczyk, 2019-06-25:
Updated the readme file

closed 2019-06-25 by adam-urbanczyk

westurner, 2019-07-10:
I installed these packages on Fedora 29 and haven't seen a segfault as a result of resizing:
```bash
dnf install -y mesa-libGLU mesa-libGL mesa-libGLU-devel
```