CadQuery/CQ-editor #52: Missing InitDriver call in viewer.py

rowanG077, 2019-06-21:
I get CQ-editor working with Nix. I had to patch the code though and I think this is a bug in CQ-editor. Once I had installed all dependencies and I ran CQ-editor I got the following error:

```
nix-shell:~/cadquery-nix]$ /nix/store/wcsmqla0193mkfvavsxpvjxpi90456hb-CQ-editor/bin/CQ-editor.sh
INFO:OCC.Display.backend:backend loaded: qt-pyqt5
/nix/store/zq1vrmls3sl2sj32502gj85vmjbr42fp-pythonocc-core-cadquery-version/lib/python3.7/site-packages/OCC/AIS.py:3: Depre
  warnings.warn("OCC.AIS is deprecated since pythonocc-0.18.2. Use OCC.Core.AIS", DeprecationWarning)
/nix/store/zq1vrmls3sl2sj32502gj85vmjbr42fp-pythonocc-core-cadquery-version/lib/python3.7/site-packages/OCC/Quantity.py:3:
  warnings.warn("OCC.Quantity is deprecated since pythonocc-0.18.2. Use OCC.Core.Quantity", DeprecationWarning)
/nix/store/zq1vrmls3sl2sj32502gj85vmjbr42fp-pythonocc-core-cadquery-version/lib/python3.7/site-packages/OCC/Geom.py:3: Depr
  warnings.warn("OCC.Geom is deprecated since pythonocc-0.18.2. Use OCC.Core.Geom", DeprecationWarning)
/nix/store/zq1vrmls3sl2sj32502gj85vmjbr42fp-pythonocc-core-cadquery-version/lib/python3.7/site-packages/OCC/gp.py:3: Deprec
  warnings.warn("OCC.gp is deprecated since pythonocc-0.18.2. Use OCC.Core.gp", DeprecationWarning)
/nix/store/zq1vrmls3sl2sj32502gj85vmjbr42fp-pythonocc-core-cadquery-version/lib/python3.7/site-packages/OCC/BRepBuilderAPI.
  warnings.warn("OCC.BRepBuilderAPI is deprecated since pythonocc-0.18.2. Use OCC.Core.BRepBuilderAPI", DeprecationWarning)
Traceback (most recent call last):
  File "/nix/store/wcsmqla0193mkfvavsxpvjxpi90456hb-CQ-editor/repo/run.py", line 7, in <module>
    main()
  File "/nix/store/wcsmqla0193mkfvavsxpvjxpi90456hb-CQ-editor/repo/src/main.py", line 16, in main
    win = MainWindow()
  File "/nix/store/wcsmqla0193mkfvavsxpvjxpi90456hb-CQ-editor/repo/src/main_window.py", line 39, in __init__
    self.viewer = OCCViewer(self)
  File "/nix/store/wcsmqla0193mkfvavsxpvjxpi90456hb-CQ-editor/repo/src/widgets/viewer.py", line 62, in __init__
    self.updatePreferences()
  File "/nix/store/wcsmqla0193mkfvavsxpvjxpi90456hb-CQ-editor/repo/src/widgets/viewer.py", line 71, in updatePreferences
    self.canvas._display.View.SetBgGradientColors(color1,color2,True)
AttributeError: 'NoneType' object has no attribute 'View'
```

I looked into the sources of pythonocc and indeed as fast as I can tell there is an `InitDriver()` call missing in CQ-editor. See https://github.com/tpaviot/pythonocc-core/blob/0.18.1/src/Display/qtDisplay.py#L134. Once I added `self.canvas.InitDriver()` in the constructor of the `OCCViewer` class the problem was fixed and I can run CQ-editor.

I'm hesitant on making a pull request because I'm not sure if this is correct since I don't have further experience with pythonocc. Please advice.

adam-urbanczyk, 2019-06-21:
This is not the case: CQ-editor is using a custom fork of pythonocc and
oce. Check out our repos.

adam-urbanczyk, 2019-06-21:
If you want to build this fork take a look at conda-packages repo in the organization. AFAIR  cmake is used to do that.

rowanG077, 2019-06-24:
Yeah I got it to work with the custom pythonocc version. I can now fully run cadquery an cq-editor without conda which is great.

closed 2019-06-24 by rowanG077

adam-urbanczyk, 2019-06-25:
@rowanG077 glad that it worked out for you. It is obviously your choice, but I'd give conda a try. I did not hear any specifics regarding why your use case would not work with a fully local install of conda.