CadQuery/CQ-editor #53: Support complete show_object syntax

adam-urbanczyk, 2019-06-25:


adam-urbanczyk, 2019-06-26:
#54 partially solves this - specifying options shouldn't cause errors anymore but for now it will have no effect.

Duckle29, 2019-06-27:
What is the current method of specifying color and name using show_object?

adam-urbanczyk, 2019-06-28:
Color is not implemented ATM. Name like so: `show_object(obj,name='name')`

adam-urbanczyk, 2019-06-30:
@Duckle29 to my understanding CQ does not specify what could be in the options dict. Do you have any preferences?

Duckle29, 2019-07-01:
@adam-urbanczyk I think it should be CQ1 compatible. I know CQ1 had an rgba option:

```python
options = { "rgba":(100, 100, 100, 0.0) } 
```

I never used others

adam-urbanczyk, 2019-07-01:
@jmwright any input on this?

jmwright, 2019-07-01:
I think that an integrated rgba option was the only one ever supported by the FreeCAD module. It would be nice to maintain backward compatibility, but it's not completely necessary to do so. I could maybe see having a separate alpha option being nice too. Sometimes you don't want to have to figure out the color code just to adjust the transparency to check things like hole depth. I'm not sure I'd use all three forms of entering a color code enough to justify the extra development time and complexity. I don't have much of an opinion I guess, other than being able to just enter "orange" instead of looking up the color code might be a nice-to-have feature.

adam-urbanczyk, 2019-07-03:
Alright, I'll add `argba`, `color` and `alpha`. For now only after 0.1 is released.

PR #127 (2020-03-26): Full syntax of show_object
Will resolve #53
closed 2020-03-26 by adam-urbanczyk