CadQuery/CQ-editor #60: How to animate object?

dimonbavly, 2019-07-11:
When I try animate this CQ -editor(or CQ) give too many objects. Although I assumed there would be an animation of box.
`import cadquery as cq
t =3
while t<=10:
    
    result = cq.Workplane("XY" ).box(t, t, t).edges("|Z").fillet((t**(1/2))/2)
    show_object(result)
    t = t + 0.1`
How can i implement this?
Is it possible to animate the kinematic slice operation? I want to write programs for CNC for multi-axis processing.

jmwright, 2019-07-12:
@dimonbavly Every call of `show_object` causes the CadQuery Gateway Interface (CQGI) to add a copy of the object to a list of objects to be rendered. The assumption was that separate objects would be included by the user, not the same object multiple times in different orientations.

@adam-urbanczyk It's an interesting idea to have a simple animation mode that would display all the objects handed back by CQGI in order, with a preset delay between each keyframe.

@fragmuffin Have you done any work with animation in cqparts?

adam-urbanczyk, 2019-07-12:
@dimonbavly as you noticed every call to `show_object` results in a new object in the object tree. There is currently no functionality to support what you need. You could implement animation by manipulating the `AIS_ColoredShape` objects that are used by the OCC kernel for displaying. For experiments you can use the embedded python console:

```python
from OCC.gp import *

T = gp_Trsf()
T.SetRotation(gp_Ax1(gp_Pnt(0., 0., 0.), gp_Dir(0., 0., 1.)), 90)

obj_tree = self.components['object_tree']
viewer =  self.components['viewer']
obj = obj_tree.CQ.child(0)
obj.ais.SetLocalTransformation(T)
viewer.redraw()
``` 

I think it is a nice idea to enable animation in CQ-editor and I think it would fit best in a separate component (PRs are very welcome!). I could add some signals and slots to the `object_tree` component to facilitate this.

prst, 2019-07-17:
Now I too very interested to implement animation in FreeCAD by using CQ. 
Yesterday try first time and.... was found problems if try to "import serial" (described here https://groups.google.com/forum/#!topic/cadquery/ijDfBsfr6SI )
So if do:
>>> import serial
got message
Traceback (most recent call last):
  File "<input>", line 1, in <module>
ModuleNotFoundError: No module named 'serial'
and in 'report view' message
_Contents of /home/stan/FreeCAD/CadQuery/new_test_cadquery.py changed, reloading 
Executing CQGI-compliant script.
Error executing CQGI-compliant script. No module named 'serial'
Executed /home/stan/FreeCAD/CadQuery/new_test_cadquery.py_

but is OK for
>>> import os
>>> import sys
>>> import time

Another point:
If edit source_code.py file in external editor and save - immediately changes is applied, it look like an animation. 
As new idea it is using open pipe-files instead of serial, and use "import os, sys, time"

jmwright, 2019-07-18:
@prst What version of FreeCAD and the CadQuery FreeCAD module are you using? I can import the serial module fine on my system.

Also, unless you have a compelling reason to stay with FreeCAD, you might want to think about switching to CQ-editor (the project within this repo) and CadQuery 2.0, which aren't FreeCAD based anymore. Almost all effort will be put into them going forward.

prst, 2019-07-18:
Issue with serial was solves. Tho main points:
1) Not work if use FreeCAD as FreeCAD_0.18-16117-Linux-Conda_Py3Qt5_glibc2.12-x86_64.AppImage by reason using Python sandbox without support of serial
If use freecad (as /usr/bin/freecad) from APT-repositories - all OK, because used system Python
2) Need to do - pip3 uninstall serial

About idea with open files or pipe - Tested this:
Open file and get data from file
...
        length      = 60.0       # Length of the block
        height      = 40.0       # Height of the block
        thickness   = 10.0    # Thickness of the block

        filepath = '~/FreeCAD/test.txt'
        cnt = 0
        #while True:
            with open(filepath) as fp:
                line = fp.readline()
                while line:
                    dat=line.strip()
                    #print( "Line {}: {}".format( cnt, line.strip() ) )
                    print( "Line {}: {}".format( cnt, dat ) )
                    line = fp.readline()
                    cnt += 1
                    #r = cq.Workplane("XY").box( length, height, thickness)
                    r = cq.Workplane("XY").box( int(dat), height, thickness)
                    show_object(r)
...
But here main loop not work  #while True:
It is look like need to implement refresh