CadQuery/CQ-editor #62: Colored STEP export

adam-urbanczyk, 2019-07-13:


tanius, 2021-01-11:
This issue probably relates to exporting STEP files with the coloring defined via CQ-editor's [`show_object()` function](https://github.com/CadQuery/CQ-editor#showing-objects).

However for those looking for colored STEP export, I just wanted to note here that it is already possible via CadQuery assemblies now, independent of CQ-editor. One would wrap a part in the assembly, and set the color on the assembly level. This coloring then is also exported to the STEP file ([source](https://github.com/CadQuery/cadquery/issues/20#issuecomment-753975315)).

adam-urbanczyk, 2021-01-11:
Yes, the issues is about CQ-editor UI supporting it.