CadQuery/CQ-editor #73: Mac installation problem

Zhongy1, 2019-07-19:
Someone I know introduced me to CadQuery, so I am trying to install it for myself on my Mac.
I installed Miniconda and installed CadQuery with this command:
`conda install -c conda-forge -c cadquery cadquery=2`
Installation was fine, but when I try to install CQ-editor, I get this error...

Command: `conda install -c cadquery cq-editor`

Collecting package metadata (current_repodata.json): done
Solving environment: failed with current_repodata.json, will retry with next repodata source.
Initial quick solve with frozen env failed.  Unfreezing env and trying again.
Solving environment: failed with current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed
Initial quick solve with frozen env failed.  Unfreezing env and trying again.
Solving environment: failed`

UnsatisfiableError: The following specifications were found to be incompatible with each other:

Package pyqtgraph conflicts for:
cq-editor -> pyqtgraph=0.10.0
Package logbook conflicts for:
cq-editor -> logbook
Package pyparsing conflicts for:
cq-editor -> pyparsing
Package pip conflicts for:
python=3.6 -> pip
Package xz conflicts for:
python=3.6 -> xz[version='>=5.2.3,<6.0a0|>=5.2.4,<6.0a0']
Package libcxx conflicts for:
python=3.6 -> libcxx[version='>=4.0.1']
Package libffi conflicts for:
python=3.6 -> libffi[version='3.2.*|>=3.2.1,<4.0a0']
Package cadquery conflicts for:
cq-editor -> cadquery=2
Package openssl conflicts for:
python=3.6 -> openssl[version='1.0.*|1.0.*,>=1.0.2l,<1.0.3a|>=1.0.2m,<1.0.3a|>=1.0.2n,<1.0.3a|>=1.0.2o,<1.0.3a|>=1.1.1a,<1.1.2a']
Package spyder conflicts for:
cq-editor -> spyder=3.3.4
Package qtconsole conflicts for:
cq-editor -> qtconsole=4.4.4
Package pyqt conflicts for:
cq-editor -> pyqt=5.9.2
Package requests conflicts for:
cq-editor -> requests
Package tk conflicts for:
python=3.6 -> tk[version='8.6.*|>=8.6.7,<8.7.0a0|>=8.6.8,<8.7.0a0']
Package readline conflicts for:
python=3.6 -> readline[version='7.*|>=7.0,<8.0a0']
Package sqlite conflicts for:
python=3.6 -> sqlite[version='>=3.20.1,<4.0a0|>=3.22.0,<4.0a0|>=3.23.1,<4.0a0|>=3.24.0,<4.0a0|>=3.25.2,<4.0a0|>=3.26.0,<4.0a0']
Package zlib conflicts for:
python=3.6 -> zlib[version='>=1.2.11,<1.3.0a0']
Package path.py conflicts for:
cq-editor -> path.py
Package ncurses conflicts for:
python=3.6 -> ncurses[version='>=6.0,<7.0a0|>=6.1,<7.0a0']

Not sure how to go about dealing with this problem

adam-urbanczyk, 2019-07-19:
Does the following work

conda install -c cadquery -c conda-forge cq-editor

Zhongy1, 2019-07-19:
Yes, that works. I can now use the cq-editor. Thanks!

adam-urbanczyk, 2019-07-19:
Great, I'll update the README.

closed 2019-07-19 by adam-urbanczyk