CadQuery/CQ-editor #74: Crash at start

roipoussiere, 2019-08-04:
After downloaded [the last release](https://github.com/CadQuery/CQ-editor/releases/tag/0.1RC1), extracted the archive and ran `./CQ-editor`, I get the following error:

```
[11718] mod is NULL - structTraceback (most recent call last):
  File "/home/adam/anaconda2/envs/cgqui4/lib/python3.6/struct.py", line 14, in <module>
ImportError: cannot import name '_clearcache'
[11718] mod is NULL - pyimod02_archiveTraceback (most recent call last):
  File "/home/adam/anaconda2/envs/cgqui4/lib/python3.6/site-packages/PyInstaller/loader/pyimod02_archive.py", line 28, in <module>
ModuleNotFoundError: No module named 'struct'
[11718] mod is NULL - pyimod03_importersTraceback (most recent call last):
  File "/home/adam/anaconda2/envs/cgqui4/lib/python3.6/site-packages/PyInstaller/loader/pyimod03_importers.py", line 24, in <module>
ModuleNotFoundError: No module named 'pyimod02_archive'
Traceback (most recent call last):
  File "site-packages/PyInstaller/loader/pyiboot01_bootstrap.py", line 15, in <module>
ModuleNotFoundError: No module named 'pyimod03_importers'
[11718] Failed to execute script pyiboot01_bootstrap
```

adam-urbanczyk, 2019-08-07:
Could you download the archive and try again? The package was updated and should work now..

roipoussiere, 2019-08-07:
There is [no new release](https://github.com/CadQuery/CQ-editor/releases). Do you mean that I should clone the repo? In this case, there is not `./CQ-editor` to execute...

adam-urbanczyk, 2019-08-07:
Indeed I updated the package for 0.1RC1 Just try it again...

roipoussiere, 2019-08-07:
> Indeed I updated the package for 0.1RC1 Just try it again...

Oh I didn't know that it was possible without creating another release.

This fixed my problem! Thank you!!

closed 2019-08-07 by roipoussiere