CadQuery/CQ-editor #75: Problem when installing from sources

roipoussiere, 2019-08-05:
I try to install CQ-editor from sources, because Conda doesn't work on my environment. I also hope that this installation procedure will help other people.

I successfully installed CadQuery, with the following commands:

```bash
INSTALLATION_DIR=$(pwd)

# Installing oce from sources

cd $INSTALLATION_DIR
git clone git://github.com/tpaviot/oce.git
mkdir oce/build
cd oce/build
cmake ..
make -j9
sudo make install/strip
make test

# Installing pythonocc-core from sources

cd $INSTALLATION_DIR
git clone https://github.com/tpaviot/pythonocc-core.git
mkdir pythonocc-core/cmake-build
cd pythonocc-core/cmake-build
cmake -DOCE_INCLUDE_PATH=/usr/local/include/oce -DOCE_LIB_PATH=/usr/local/lib ..
make -j9
sudo make install
echo 'export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib/' > ~/.zshrc
source ~/.zshrc
cd ../test
python run_tests.py

# Installing CadQuery from sources

cd $INSTALLATION_DIR
git clone https://github.com/CadQuery/cadquery.git
sudo python setup.py install --single-version-externally-managed --record=record.txt
python runtests.py
# python
# > import cadquery
# > cadquery.Workplane('XY').box(1,2,3).toSvg() # --> success!! \o/
```

Now for CQ-editor, based on the [Conda file](https://github.com/CadQuery/CQ-editor/blob/master/conda/meta.yaml) and some commands outputs, I ran:

```
cd $INSTALLATION_DIR
# getting sources:
git clone https://github.com/CadQuery/CQ-editor.git
# installing python dependencies:
sudo pip install Logbook PyQt5 pyparsing pyqtgraph spyder path.py qtconsole requests2
# installing CQ-editor:
sudo python setup.py install --single-version-externally-managed --record=record.txt
# installing python dependencies required for tests:
sudo pip install pytest pytest-qt pytest-mock
# executing tests:
python -m pytest -s
```

But I get [this error](https://gist.github.com/roipoussiere/4d921e6aebc4b95990acd3682e55cef8) which suggests that a variable `self.canvas._display` is null.

Do you have any advice to give? Did I missed to install some dependency?

adam-urbanczyk, 2019-08-05:
I think you are using the stock PythonOCC - CQ-editor requires a custom fork https://github.com/CadQuery/pythonocc-core

If I were you I'd really try using conda - you can run it fully locally without any admin privileges.

roipoussiere, 2019-08-06:
> I think you are using the stock PythonOCC - CQ-editor requires a custom fork

Indeed I was using the stock PythonOCC, with the custom fork I don't have this issue, thank you!

> If I were you I'd really try using conda - you can run it fully locally without any admin privileges.

I trust you, but Conda doesn't work on my system, I spent a couple of hours to debug it but it seems hopeless.

But now I have [an other issue](https://github.com/CadQuery/CQ-editor/issues/9#issuecomment-518742607) to fix (maybe not related to the fact that I compiled the program from sources).

adam-urbanczyk, 2019-08-10:
@roipoussiere  did you follow my comment in #9? If so did it help with the issue?

roipoussiere, 2019-08-10:
Not yet, but in any case, this is not really related to this issue so we can close this one I think (or rename it to *can not compile from sources* maybe).

adam-urbanczyk, 2019-08-10:
Alright, closing.

closed 2019-08-10 by adam-urbanczyk