CadQuery/CQ-editor #77: Use a monospace font for the editor

roipoussiere, 2019-08-12:
Usually a monospace font is use on IDEs, it allows doing pretty indentations.

It's notably useful in Python, in order to respect the [pep8 standard](https://www.python.org/dev/peps/pep-0008/#indentation):

```python
# Aligned with opening delimiter.
foo = long_function_name(var_one, var_two,
                         var_three, var_four)
```

We can't do this in QC-editor, because spaces are rendered smaller than other letters.

rdgraham, 2019-11-01:
I also noticed this issue immediately when trying out cq-editor. It is very unnatural to not use a monospace font when writing python.

I found it was quite easy to change this. In `cqeditor/widgets/editor.py` add the import

`from PyQt5.QtGui import QFont`

and in the `self.setup_editor` add `font=QFont('Courier New', 10)`.

I think monospace should be the default and the option added to the preferences. I can make a pull request to add this if that is useful.

adam-urbanczyk, 2019-11-02:
Solved in #95. I will open a new issue for making the font user selectable.

closed 2019-11-02 by adam-urbanczyk