CadQuery/CQ-editor #80: CQGI debug() colors

roipoussiere, 2019-08-13:
As a beginner, I would like to understand what is in the object stack and what I select. As far as I understand, the `debug()` is here for that purpose.

When executing the CQGI script example, [provided in the documentation](https://cadquery.readthedocs.io/en/latest/cqgi.html?highlight=color#the-script-side):

```python
base_cube = cq.Workplane('XY').rect(1.0,1.0).extrude(1.0)
top_of_cube_plane = base_cube.faces(">Z").workplane()
debug(top_of_cube_plane, { 'color': 'yellow', } )
debug(top_of_cube_plane.center, { 'color' : 'blue' } )

circle=top_of_cube_plane.circle(0.5)
debug(circle, { 'color': 'red' } )

show_object( circle.extrude(1.0) )
```

I get this error:

```
TypeError: () takes one positional argument but 2 were given
```

Is CQGI not yet fully implemented in CadQuery Editor, or I missed something?

jmwright, 2019-08-13:
@adam-urbanczyk can say for sure, but it may be related to [this issue](https://github.com/CadQuery/CQ-editor/issues/53).

Essentially the options for `show_object`, and I assume `debug`, haven't been implemented yet. Just drop the options from the debug call and the error should go away (it did for me).

roipoussiere, 2019-08-13:
I though that `debug` was used to color some parts of an object in order to understand what we are selecting. What the point of `debug()` without any argument, then?

jmwright, 2019-08-13:
> What the point of debug() without any argument, then?

I think the CQGI implementation is just lagging a little after the transition to CQ 2.0. I haven't seen many people use the `debug` functionality, even though it can be pretty useful. That may be why this hasn't been brought up yet.

In CadQuery's FreeCAD workbench (CQ 1.0) I rendered debug objects in red with some transparency by default, so the option wasn't used much. I'm not sure if @adam-urbanczyk has something like that planned for CQ-editor or not.

adam-urbanczyk, 2019-08-18:
@roipoussiere CQ-editor does not implement the CQGI interface. Currently `debug` just displays the argument in the application log.

BTW: there is a dedicated graphical stack inspector in CQ-editor if you are interested in seeing what is on your stack.

adam-urbanczyk, 2020-03-26:
After merging #127 it is possible to specify color like so
```python
show_object(obj,name='a',options=dict(color='red'))
```

roipoussiere, 2020-05-01:
@adam-urbanczyk really nice! Documentation should be updated as well but I don't know why I'm not currently able to access to the wiki...

roipoussiere, 2021-03-02:
Closing since #127 solves this issue and enough documentation has been added in the readme. :)

closed 2021-03-02 by roipoussiere