CadQuery/CQ-editor #85: Help command can not open web browser

roipoussiere, 2019-08-13:
The help commands (*Help Menu/Documentation* and *Help Menu/CQ Documentation*) doesn't open the web browser.

Here is the stack trace:

```
/usr/bin/xdg-mime: ligne 323 : [: /home/nath/.local/share/applications/userapp/Pale : opérateur binaire attendu
/usr/bin/xdg-mime: ligne 325 : [: /home/nath/.local/share/applnk/userapp/Pale : opérateur binaire attendu
XPCOMGlueLoad error for file /usr/lib/palemoon/libxul.so:
/usr/lib/libgio-2.0.so.0: undefined symbol: g_ref_count_inc
Couldn't load XPCOM.
XPCOMGlueLoad error for file /usr/lib/palemoon/libxul.so:
/usr/lib/libgio-2.0.so.0: undefined symbol: g_ref_count_inc
Couldn't load XPCOM.
XPCOMGlueLoad error for file /usr/lib/palemoon/libxul.so:
/usr/lib/libgio-2.0.so.0: undefined symbol: g_ref_count_inc
Couldn't load XPCOM.
xdg-open: no method available for opening 'https://dcowden.github.io/cadquery'
```

"opérateur binaire attendu" means "expecting binary operator".

OS: Linux Manjaro 18.04

---

Side note:

The url for *Help Menu/CQ Documentation*, which is currently https://dcowden.github.io/cadquery should be updated to https://cadquery.readthedocs.io or even https://cadquery.readthedocs.io/en/latest/apireference.html

adam-urbanczyk, 2019-08-18:
Could it be something with your distribution? I don't know any better way of opening web-pages than using `xdg-open`.

roipoussiere, 2021-03-02:
> Could it be something with your distribution?

Probably, I cannot reproduce this now. I close.

closed 2021-03-02 by roipoussiere