CadQuery/CQ-editor #86: AttributeError: 'NoneType' object has no attribute 'View'

l29ah, 2019-08-21:
Just installed CQ-editor, tried to run it:
```
‰ cq-editor
INFO:OCC.Display.backend:backend loaded: qt-pyqt5
Traceback (most recent call last):
  File "/usr/lib/python-exec/python3.6/cq-editor", line 11, in <module>
    load_entry_point('CQ-editor==0.1.0.dev0', 'gui_scripts', 'cq-editor')()
  File "/usr/lib64/python3.6/site-packages/cq_editor/__main__.py", line 16, in main
    win = MainWindow()
  File "/usr/lib64/python3.6/site-packages/cq_editor/main_window.py", line 39, in __init__
    self.viewer = OCCViewer(self)
  File "/usr/lib64/python3.6/site-packages/cq_editor/widgets/viewer.py", line 62, in __init__
    self.updatePreferences()
  File "/usr/lib64/python3.6/site-packages/cq_editor/widgets/viewer.py", line 71, in updatePreferences
    self.canvas._display.View.SetBgGradientColors(color1,color2,True)
AttributeError: 'NoneType' object has no attribute 'View'
```

adam-urbanczyk, 2019-08-21:
Did you follow the installation instructions from the `README`? CQ-editor requires a custom fork of pythonocc (https://github.com/CadQuery/pythonocc-core). It is available via the cadquery conda channel.

fake-name, 2019-09-26:
The need for a custom fork of pythonocc isn't mentioned anywhere in https://github.com/CadQuery/cadquery/blob/master/README.md

If you're trying to install *without* using anaconda (which I am), there isn't any guidance whatsoever.

adam-urbanczyk, 2019-09-28:
Indeed, you are right. The easiest way to try out CQ-editor us to use the self-contained zip (https://github.com/CadQuery/CQ-editor/releases/tag/0.1RC1). Let me know if that works for you.

fake-name, 2019-09-28:
In this case, I was trying to run it with the version of PythonOCC (and it's associated dependencies) I had just built, so I was explicitly trying to build *everything* from source, so another different way to get a prebuilt package isn't exactly useful.

I wound up giving up and using anaconda, which had a bunch of issue with dependency resolution but did eventually work.

adam-urbanczyk, 2019-10-01:
I'm going to close that issue then. Feel free to open a new one regarding building from sources docs (or even better share your experience in a PR).

closed 2019-10-01 by adam-urbanczyk

greyltc, 2019-11-27:
I have the same problem. It's very unfortunate that this project relies on a fork of PythonOCC. I've been unpleasantly surprised to learn that this doesn't play nicely with the upstream project I thought it stemmed from.
Reliance on anaconda for this forked dep is a non-starter for me personally.