CadQuery/CQ-editor #90: Intel MKL FATAL ERROR: Cannot load mkl_intel_thread.dll. (FIX)

slackmage, 2019-09-13:
I was stumped for a couple days with this error when trying to run cq-editor. As it turns out the issue was that I had a music production product called Amplitube installed. It places some incompatible files in windows/system32 that causes the load of load mkl_intel_thread.dll to fail . I uninstalled it and everything worked fine. I have also read that ASIO4ALL causes this error as well, though I have not tested that myself.

adam-urbanczyk, 2019-09-13:
Thanks for trying CQ-editor! I am not sure how to reproduce this error. Could you e.g. share some details of your environment by running the command below? Or are you maybe using the executable version form here https://github.com/CadQuery/CQ-editor/releases/tag/0.1RC1 ?
```
conda info -a
```
It would be also helpful to know which files are causing the issue.

slackmage, 2019-09-13:
active environment : None
       user config file : C:\Users\User\.condarc
 populated config files :
          conda version : 4.7.11
    conda-build version : 3.18.8
         python version : 3.7.3.final.0
       virtual packages : __cuda=10.1
       base environment : C:\Users\User\Anaconda3  (writable)
           channel URLs : https://repo.anaconda.com/pkgs/main/win-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/win-64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/msys2/win-64
                          https://repo.anaconda.com/pkgs/msys2/noarch
          package cache : C:\Users\User\Anaconda3\pkgs
                          C:\Users\User\.conda\pkgs
                          C:\Users\User\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\User\Anaconda3\envs
                          C:\Users\User\.conda\envs
                          C:\Users\User\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/4.7.11 requests/2.22.0 CPython/3.7.3 Windows/10 Windows/10.0.18362
          administrator : False
             netrc file : None
           offline mode : False

# conda environments:
#
base                  *  C:\Users\User\Anaconda3
cqgui                    C:\Users\User\Anaconda3\envs\cqgui
                         E:\Anaconda3
                         E:\Anaconda3\envs\cqgui

sys.version: 3.7.3 (default, Apr 24 2019, 15:29:51) [...
sys.prefix: C:\Users\User\Anaconda3
sys.executable: C:\Users\User\Anaconda3\python.exe
conda location: C:\Users\User\Anaconda3\lib\site-packages\conda
conda-build: C:\Users\User\Anaconda3\Scripts\conda-build.exe
conda-convert: C:\Users\User\Anaconda3\Scripts\conda-convert.exe
conda-debug: C:\Users\User\Anaconda3\Scripts\conda-debug.exe
conda-develop: C:\Users\User\Anaconda3\Scripts\conda-develop.exe
conda-env: C:\Users\User\Anaconda3\Scripts\conda-env.exe
conda-index: C:\Users\User\Anaconda3\Scripts\conda-index.exe
conda-inspect: C:\Users\User\Anaconda3\Scripts\conda-inspect.exe
conda-metapackage: C:\Users\User\Anaconda3\Scripts\conda-metapackage.exe
conda-render: C:\Users\User\Anaconda3\Scripts\conda-render.exe
conda-server: C:\Users\User\Anaconda3\Scripts\conda-server.exe
conda-skeleton: C:\Users\User\Anaconda3\Scripts\conda-skeleton.exe
conda-verify: C:\Users\User\Anaconda3\Scripts\conda-verify.exe
user site dirs: C:\Users\User\AppData\Roaming\Python\Python37

CIO_TEST: <not set>
CONDA_EXE: C:\Users\User\Anaconda3\condabin\..\Scripts\conda.exe
CONDA_EXES: "C:\Users\User\Anaconda3\condabin\..\Scripts\conda.exe"
CONDA_ROOT: C:\Users\User\Anaconda3
HOMEPATH: \Users\User
MIC_LD_LIBRARY_PATH: C:\Program Files (x86)\Common Files\Intel\Shared Libraries\compiler\lib\intel64_win_mic
PATH: C:\Users\User\Anaconda3;C:\Users\User\Anaconda3\Library\mingw-w64\bin;C:\Users\User\Anaconda3\Library\usr\bin;C:\Users\User\Anaconda3\Library\bin;C:\Users\User\Anaconda3\Scripts;C:\Users\User\Anaconda3\bin;C:\Program Files (x86)\Common Files\Intel\Shared Libraries\redist\intel64_win\compiler;C:\Program Files\Oculus\Support\oculus-runtime;C:\Program Files (x86)\Common Files\Intel\Shared Files\cpp\bin\Intel64;C:\Program Files (x86)\Common Files\Intel\Shared Files\cpp\bin\ia32;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\PuTTY\;C:\Program Files\Git\cmd;C:\Program Files\Common Files\Autodesk Shared\;C:\Program Files (x86)\Autodesk\Backburner\;C:\Program Files\dotnet\;C:\Program Files\nodejs\;C:\ProgramData\chocolatey\bin;C:\Users\User\Anaconda3;C:\Users\User\Anaconda3\Library\mingw-w64\bin;C:\Users\User\Anaconda3\Library\usr\bin;C:\Users\User\Anaconda3\Library\bin;C:\Users\User\Anaconda3\Scripts;C:\Users\User\AppData\Local\Microsoft\WindowsApps;C:\Users\User\AppData\Local\GitHubDesktop\bin;C:\Users\User\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\User\AppData\Roaming\npm;E:\Nmap
PSMODULEPATH: C:\Program Files\WindowsPowerShell\Modules;C:\WINDOWS\system32\WindowsPowerShell\v1.0\Modules
REQUESTS_CA_BUNDLE: <not set>
SSL_CERT_FILE: <not set>

slackmage, 2019-09-13:
I ran into the issue when using the binary and also building from the source

adam-urbanczyk, 2019-09-22:
@slackmage are you able to identify the file that causes the issue? At the moment I'm not sure how to reproduce it locally.

slackmage, 2019-09-23:
The way to reproduce it would be to install Amplitube. When uninstalling it I wasn't able to determine which file Amplitube had put in system32 that was removed.

adam-urbanczyk, 2019-09-23:
Is the free version what you used https://www.ikmultimedia.com/products/index.php?R=INIT&FV=all-amplitube-for-macpc-related-products&CV=Other%20Filter&PSEL=all-amplitube ?

slackmage, 2019-09-23:
No but it's likely that the demo version would cause the same issue.

adam-urbanczyk, 2020-03-26:
I don't have the bandwith to reproduce this issue.

closed 2020-03-26 by adam-urbanczyk