CadQuery/CQ-editor #94: Crash when compiling a specific model

jduquennoy, 2019-10-05:
Hi guys,

I have been using CQ-editor on mac to build a 3d model (my most complex model to date :-) ).
It is really a pleasure to be able to do that on my machine without using a virtualized linux ! Thanks a lot to you, who made that possible.

i recently updated to the latest HEAD (2ddbc1c), and now, CA-Editor crashes when compiling the project, due to an access to a released of a non allocated object.
Does anyone experienced such a problem ?

Please find attached the model that causes the crash.

[flashBrackets.py.zip](https://github.com/CadQuery/CQ-editor/files/3693061/flashBrackets.py.zip)

adam-urbanczyk, 2019-10-06:
I tired your code on linux and I get a `Standard_ConstructionError` on this line:
```python
 armSB200 = makeArm2SB200(armFactory).translate((80, 0))
```
If I change it to the following
```python
armSB200 = makeArm2SB200(Arm()).translate((80, 0))
```
I get:
[image]

I have two questions:
1. Is this the intended result?
2. Could you share more details about the crash (BTW: I assume we are talking about a segfault)? It would be good to have a Python stacktrace to confirm that it is triggered by the same line and possibly a backtrace from a debugger to check where it occurs in PythonOCC/OCE.

jduquennoy, 2019-10-06:
Yep, the result seems to be perfect :-)

It does compile with the corrected version on my machine, but it crashes with the construction error. Here is the stack trace (I should have added that in first place, sorry for that):

```
python(46607,0x1073375c0) malloc: *** error for object 0x7fdcf0165698: pointer being freed was not allocated
python(46607,0x1073375c0) malloc: *** set a breakpoint in malloc_error_break to debug
Fatal Python error: Aborted

Thread 0x000070001023a000 (most recent call first):
  File "/Users/jerome/miniconda3/envs/cqgui/lib/python3.6/selectors.py", line 577 in select
  File "/Users/jerome/miniconda3/envs/cqgui/lib/python3.6/asyncio/base_events.py", line 1415 in _run_once
  File "/Users/jerome/miniconda3/envs/cqgui/lib/python3.6/asyncio/base_events.py", line 438 in run_forever
  File "/Users/jerome/miniconda3/envs/cqgui/lib/python3.6/site-packages/tornado/platform/asyncio.py", line 148 in start
  File "/Users/jerome/miniconda3/envs/cqgui/lib/python3.6/site-packages/ipykernel/iostream.py", line 78 in _thread_main
  File "/Users/jerome/miniconda3/envs/cqgui/lib/python3.6/threading.py", line 864 in run
  File "/Users/jerome/miniconda3/envs/cqgui/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/Users/jerome/miniconda3/envs/cqgui/lib/python3.6/threading.py", line 884 in _bootstrap

Thread 0x000070000fd37000 (most recent call first):
  File "/Users/jerome/miniconda3/envs/cqgui/lib/python3.6/threading.py", line 295 in wait
  File "/Users/jerome/miniconda3/envs/cqgui/lib/python3.6/threading.py", line 551 in wait
  File "/Users/jerome/miniconda3/envs/cqgui/lib/python3.6/site-packages/IPython/core/history.py", line 829 in run
  File "/Users/jerome/miniconda3/envs/cqgui/lib/python3.6/site-packages/IPython/core/history.py", line 58 in needs_sqlite
  File "</Users/jerome/miniconda3/envs/cqgui/lib/python3.6/site-packages/decorator.py:decorator-gen-24>", line 2 in run
  File "/Users/jerome/miniconda3/envs/cqgui/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/Users/jerome/miniconda3/envs/cqgui/lib/python3.6/threading.py", line 884 in _bootstrap

Current thread 0x00000001073375c0 (most recent call first):
  File "/Users/jerome/miniconda3/envs/cqgui/lib/python3.6/site-packages/cadquery/occ_impl/shapes.py", line 208 in clean
  File "/Users/jerome/miniconda3/envs/cqgui/lib/python3.6/site-packages/cadquery/cq.py", line 2887 in <listcomp>
  File "/Users/jerome/miniconda3/envs/cqgui/lib/python3.6/site-packages/cadquery/cq.py", line 2887 in clean
  File "/Users/jerome/miniconda3/envs/cqgui/lib/python3.6/site-packages/cadquery/cq.py", line 2287 in extrude
  File "<string>", line 38 in make
  File "<string>", line 176 in makeArm2SB200
  File "<string>", line 258 in <module>
  File "/Users/jerome/CQ-editor/cq_editor/widgets/debugger.py", line 183 in _exec
  File "/Users/jerome/CQ-editor/cq_editor/widgets/debugger.py", line 214 in render
  File "/Users/jerome/CQ-editor/cq_editor/__main__.py", line 27 in main
  File "run.py", line 11 in <module>
Abort trap: 6
```

adam-urbanczyk, 2019-10-08:
I am confused @jduquennoy . Does the code I pasted run or does it still segfault or does it result in an exception without a crash?

jduquennoy, 2019-10-08:
Sorry, i wasn’t clear.
The code you pasted runs perfectly.
The stack trace is produced by my old code, that generates a construction error.

adam-urbanczyk, 2019-11-10:
I will close this, will handle the crash on exception issue here: https://github.com/CadQuery/cadquery/issues/207

closed 2019-11-10 by adam-urbanczyk