CadQuery/CQ-editor #99: Reloading imported modules

marcus7070, 2019-11-20:
I have a feature request, or at least a discussion on one.

#### Use case

*NB. This is very simplified, I have one project with 8 different local module imports, and I intend to use more in the next.*

driving_dims.py:
```python
box = (100, 20, 30)
```

main.py:
```python
import cadquery as cq
import driving_dims

a = cq.Workplane("front").box(*driving_dims.box)

show_object(a)
```

1. start cq-editor, load `main.py`, render
2. change `box` in `driving_dims.py`
3. render again, new dimensions are not loaded

#### User side solution

I naively restarted cq-editor a lot until I discovered I could change my import from
```python
import driving_dims
```
to
```python
import importlib
import driving_dims
importlib.reload(driving_dims)
```
to have the module reloaded every time the script is executed. 

I kind of like this solution because you can be explicit about what needs to be reloaded every time. 

#### Feature suggestion

I would be happy with just documenting the `importlib.reload` method, but the devs seem to be going for a quite user-friendly feel, so perhaps they want to implement automatic reloading of any module in the working script directory?

adam-urbanczyk, 2020-02-05:
It should be possible, e.g. like so:
 *   parse the code with AST and find import statements
 *   reload the modules is they are already loaded
This will have low prio for me at the moment though.

adam-urbanczyk, 2020-09-13:
Related issue: #185

adam-urbanczyk, 2020-09-13:
External modules that might be helpful:

* https://github.com/ronaldoussoren/modulegraph
* https://github.com/thebjorn/pydeps
* https://github.com/pawamoy/dependenpy

* https://github.com/jparise/python-reloader
*

fedorkotov, 2021-04-24:
I encountered this problem when I was developing my plugin for cadquery-plugins https://github.com/CadQuery/cadquery-plugins/pull/20 (see examples in `README.md`). My plugin adds a method to `cq.Workplane`. This works for the first run after starting CQ-editor but second and subsequent runs fail. I had to manually force module reloading on each run

```python
import sys

# ...

if "apply_to_each_face" in sys.modules:
    del sys.modules["apply_to_each_face"]
from apply_to_each_face import # ...
```

lorenzncode, 2021-08-21:
This was implemented in #272.

1. start cq-editor, load `main.py`, render
2. change `box` in `driving_dims.py`
3. render again, new dimensions **are** loaded

With Preferences "Autoreload: watch imported modules" selected:

1. start cq-editor, load `main.py`, render
2. change box in `driving_dims.py`, render updates automatically

closed 2021-08-25 by marcus7070