CadQuery/CQ-editor #102: 2D sketcher

dumblob, 2019-11-27:
It seems the major downside (and the reason why CadQuery is difficult to use in practice) is the missing support for sketching.

Sketching can't be done programatically at all in practice or very inefficiently (and creating paths with external tools is very frustrating), so sketcher with the same functionality as the FreeCAD sketcher (including the constraint solver) could be an option. But way better integrated rather than switching back and forth without context sharing like in FreeCAD.

To be clear, from my experience gathered from several long-time engineers, sketching accounts for the vast majority of time spent with 3D CAD modeling. Optimizing sketching possibilities, interactivity, and integration seems like the key to success.

Thoughts?

jmwright, 2019-11-27:
@dumblob This doesn't address your core concern, but here are a few issues relevant to 2D sketching in CadQuery.

[Sketchlets](https://github.com/CadQuery/cadquery/issues/52)

[Parametric Sketching](https://github.com/CadQuery/cadquery/issues/42)

[DXF Import](https://github.com/CadQuery/cadquery/issues/17)

dumblob, 2019-11-28:
Thanks @jmwright, the links look like a solid foundation for what I envision in this issue.