CadQuery/CQ-editor #103: Feature request: Import a STEP file from a relative path

marcus7070, 2019-11-30:
Download https://www.molex.com/pdm_docs/stp/48037-1000_stp.zip to get the STEP file. Stick it in a folder with the following script:
```python
import cadquery as cq

part = (cq
    .importers.importStep('480371000.stp')
)
print(part.toSvg())
```
Running `python3 ./script.py` produces the expected SVG output, but loading the same file through CQ-editor (started through a launcher, like you would usually do with a graphical program) fails because it can't find the STEP file. 

Changing the file to an absolute path does solve this, but I'd like to distribute a CadQuery example and have users just open the script.py and it works. Distributing an example with `/home/marcus/tempbuild/attemptNumber89/480371000.stp` in the code is always a bit embarrassing and unprofessional. 

It's a pretty unimportant feature to bother you about, but I'm hoping there is a trivial solution. If I launch CQ-editor from a command line with the current working directory containing the script, relative paths work. So perhaps just setting the working directory before CQ-editor runs the script would work?

marcus7070, 2019-11-30:
With regard to #14, I see `subprocess.run` takes an argument for current working directory.

adam-urbanczyk, 2019-12-01:
I think this should be relatively easy - just wrap the execution of the cq script into a `with Path(file_path):` block.

adam-urbanczyk, 2020-02-05:
Solved by #104

closed 2020-02-05 by adam-urbanczyk