CadQuery/CQ-editor #106: MacOS : Recursive repaint detected

osterwood, 2019-12-16:
I created a new Conda env with a head checkout of CQ-editor and get the following error when trying to launch it:

```
› python run.py        
Warning: QApplication was created before pyqtgraph was imported; there may be problems (to avoid bugs, call QApplication.setGraphicsSystem("raster") before the QApplication is created).
Namespace(filename=None)
QWidget::repaint: Recursive repaint detected
Fatal Python error: Segmentation fault

Thread 0x000070000e39b000 (most recent call first):
  File "/Users/chris/anaconda3/envs/cqgui/lib/python3.7/site-packages/zmq/utils/garbage.py", line 47 in run
  File "/Users/chris/anaconda3/envs/cqgui/lib/python3.7/threading.py", line 926 in _bootstrap_inner
  File "/Users/chris/anaconda3/envs/cqgui/lib/python3.7/threading.py", line 890 in _bootstrap

Thread 0x000070000de98000 (most recent call first):
  File "/Users/chris/anaconda3/envs/cqgui/lib/python3.7/selectors.py", line 558 in select
  File "/Users/chris/anaconda3/envs/cqgui/lib/python3.7/asyncio/base_events.py", line 1735 in _run_once
  File "/Users/chris/anaconda3/envs/cqgui/lib/python3.7/asyncio/base_events.py", line 534 in run_forever
  File "/Users/chris/anaconda3/envs/cqgui/lib/python3.7/site-packages/tornado/platform/asyncio.py", line 148 in start
  File "/Users/chris/anaconda3/envs/cqgui/lib/python3.7/site-packages/ipykernel/iostream.py", line 78 in _thread_main
  File "/Users/chris/anaconda3/envs/cqgui/lib/python3.7/threading.py", line 870 in run
  File "/Users/chris/anaconda3/envs/cqgui/lib/python3.7/threading.py", line 926 in _bootstrap_inner
  File "/Users/chris/anaconda3/envs/cqgui/lib/python3.7/threading.py", line 890 in _bootstrap

Thread 0x000070000d88f000 (most recent call first):
  File "/Users/chris/anaconda3/envs/cqgui/lib/python3.7/threading.py", line 296 in wait
  File "/Users/chris/anaconda3/envs/cqgui/lib/python3.7/threading.py", line 552 in wait
  File "/Users/chris/anaconda3/envs/cqgui/lib/python3.7/site-packages/IPython/core/history.py", line 829 in run
  File "/Users/chris/anaconda3/envs/cqgui/lib/python3.7/site-packages/IPython/core/history.py", line 58 in needs_sqlite
  File "</Users/chris/anaconda3/envs/cqgui/lib/python3.7/site-packages/decorator.py:decorator-gen-25>", line 2 in run
  File "/Users/chris/anaconda3/envs/cqgui/lib/python3.7/threading.py", line 926 in _bootstrap_inner
  File "/Users/chris/anaconda3/envs/cqgui/lib/python3.7/threading.py", line 890 in _bootstrap

Current thread 0x00000001107f65c0 (most recent call first):
  File "/Users/chris/anaconda3/envs/cqgui/lib/python3.7/site-packages/OCC/Display/OCCViewer.py", line 174 in Create
  File "/Users/chris/anaconda3/envs/cqgui/lib/python3.7/site-packages/OCC/Display/qtDisplay.py", line 120 in InitDriver
  File "/Users/chris/anaconda3/envs/cqgui/lib/python3.7/site-packages/OCC/Display/qtDisplay.py", line 149 in paintEvent
  File "/Users/chris/anaconda3/envs/cqgui/lib/python3.7/site-packages/OCC/Display/OCCViewer.py", line 174 in Create
  File "/Users/chris/anaconda3/envs/cqgui/lib/python3.7/site-packages/OCC/Display/qtDisplay.py", line 120 in InitDriver
  File "/Users/chris/anaconda3/envs/cqgui/lib/python3.7/site-packages/OCC/Display/qtDisplay.py", line 149 in paintEvent
  File "/Users/chris/Documents/github/cadquery/CQ-editor/cq_editor/__main__.py", line 26 in main
  File "run.py", line 11 in <module>
zsh: segmentation fault  python run.py
```

Any ideas about what might be causing this?  The repo runs without issue on a Ubuntu machine I have, with a Conda env created the exact same way.

jmwright, 2019-12-16:
@osterwood I can't replicate this. I get some weird visual artifacts in the toolbar, but the app runs. Below is the system info.

[image]

osterwood, 2019-12-16:
I'm running 10.14.4.  I'll try updating the 10.14.6 and see if that changes anything.

adam-urbanczyk, 2019-12-16:
I don't have access to mac but I'll try to finalize #101 to check if this can be reproduced. In the meantime could you try to paste a backtrace from lldb ?

jmwright, 2019-12-27:
@osterwood Any progress on this?

osterwood, 2019-12-27:
Haven't had a chance to upgrade yet, but I was able to get it to launch (once) and run code successfully.  

Since then, I am getting the same "Recursive repaint" error.  Exiting the conda ENV and re-entering it does not help, no apparent python process lingering in the background -- not sure why it worked once and not again.

Here is what happened when running via lldb (not sure how useful this is).

```
› lldb python       
(lldb) target create "python"
Current executable set to 'python' (x86_64).
(lldb) settings set -- target.run-args "run.py"
(lldb) r
Process 22568 launched: '/Users/chris/anaconda3/envs/cqgui/bin/python' (x86_64)
Warning: QApplication was created before pyqtgraph was imported; there may be problems (to avoid bugs, call QApplication.setGraphicsSystem("raster") before the QApplication is created).
2019-12-26 22:15:20.526739-0500 python[22568:32536228] SecTaskLoadEntitlements failed error=22 cs_flags=20, pid=22568
2019-12-26 22:15:20.526828-0500 python[22568:32536228] SecTaskCopyDebugDescription: python3.7[22568]/0#-1 LF=0
2019-12-26 22:15:20.851401-0500 python[22568:32536228] SecTaskLoadEntitlements failed error=22 cs_flags=20, pid=22568
2019-12-26 22:15:20.851482-0500 python[22568:32536228] SecTaskCopyDebugDescription: python3.7[22568]/0#-1 LF=0
Namespace(filename=None)
QWidget::repaint: Recursive repaint detected
Process 22568 stopped
* thread #1, name = 'ZMQbg/1', queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0xa0)
    frame #0: 0x000000012022136d libTKOpenGl.11.dylib`OpenGl_ShaderManager::BindProgram(OpenGl_AspectLine const*, Handle_OpenGl_Texture const&, bool, bool, Handle_OpenGl_ShaderProgram const&) + 29
libTKOpenGl.11.dylib`OpenGl_ShaderManager::BindProgram:
->  0x12022136d <+29>: movq   0xa0(%rsi), %rax
    0x120221374 <+36>: cmpb   $0x0, 0xf(%rax)
    0x120221378 <+40>: je     0x120221392               ; <+66>
    0x12022137a <+42>: movq   %rbx, %rdi
```

jmwright, 2019-12-27:
Sounds like maybe a race condition in the app's startup.

adam-urbanczyk, 2019-12-29:
Looking at the python stacktrace, it looks like `InitDriver` is called from within `InitDriver`. This should not happen, but I have no clue at this point why it does.
```
Current thread 0x00000001107f65c0 (most recent call first):
  File "/Users/chris/anaconda3/envs/cqgui/lib/python3.7/site-packages/OCC/Display/OCCViewer.py", line 174 in Create
  File "/Users/chris/anaconda3/envs/cqgui/lib/python3.7/site-packages/OCC/Display/qtDisplay.py", line 120 in InitDriver
  File "/Users/chris/anaconda3/envs/cqgui/lib/python3.7/site-packages/OCC/Display/qtDisplay.py", line 149 in paintEvent
  File "/Users/chris/anaconda3/envs/cqgui/lib/python3.7/site-packages/OCC/Display/OCCViewer.py", line 174 in Create
  File "/Users/chris/anaconda3/envs/cqgui/lib/python3.7/site-packages/OCC/Display/qtDisplay.py", line 120 in InitDriver
  File "/Users/chris/anaconda3/envs/cqgui/lib/python3.7/site-packages/OCC/Display/qtDisplay.py", line 149 in paintEvent
```

osterwood, 2019-12-31:
I was just able to launch the GUI 6 times in a row.  No changes to the conda ENV or the OS.  No idea what's going on here, but closing the issue until I can reproduce the issue reliably.  

Adam, thanks for noticing the recursive call there.  If this reproduces, that gives me a good starting point to determine the root cause.

closed 2019-12-31 by osterwood

adam-urbanczyk, 2020-01-01:
@osterwood  did you maybe update graphics card drivers recently?

osterwood, 2020-01-01:
Nope - this machine (which originally had the issue and now does not) has not been rebooted in 50 days.