CadQuery/CQ-editor #107: Package Build Instructions

jmwright, 2019-12-31:
It would probably be a good idea to have documentation on building the stand-alone packages of CQ-editor. This would allow someone with access to a Mac to work on building a package so that we could have all three major OSes covered by the releases.

adam-urbanczyk, 2020-01-02:
Good idea! The general approach is to use pyinstaller, I have no experience using it on Macs.

jmwright, 2020-01-02:
My hope is that @michaelgale could help us with this if we provide good info on how to do it.

https://github.com/CadQuery/cadquery/issues/262#issuecomment-569942144

jmwright, 2020-01-02:
@adam-urbanczyk Could you maybe document that you did for Linux and we can use that as a base for what to do on MaxOS?

michaelgale, 2020-01-03:
@jmwright @adam-urbanczyk I don't have direct experience with the CQ-Editor; however, I would be happy to try to build the macOS version.  I would likely go through the process on a linux VM first to get the hang of things.  In any case, I have macOS Mojave and Catalina build platforms to verify stability across OS the latest releases. 
Forgive my ignorance, but is CQ-Editor distributed as a conda build recipe?

jmwright, 2020-01-03:
@michaelgale There is a [conda yaml file](https://github.com/CadQuery/CQ-editor/blob/master/cqgui_env.yml), but there are also [pre-built packages](https://github.com/CadQuery/CQ-editor/releases/tag/0.1RC0) that are designed to be self-contained, meaning the user shouldn't need to install dependencies or set up an Anaconda environment. CadQuery 1.0 always supported all three major operating systems, and I think that having a pre-built CQ-editor package for MacOS would help us maintain that compatibility for CadQuery 2.0.

With that said, if you'll never use CQ-editor, I don't want to force you into contributing to a project that doesn't scratch an itch for you.

@adam-urbanczyk I might look into this for consistent access to a Mac that I don't have to worry about breaking. Looks like the first 30 hours are free. https://checkout.macincloud.com/

michaelgale, 2020-01-03:
@jmwright Understood.  I was unaware of the pyinstaller bundles you have made for CQ-Editor.  I don’t mind attempting a macOS build since this exercise should benefit the community beyond just CQ-Editor.  However, CQ-Editor does carry a lot of baggage! ;) Since runtime QT and oce libraries need to be packaged with the bundle it makes very large distribution bundles.  I presume you are not building the “one file” executable flavor?  In the macOS world, end users will prefer traditional App bundles either raw or in a DMG file. Furthermore, with Apple tightening their restrictions on distributing apps outside the app store, issues such as signing both the executable and the bundle properly will be important to avoid nuisance security override intervention from the end user (assuming they bother).  The last issue is macOS deprecating support for OpenGL. Catalina is the last “grace period” release before fully dropping OpenGL. Therefore, CQ-Editor with its openGL dependency could have a short lifecycle on macOS unless an alternative cross platform 3D engine is adopted.

adam-urbanczyk, 2020-01-03:
@michaelgale here are the bundles https://github.com/CadQuery/CQ-editor/releases/tag/0.1RC1 they are just zipped portable executables. You could try by running `pyinstaller pyinstaller.spec` in a suitable conda environment (i.e. one that can run CQ-editor) but you will likely need to add mac specific workarounds to the spec file.

For now I wouldn't worry about app store and signing stuff.

michaelgale, 2020-01-05:
1. I've verified CQ-Editor works on macOS Catalina deployed inside a conda environment.
2. I've struggled for several hours building a PyInstaller bundle for macOS.  I can perform an error free build once all the dependencies and configuration is setup in the .spec file.  However, I have not been able to get a working executable.
3. The crash mechanism is due to conflicting QT dynamic libraries:  those bundled with the app bundle and those which are discovered in the miniconda environment.  
4. It appears that this is a very long standing issue (on all platforms).  I am trying to inject debug statements in run.py to see if it can introspect the QT library and find out what is going on.  
I'll keep plugging away to see if I can make progress.  If so, I should be able make a convenient .DMG installer for macOS.  And, if successful, I'll make a automated build/code-signing script for on going releases.

jmwright, 2020-01-06:
@michaelgale Thanks for working on this. There have been all sorts of issues for us related to Qt on MacOS. It seems to be very finicky to get working.

michaelgale, 2020-01-06:
FYI, I now have a working executable in the "dist" folder--a typical case of resolving stale dependencies. However it only "launches" from the command line.  When launched from the macOS Finder or from the App bundle it appears that the executable is "sandboxed" by macOS SIP (System Integrity Protection) for non-signed binaries.  Therefore, the app cannot introspect itself properly, i.e. to get the CASROOT / oce path.  This is necessary to initialize OCCViewer (with its Create method).  That's as far as I can navigate through the stack trace without getting into OCC.Core object files.  

I've had to comment out the `context.Display(his)` call made in viewer.py during app startup to draw the axis lines.  This lets me debug deeper to find that essentially OCC (or likely OCE) cannot initialize OpenGL.
```
 def display_many(self,ais_list,fit=None):

        context = self._get_context()
        
        # for ais in ais_list:
        #     context.Display(ais)

        self.canvas._display.Repaint()

        if self.preferences['Fit automatically'] and fit is None:
            self.fit()
        elif fit:
            self.fit()
```

The stack trace so far...

```
My path: /Users/michaelgale/src/cqgui/CQ-editor/dist/CQ-Editor
casroot path: /Users/michaelgale/src/cqgui/CQ-editor/dist/CQ-Editor/oce
Warning: QApplication was created before pyqtgraph was imported; there may be problems (to avoid bugs, call QApplication.setGraphicsSystem("raster") before the QApplication is created).
My path: /Users/michaelgale/src/cqgui/CQ-editor/dist/CQ-Editor
casroot path: /Users/michaelgale/src/cqgui/CQ-editor/dist/CQ-Editor/oce
Standard GLSL programs are not found in: oce/src/Shaders
CQ-Editor(91024,0x104f73dc0) malloc: *** error for object 0x7fac78911968: pointer being freed was not allocated
CQ-Editor(91024,0x104f73dc0) malloc: *** set a breakpoint in malloc_error_break to debug
Fatal Python error: Aborted

Thread 0x0000700003500000 (most recent call first):
  File "selectors.py", line 558 in select
  File "asyncio/base_events.py", line 1735 in _run_once
  File "asyncio/base_events.py", line 534 in run_forever
  File "site-packages/tornado/platform/asyncio.py", line 148 in start
  File "site-packages/ipykernel/iostream.py", line 78 in _thread_main
  File "threading.py", line 870 in run
  File "threading.py", line 926 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x0000700002ef7000 (most recent call first):
  File "threading.py", line 296 in wait
  File "threading.py", line 552 in wait
  File "site-packages/IPython/core/history.py", line 829 in run
  File "site-packages/IPython/core/history.py", line 58 in needs_sqlite
  File "</Users/michaelgale/src/cqgui/CQ-editor/dist/CQ-Editor/decorator.pyc:decorator-gen-24>", line 2 in run
  File "threading.py", line 926 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Current thread 0x0000000104f73dc0 (most recent call first):
  File "site-packages/OCC/Display/OCCViewer.py", line 174 in Create
  File "site-packages/OCC/Display/qtDisplay.py", line 120 in InitDriver
  File "site-packages/OCC/Display/qtDisplay.py", line 149 in paintEvent
  File "cq_editor/__main__.py", line 42 in main
  File "run.py", line 36 in <module>
Abort trap: 6
logout
Saving session...completed.
```

The references to:

```
My path: /Users/michaelgale/src/cqgui/CQ-editor/dist/CQ-Editor
casroot path: /Users/michaelgale/src/cqgui/CQ-editor/dist/CQ-Editor/oce
```

are my debugging messages to introspect the path of the executable.  Eventually the program crashes with the last clue:

```
Standard GLSL programs are not found in: oce/src/Shaders
```

This message is generated somewhere in OCC.Core and it looks like it cannot find that folder in the App dist bundle; either because it is running in a symlinked sandbox or is trying to resolve a relative path it found in a conda environment.

michaelgale, 2020-01-06:
I think I can all but confirm that its a macOS SIP issue from this lldb trace:

```shell
(lldb) process launch
Process 91622 launched: '/opt/miniconda3/envs/cqgui/bin/python' (x86_64)
Python 3.7.5 (default, Oct 25 2019, 10:52:18)
[Clang 4.0.1 (tags/RELEASE_401/final)] :: Anaconda, Inc. on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> exec(open("run.py").read())
My path: /Users/michaelgale/src/cqgui
casroot path: /Users/michaelgale/src/cqgui/oce
2020-01-05 21:26:03.246278-0500 python[91622:3261625] flock failed to lock maps file: errno = 35
2020-01-05 21:26:03.246955-0500 python[91622:3261625] flock failed to lock maps file: errno = 35
Warning: QApplication was created before pyqtgraph was imported; there may be problems (to avoid bugs, call QApplication.setGraphicsSystem("raster") before the QApplication is created).
My path: /Users/michaelgale/src/cqgui
casroot path: /Users/michaelgale/src/cqgui/oce
2020-01-05 21:26:17.413221-0500 python[91622:3261393] SecTaskLoadEntitlements failed error=22 cs_flags=20, pid=91622
2020-01-05 21:26:17.413342-0500 python[91622:3261393] SecTaskCopyDebugDescription: python3.7[91622]/0#-1 LF=0
2020-01-05 21:26:18.022475-0500 python[91622:3261393] SecTaskLoadEntitlements failed error=22 cs_flags=20, pid=91622
2020-01-05 21:26:18.022593-0500 python[91622:3261393] SecTaskCopyDebugDescription: python3.7[91622]/0#-1 LF=0
Standard GLSL programs are not found in: /Users/michaelgale/src/cqgui/oce/src/Shaders
python(91622,0x1004e7dc0) malloc: *** error for object 0x133f668a8: pointer being freed was not allocated
python(91622,0x1004e7dc0) malloc: *** set a breakpoint in malloc_error_break to debug
Process 91622 stopped
* thread #1, name = 'ZMQbg/1', queue = 'com.apple.main-thread', stop reason = signal SIGABRT
    frame #0: 0x00007fff670247fa libsystem_kernel.dylib`__pthread_kill + 10
libsystem_kernel.dylib`__pthread_kill:
->  0x7fff670247fa <+10>: jae    0x7fff67024804            ; <+20>
    0x7fff670247fc <+12>: movq   %rax, %rdi
    0x7fff670247ff <+15>: jmp    0x7fff6701ea89            ; cerror_nocancel
    0x7fff67024804 <+20>: retq
Target 0: (python) stopped.
```

The app has been "sandboxed" and when the attempt to access the `oce/src/Shaders` folder is made, it triggers macOS SecTask... to verify the executable's entitlements.  It resolves an incorrect path and therefore OpenGL does not initialize thereby crashing the app.

I've read that by signing the executable and the app bundle with valid developer credentials it is possible for the app to run without the SIP sandbox and therefore introspect itself properly to get correct relative paths in its bundle.  I do have an Apple dev account and will try signing the app when I remember how the signing process works!  I usually either have Xcode perform the signing or I have some scripts which do this automatically when generating DMG files for public distribution.

michaelgale, 2020-01-06:
One more note:  If it is enough for end users to download the current app bundle and launch from the command line; then its pretty much job done.  Although clunky to launch (i.e. launch a Terminal, cd to the app path, and execute) it does work.  The app seems to work, the 3D viewer works, valid cq code runs, etc.  

However, the goal should be to offer a turnkey app DMG file.  Once the DMG is mounted, the app bundle can be dragged-dropped into the Applications folder (as is customary in the macOS world!) and thereby launched just like any other App without security warnings etc.  Incidentally, the DMG will be necessary since the resulting dist folder is 1.0 GB!!! Hopefully its inherent compression will bring that down to a sane level.  I have not yet investigated the culprits for the bulk of this greedy use of disk space.

michaelgale, 2020-01-06:
Just for interest, here's a screenshot of CQ-Editor running in macOS Catalina as standalone binary distribution (launched from Terminal only)...
[image]

michaelgale, 2020-01-06:
I suspect the issues described in #106 might be related.  I noticed SecTask... references in the lldb traces of the OP.  Where is the source code for the OpenGL driver initialization in OCC?

jmwright, 2020-01-06:
@michaelgale That's great that you got it running. Maybe a line could be added in the readme telling MacOS users to run it from the command line for now?

In digging around looking through the OpenGL initialization code, I found [this](https://github.com/tpaviot/pythonocc-core/blob/master/src/Visualization/Visualization.h#L27).

Here's the reference for [Cocoa_Window.hxx](https://www.opencascade.com/doc/occt-7.0.0/refman/html/_cocoa___window_8hxx.html)

I believe this is a reference for the [OpenGL graphics driver](https://www.opencascade.com/doc/occt-7.3.0/refman/html/class_open_gl___graphic_driver.html). Here's a snippet of init code from that class.
```c++
  //! Constructor.
  //! @param theDisp connection to display, required on Linux but optional on other systems
  //! @param theToInitialize perform initialization of default OpenGL context on construction
  Standard_EXPORT OpenGl_GraphicDriver (const Handle(Aspect_DisplayConnection)& theDisp,
                                        const Standard_Boolean                  theToInitialize = Standard_True);

  //! Destructor.
  Standard_EXPORT virtual ~OpenGl_GraphicDriver();

  //! Release default context.
  Standard_EXPORT void ReleaseContext();

  //! Perform initialization of default OpenGL context.
  Standard_EXPORT Standard_Boolean InitContext();
```

michaelgale, 2020-01-06:
I think what is happening is as follows...
1. In `OCC/Core/Graphic3d.py`, `ShadersFolder()` is called which in turn calls the binary object function `_Graphic3d.Graphic3d_ShaderProgram_ShadersFolder(*args)`
2. This function is in `oce/Graphic3d_ShaderProgram.cxx` and it is as follows:

```c++
// =======================================================================
// function : ShadersFolder
// purpose  :
// =======================================================================
const TCollection_AsciiString& Graphic3d_ShaderProgram::ShadersFolder()
{
  static Standard_Boolean        THE_IS_DEFINED = Standard_False;
  static TCollection_AsciiString THE_SHADERS_FOLDER;
  if (!THE_IS_DEFINED)
  {
    THE_IS_DEFINED = Standard_True;
    OSD_Environment aDirEnv ("CSF_ShadersDirectory");
    THE_SHADERS_FOLDER = aDirEnv.Value();
    if (THE_SHADERS_FOLDER.IsEmpty())
    {
      OSD_Environment aCasRootEnv ("CASROOT");
      THE_SHADERS_FOLDER = aCasRootEnv.Value();
#ifdef OCE_INSTALL_DATA_DIR
      if (THE_SHADERS_FOLDER.IsEmpty())  {
        THE_SHADERS_FOLDER = OCE_INSTALL_DATA_DIR;
      }
#endif
      if (!THE_SHADERS_FOLDER.IsEmpty())
      {
        THE_SHADERS_FOLDER += "/src/Shaders";
      }
    }

    if (THE_SHADERS_FOLDER.IsEmpty())
    {
      std::cerr << "Both environment variables CSF_ShadersDirectory and CASROOT are undefined!\n"
                << "At least one should be defined to use standard GLSL programs.\n";
      Standard_Failure::Raise ("CSF_ShadersDirectory and CASROOT are undefined");
      return THE_SHADERS_FOLDER;
    }

    const OSD_Path aDirPath (THE_SHADERS_FOLDER);
    OSD_Directory aDir (aDirPath);
    const TCollection_AsciiString aProgram = THE_SHADERS_FOLDER + "/Declarations.glsl";
    OSD_File aProgramFile (aProgram);
    if (!aDir.Exists()
     || !aProgramFile.Exists())
    {
      std::cerr << "Standard GLSL programs are not found in: " << THE_SHADERS_FOLDER.ToCString() << std::endl;
      Standard_Failure::Raise ("CSF_ShadersDirectory or CASROOT is set incorrectly");
      return THE_SHADERS_FOLDER;
    }
  }
  return THE_SHADERS_FOLDER;
}
```

Depending on runtime SIP sandboxing, the path that this function infers may turn out to be incorrect or obscured. Therefore, when it tests `Exists()` at the end of the function it raises the error. It is difficult to tell if this function falls back on the hardcoded `OCE_INSTALL_DATA_DIR` since that will defined when oce is built. If it is, then it is yet another mechanism to not find the Shaders folder.  

At the end of the day, the program crashes because OpenGL does not get initialized since it cannot find its Shaders/etc. resources in `oce/src`.

michaelgale, 2020-01-06:
@adam-urbanczyk  @jmwright  After more investigation, I think I have found second (but related) mechanism for macOS app crashing.  It might be related to #106 and involve a race condition:

1. `MainWindow`'s `__init__` method calls `self.components['object_tree'].addLines()`
2. This triggers an event ` self.sigObjectsAdded.emit(ais_list)`
3. which is responded by `def display_many(self,ais_list,fit=None):` in `viewer.py`
4. This calls `context.Display(ais)` which resolves to the `Display` method of `OCC.Display.qtDisplay.qtViewer3d` which resolves to `_AIS.AIS_InteractiveContext_Display(self, *args)` 

This last call to AIS display occurs before the thread responsible for initializing the OpenGL driver and display context.  We infer this is the case since we see no evidence of the initialization completed (usually a console message saying something about the layer manager).  

There are definitely non thread safe conditions being established at application launch, especially the call to add the axis lines.  When this is commented out, the app may or may not crash depending on the launch context (i.e. direct from terminal, lldb, or via the macOS `launchd` when initiated by the Finder or the terminal `open` command).

michaelgale, 2020-01-06:
p.s. I forgot to attach the stack trace of above...

```shell
* thread #1: tid = 0x34ddc4, 0x00007fff670247fa libsystem_kernel.dylib`__pthread_kill + 10, name = 'ZMQbg/1', queue = 'com.apple.main-thread', stop reason = signal SIGABRT
  thread #2: tid = 0x34de09, 0x00007fff6701f92e libsystem_kernel.dylib`__workq_kernreturn + 10
  thread #5: tid = 0x34df34, 0x00007fff6701f92e libsystem_kernel.dylib`__workq_kernreturn + 10
  thread #6: tid = 0x34df44, 0x00007fff67020ce6 libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #7: tid = 0x34df48, 0x00007fff67022bce libsystem_kernel.dylib`kevent + 10
  thread #8: tid = 0x34df49, 0x00007fff67022bce libsystem_kernel.dylib`kevent + 10
  thread #9: tid = 0x34df4a, 0x00007fff67022bce libsystem_kernel.dylib`kevent + 10
(lldb) thread list all
Process 94969 stopped
* thread #1: tid = 0x34ddc4, 0x00007fff670247fa libsystem_kernel.dylib`__pthread_kill + 10, name = 'ZMQbg/1', queue = 'com.apple.main-thread', stop reason = signal SIGABRT
  thread #2: tid = 0x34de09, 0x00007fff6701f92e libsystem_kernel.dylib`__workq_kernreturn + 10
  thread #5: tid = 0x34df34, 0x00007fff6701f92e libsystem_kernel.dylib`__workq_kernreturn + 10
  thread #6: tid = 0x34df44, 0x00007fff67020ce6 libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #7: tid = 0x34df48, 0x00007fff67022bce libsystem_kernel.dylib`kevent + 10
  thread #8: tid = 0x34df49, 0x00007fff67022bce libsystem_kernel.dylib`kevent + 10
  thread #9: tid = 0x34df4a, 0x00007fff67022bce libsystem_kernel.dylib`kevent + 10
(lldb) thread backtrace all
* thread #1, name = 'ZMQbg/1', queue = 'com.apple.main-thread', stop reason = signal SIGABRT
  * frame #0: 0x00007fff670247fa libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fff670e1bc1 libsystem_pthread.dylib`pthread_kill + 432
    frame #2: 0x00007fff66faba1c libsystem_c.dylib`abort + 120
    frame #3: 0x00007fff670a1647 libsystem_malloc.dylib`malloc_vreport + 548
    frame #4: 0x00007fff670a440e libsystem_malloc.dylib`malloc_report + 151
    frame #5: 0x000000010d851d69 libc++abi.1.dylib`__cxa_end_catch + 137
    frame #6: 0x000000012d3e45a4 _AIS.so`_wrap_AIS_InteractiveContext_Display__SWIG_0(_object*, int, _object**) + 1332
    frame #7: 0x000000012d2ad902 _AIS.so`_wrap_AIS_InteractiveContext_Display(_object*, _object*) + 98
    frame #8: 0x000000010192bff3 libpython3.7m.dylib`_PyMethodDef_RawFastCallKeywords + 259
    frame #9: 0x000000010192bbd2 libpython3.7m.dylib`_PyObject_FastCallKeywords + 306
    frame #10: 0x0000000101a34244 libpython3.7m.dylib`call_function + 708
    frame #11: 0x0000000101a2b3d5 libpython3.7m.dylib`_PyEval_EvalFrameDefault + 7893
    frame #12: 0x0000000101a2906c libpython3.7m.dylib`_PyEval_EvalCodeWithName + 3308
    frame #13: 0x000000010192afd1 libpython3.7m.dylib`_PyFunction_FastCallDict + 481
    frame #14: 0x000000010192ca84 libpython3.7m.dylib`_PyObject_Call_Prepend + 164
    frame #15: 0x000000010192e494 libpython3.7m.dylib`method_call + 36
    frame #16: 0x000000010192c496 libpython3.7m.dylib`PyObject_Call + 246
    frame #17: 0x00000001082dd007 QtCore.so`PyQtSlot::call(_object*, _object*) const + 39
    frame #18: 0x00000001082dcf16 QtCore.so`PyQtSlot::invoke(void**, _object*, void*, bool) const + 374
    frame #19: 0x00000001082dcfac QtCore.so`PyQtSlot::invoke(void**, _object*, void*) const + 12
    frame #20: 0x00000001082dfe0b QtCore.so`qt_metacall_worker(_sipSimpleWrapper*, _typeobject*, _sipTypeDef*, QMetaObject::Call, int, void**) + 683
    frame #21: 0x00000001082dfb32 QtCore.so`qpycore_qobject_qt_metacall(_sipSimpleWrapper*, _sipTypeDef*, QMetaObject::Call, int, void**) + 66
    frame #22: 0x000000010586b37d QtCore`QMetaObject::activate(QObject*, int, int, void**) + 1773
    frame #23: 0x00000001082d971f QtCore.so`pyqtBoundSignal_emit(_object*, _object*) + 655
    frame #24: 0x000000010192bff3 libpython3.7m.dylib`_PyMethodDef_RawFastCallKeywords + 259
    frame #25: 0x0000000101936885 libpython3.7m.dylib`_PyMethodDescr_FastCallKeywords + 85
    frame #26: 0x0000000101a34274 libpython3.7m.dylib`call_function + 756
    frame #27: 0x0000000101a2b3bb libpython3.7m.dylib`_PyEval_EvalFrameDefault + 7867
    frame #28: 0x000000010192b94e libpython3.7m.dylib`function_code_fastcall + 254
    frame #29: 0x0000000101a3424e libpython3.7m.dylib`call_function + 718
    frame #30: 0x0000000101a2b3d5 libpython3.7m.dylib`_PyEval_EvalFrameDefault + 7893
    frame #31: 0x0000000101a2906c libpython3.7m.dylib`_PyEval_EvalCodeWithName + 3308
    frame #32: 0x000000010192afd1 libpython3.7m.dylib`_PyFunction_FastCallDict + 481
    frame #33: 0x000000010192ca84 libpython3.7m.dylib`_PyObject_Call_Prepend + 164
    frame #34: 0x000000010199431a libpython3.7m.dylib`slot_tp_init + 298
    frame #35: 0x000000010199da17 libpython3.7m.dylib`type_call + 295
    frame #36: 0x000000010192bd53 libpython3.7m.dylib`_PyObject_FastCallKeywords + 691
    frame #37: 0x0000000101a34244 libpython3.7m.dylib`call_function + 708
    frame #38: 0x0000000101a2b466 libpython3.7m.dylib`_PyEval_EvalFrameDefault + 8038
    frame #39: 0x000000010192b94e libpython3.7m.dylib`function_code_fastcall + 254
    frame #40: 0x0000000101a3424e libpython3.7m.dylib`call_function + 718
    frame #41: 0x0000000101a2b466 libpython3.7m.dylib`_PyEval_EvalFrameDefault + 8038
    frame #42: 0x0000000101a2906c libpython3.7m.dylib`_PyEval_EvalCodeWithName + 3308
    frame #43: 0x0000000101a28370 libpython3.7m.dylib`PyEval_EvalCode + 48
[... cut ...]
    frame #31: 0x000000010192ca84 libpython3.7m.dylib`_PyObject_Call_Prepend + 164
    frame #32: 0x000000010192e494 libpython3.7m.dylib`method_call + 36
    frame #33: 0x000000010192c496 libpython3.7m.dylib`PyObject_Call + 246
    frame #34: 0x0000000101aeae19 libpython3.7m.dylib`t_bootstrap + 169
    frame #35: 0x0000000101a8aa57 libpython3.7m.dylib`pythread_wrapper + 39
    frame #36: 0x00007fff670e1e65 libsystem_pthread.dylib`_pthread_start + 148
    frame #37: 0x00007fff670dd83b libsystem_pthread.dylib`thread_start + 15
```

jmwright, 2020-01-06:
@michaelgale Thanks for the in-depth debugging! It looks to me like to address #108 and #109 we'll need to coordinate with the upstream OCE/OCCT projects. Is that your view as well?

adam-urbanczyk, 2020-01-06:
@michaelgale I am not sure that I follow. Did you bundle the required shaders? If they are bundled why would some sandboxing mechanism prevent their usage?

michaelgale, 2020-01-06:
@adam-urbanczyk I did indeed bundle the shaders/etc. with the build product and confirmed they were in fact in the bundle.  However, it seems that oce is unable to either resolve the path or open the file.  

This runtime trace is performed with the `self.components['object_tree'].addLines()` statement commented out from the `MainWindow.__init__` method to prevent the race condition and attempt OpenGL initialization.  I put a couple of introspection checks of the environment to ensure it is set correctly which appears at the beginning of this trace.

```shell
$ /Users/michaelgale/src/cqgui/CQ-editor/dist/CQ-Editor/CQ-Editor ; exit;
My path: /Users/michaelgale/src/cqgui/CQ-editor/dist/CQ-Editor
env[CASROOT]: oce
env[CSF_ShadersDirectory]: oce/src/Shaders
Warning: QApplication was created before pyqtgraph was imported; there may be problems (to avoid bugs, call QApplication.setGraphicsSystem("raster") before the QApplication is created).
Standard GLSL programs are not found in: oce/src/Shaders
CQ-Editor(98811,0x107f2fdc0) malloc: *** error for object 0x7fa767481c88: pointer being freed was not allocated
CQ-Editor(98811,0x107f2fdc0) malloc: *** set a breakpoint in malloc_error_break to debug
Fatal Python error: Aborted

Thread 0x0000700008c2f000 (most recent call first):
  File "selectors.py", line 558 in select
  File "asyncio/base_events.py", line 1735 in _run_once
  File "asyncio/base_events.py", line 534 in run_forever
  File "site-packages/tornado/platform/asyncio.py", line 148 in start
  File "site-packages/ipykernel/iostream.py", line 78 in _thread_main
  File "threading.py", line 870 in run
  File "threading.py", line 926 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x0000700008626000 (most recent call first):
  File "threading.py", line 296 in wait
  File "threading.py", line 552 in wait
  File "site-packages/IPython/core/history.py", line 829 in run
  File "site-packages/IPython/core/history.py", line 58 in needs_sqlite
  File "</Users/michaelgale/src/cqgui/CQ-editor/dist/CQ-Editor/decorator.pyc:decorator-gen-24>", line 2 in run
  File "threading.py", line 926 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Current thread 0x0000000107f2fdc0 (most recent call first):
  File "site-packages/OCC/Display/OCCViewer.py", line 174 in Create
  File "site-packages/OCC/Display/qtDisplay.py", line 120 in InitDriver
  File "site-packages/OCC/Display/qtDisplay.py", line 149 in paintEvent
  File "cq_editor/__main__.py", line 28 in main
  File "run.py", line 18 in <module>
Abort trap: 6
```

As we can see, oce throws `Standard GLSL programs are not found in: oce/src/Shaders` error.  The path it reports is correct if it is relative to the executable.  

I will continue to debug to see if I can find any more clues.

adam-urbanczyk, 2020-01-06:
OK, got it. Maybe using `strace` would help (assuming it exists on mac) ?

michaelgale, 2020-01-06:
Experiment:
1. I put this code in `run.py`:
```python
if 'CASROOT' in os.environ:
    print("env[CASROOT]: %s" % (os.environ['CASROOT']))
else:
    print("Did not find CASROOT in environment")

if 'CSF_ShadersDirectory' in os.environ:
    print("env[CSF_ShadersDirectory]: %s" % (os.environ['CSF_ShadersDirectory']))
else:
    print("Did not find CSF_ShadersDirectory in environment")
```

2. I changed `MainWindow.__init__` to: 
```python
from OCC.Core.Graphic3d import Graphic3d_ShaderProgram
from OCC.Core.TCollection import TCollection_AsciiString

class MainWindow(QMainWindow,MainMixin):

    name = 'CQ GUI'
    org = 'CadQuery'

    def __init__(self,parent=None):

        super(MainWindow,self).__init__(parent)
        MainMixin.__init__(self)

        self.setWindowIcon(icon('app'))

        self.viewer = OCCViewer(self)
        if 'CASROOT' in os.environ:
            print("env[CASROOT]: %s" % (os.environ['CASROOT']))
        else:
            print("Did not find CASROOT in environment")

        if 'CSF_ShadersDirectory' in os.environ:
            print("env[CSF_ShadersDirectory]: %s" % (os.environ['CSF_ShadersDirectory']))
        else:
            print("Did not find CSF_ShadersDirectory in environment")

        print(type(Graphic3d_ShaderProgram.ShadersFolder))
        print(dir(Graphic3d_ShaderProgram))
        shf = Graphic3d_ShaderProgram.ShadersFolder()
        print(type(shf))
        x = shf.PrintToString()
        print(type(x))
        print(str(x))
```

When the executable is run from the Terminal:
```shell
$ ./CQ-Editor
My path: /Users/michaelgale/src/cqgui/CQ-editor/dist/CQ-Editor
env[CASROOT]: oce
env[CSF_ShadersDirectory]: oce/src/Shaders
Warning: QApplication was created before pyqtgraph was imported; there may be problems (to avoid bugs, call QApplication.setGraphicsSystem("raster") before the QApplication is created).
env[CASROOT]: oce
env[CSF_ShadersDirectory]: oce/src/Shaders
<class 'function'>
['AttachShader', 'ClearVariables', 'Delete', 'Destroy', 'DetachShader', 'DownCast', 'DynamicType', 'GetHandle', 'GetId', 'GetObject', 'GetRefCount', 'Header', 'IsDone', 'IsInstance', 'IsKind', 'IsNull', 'SetHeader', 'ShaderName_Phong', 'ShaderName_UNKNOWN', 'ShaderObjects', 'ShadersFolder', 'This', 'Variables', '__class__', '__delattr__', '__dict__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__le__', '__lt__', '__module__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '__swig_destroy__', '__weakref__', 'thisown']
<class 'OCC.Core.TCollection.TCollection_AsciiString'>
<class 'str'>
oce/src/Shaders
Layer manager created
Layer dimensions: 679, 555
```
which is what we expect and the app runs without crashing.

[... 93 lines cut ...]
32  CQ-Editor                     	0x0000000100014de4 0x100013000 + 7652

Thread 1:
0   libsystem_pthread.dylib       	0x00007fff670dd818 start_wqthread + 0

Thread 2:
0   libsystem_pthread.dylib       	0x00007fff670dd818 start_wqthread + 0

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000000  rbx: 0x000000010e6e1dc0  rcx: 0x000000010124ca58  rdx: 0x0000000000000000
  rdi: 0x0000000000000307  rsi: 0x0000000000000006  rbp: 0x000000010124ca80  rsp: 0x000000010124ca58
   r8: 0x000000010124cf48   r9: 0x4635906d2be3c70b  r10: 0x000000010e6e1dc0  r11: 0x0000000000000246
  r12: 0x0000000000000307  r13: 0x0000000000000050  r14: 0x0000000000000006  r15: 0x0000000000000016
  rip: 0x00007fff670247fa  rfl: 0x0000000000000246  cr2: 0x0000000100f60fb0
  
Logical CPU:     0
Error Code:      0x02000148
Trap Number:     133
```

Therefore when we call `Graphic3d_ShaderProgram.ShadersFolder()` it either:
i. succeeds and returns a string with the path or
ii. crashes with an invalid memory access

Its difficult to know where it fails exactly, but its the result of a memory allocation possibly with a class constructor?  The only difference is the launch context.  It appears the `oce` paths are correct in the environment before calling `ShadersFolder()`.  I need to find more clues, but at least I can trigger the crash simply by calling `ShadersFolder` directly.

adam-urbanczyk, 2020-01-07:
Side note: the crash is actually caused by this issue https://github.com/CadQuery/cadquery/issues/207  and this https://github.com/swig/swig/issues/1658 . My hypothesis is that it is some kind of ABI mismatch due to wrong compiler version.

michaelgale, 2020-01-07:
Interesting.  Given that the error message makes it to the console, it could be SWIG exception handling and propagation?

rainman110, 2020-02-25:
> My hypothesis is that it is some kind of ABI mismatch due to wrong compiler version.

@adam-urbanczyk  I experienced the same issue in our c++ only library TiGL. Like you, I also suspect a compiler ABI mismatch. Compiling OCCT from scratch (not using conda builds) solves the problem. I still don't understand, what is really happening though.

phrz, 2020-09-30:
Apologies for popping into this older issue, what is the present status of installability and usability on macOS Catalina? (Or the status of such documentation) — I'm very interested in this project.

adam-urbanczyk, 2020-10-01:
You can try the conda package from our (cadquery) channel (note that it will require py3.7 in your env). I don't have access to a mac so I'm not sure if it works. If you try please share the result.