CadQuery/CQ-editor #112: New version of cadquery available. how to update

sander76, 2020-01-18:
I have run both the Win executable and the conda/github path.

Both work, but when checking for updates the app tells me newer versions are available (2.0rc1 and 2 I believe). Looking with 'conda list' I see cadquery version 0.5. 

Can anyone clarify a bit? Should I update from the seemingly outdated v0.5 to 2.0?

Thanks!

jmwright, 2020-01-20:
@sander76 The Win executable should already have 2.0rc1 included with it I think. v0.5 is from the CadQuery 1.0 iteration, and wouldn't work anyway. That version number has probably been left in somewhere it shouldn't as a copy-paste error. I think you're fine with what you're running right now.

sander76, 2020-01-21:
@jmwright Thanks.
I am not familiar with conda. But I think I remember doing `conda list` or something and seeing that v0.5 number.

Could https://github.com/CadQuery/cadquery/blob/master/setup.py#L19 be an error then ?

jmwright, 2020-01-21:
@sander76 setup.py does need to be updated, but I think what's causing the problem in this case is that CadQuery's `__init__.py` `__version__` field hasn't been updated.

https://github.com/CadQuery/cadquery/blob/102c16c14e42b6acca484efbf017f9ac2cc75789/cadquery/__init__.py#L65

jmwright, 2020-01-21:
Here's where CQ-editor is making the check the causes the update message to trigger: https://github.com/CadQuery/CQ-editor/blob/df0c201c5819627253a41b5c5ae57cab04285d3e/cq_editor/utils.py#L117

sander76, 2020-01-22:
Thanks! Slowly but surely I'm getting up to speed with this tool. Really nice!

closed 2020-01-29 by sander76