CadQuery/CQ-editor #114: proper importing for external editor

r4dr3fr4d, 2020-01-30:
Currently the `__init__.py`'s for cq_editor and widgets are empty.  I'd like to be able to use cq_editor viewer alongside vim or another editor - is there an easy way to include everything required?  Would it be a good idea to add to the `__init__` files?  @justbuchanan you seem to have done this successfully - any thoughts?

adam-urbanczyk, 2020-01-30:
There is `Automatic reload and preview` if that is what you mean. Just toggle it and edit your file in the editor of choice. On save to disk the model will be regenerated automatically.

adam-urbanczyk, 2020-02-05:
Did you manage @r4dr3fr4d  to get it working?

r4dr3fr4d, 2020-02-05:
I haven't tried yet, but I don't believe I was thinking clearly when I
posted - I meant to ask for an easy to run cq-editor from the script
itself, after computing a cadquery model - but I suppose that may be as
easy as writing out the file and launching it.  I sort of envision the
workflow being "run the script, which will then show the result."

r4dr3fr4d, 2020-02-08:
@adam-urbanczyk tried the auto-reload - really neat, thank you!  I guess I was used to doing this a little differently with python and openscad.

closed 2020-02-08 by r4dr3fr4d