CadQuery/CQ-editor #115: conda install instructions in README incompete/broken

KAndrsn, 2020-02-01:
I followed the README instructions to install from conda.
> Alternatively one can use conda:
> `conda install -c cadquery -c conda-forge cq-editor`
> and then simply type `cq-editor` to run it.

The install runs and completes.  Re-running it reports that everything is already successfully installed.  But there is no `cq-editor` command.

The cq-editor tree in Miniconda3\pkgs only contains an 'info' directory.  The 'Lib' directory is absent.

Win10 x64
conda 4.8.2 (miniconda)

jmwright, 2020-02-01:
I always use the alternative method where I clone the repo and then run.
```
conda env create -f cqgui_env.yml -n cqgui
```
It works for me. Not sure about the standard method.

KAndrsn, 2020-02-01:
@jmwright : That may work as yet another alternative, but the instructions I referenced do not work.  I would like to see either the install fixed, or the instructions changed to add whatever mojo is missing.  I grabbed the pre-built zip, so I could set least play around some.

adam-urbanczyk, 2020-02-02:
Confirmed @KAndrsn  - for some reason the package is empty. It used to be OK for the 0.1RC1 package, will investigate.

adam-urbanczyk, 2020-02-03:
I think it is solved @KAndrsn , could you try again and let us know?

KAndrsn, 2020-02-03:
@adam-urbanczyk 
Closer.  The exe is in the Scripts directory, but it seems to exit without doing much (there's a reason to using mechanical hard drives still!).  There is no 'cqgui' env, and I can't find that yaml file either.

adam-urbanczyk, 2020-02-04:
Thanks. I confirm that the entry points fail silently. What does work is `python -m cq_editor`. TBC

adam-urbanczyk, 2020-02-04:
It works now on my laptop, the secret was to use `entry_points` in `meta.yaml`. Let me know if it works for you too @KAndrsn !

KAndrsn, 2020-02-05:
@adam-urbanczyk  Confirmed.

adam-urbanczyk, 2020-02-05:
Alright, closing the issue.

closed 2020-02-05 by adam-urbanczyk