CadQuery/CQ-editor #116: UnsatisfiableError on Linux Mint 19.2

KarlZeilhofer, 2020-02-08:
Hi, 

I just tried to install CQ and the editor on Anaconda, which I've installed into /opt/anaconda3 (with sudo). 

```
(base) karl@karls-p70 ~ $ conda install -c cadquery -c conda-forge cq-editor
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: \ 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed                                                                                                                                                            
                                                                                                                                                                  
UnsatisfiableError: The following specifications were found to be incompatible with each other:                                                                   

Package spyder-kernels conflicts for:
anaconda==2019.10 -> spyder-kernels==0.5.2[build='py37_0|py27_0|py36_0']
spyder -> spyder-kernels[version='0.*|<1.0|>=0.1,<1|>=0.4.2,<1|>=0.4.3,<1|>=0.5.0,<1|>=1.8.1,<2.0.0']
anaconda==2019.10 -> spyder==3.3.6=py37_0 -> spyder-kernels[version='>=0.5.0,<1']
cq-editor -> spyder=3.3.4 -> spyder-kernels[version='>=0.4.3,<1']
Package qtconsole conflicts for:
anaconda==2019.10 -> qtconsole==4.5.5=py_0
anaconda==2019.10 -> jupyter==1.0.0=py37_7 -> qtconsole[version='>=4.2']
qtconsole
jupyter -> qtconsole
cq-editor -> spyder=3.3.4 -> qtconsole[version='>=4.2']
spyder -> qtconsole[version='>=4.2|>=4.6.0']
cq-editor -> qtconsole=4.4.4
Package spyder conflicts for:
cq-editor -> spyder=3.3.4
spyder
```

Did I do anything wrong?

Greets, Karl

jmwright, 2020-02-09:
@KarlZeilhofer Does it work if you build from the environment yaml file?
```
conda env create -f cqgui_env.yml -n cqgui
```

KarlZeilhofer, 2020-02-09:
Thanks, yes it did!

[image]

closed 2020-02-09 by KarlZeilhofer