CadQuery/CQ-editor #119: Problem with conda installation : failed with initial frozen solve

lalebarde, 2020-02-22:
Hi, I have installed cadquery successfully with:

`conda install -c conda-forge -c cadquery cadquery=2`

I am trying now to install CQ-editor as explained [here](https://github.com/CadQuery/CQ-editor#installation). It fails with this messages:

```
(base) lalebarde@PC:/opt/openscad$ conda install -c cadquery -c conda-forge cq-editor
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: - 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
-
```
I have also tried the solution [here](https://github.com/CadQuery/CQ-editor/issues/116), but without success:

```
(base) lalebarde@PC:/opt/openscad$ conda env create -f cqgui_env.yml -n cqgui
SpecNotFound: Invalid name, try the format: user/package
```
I am not a conda user, so I don't know if the installation sequence for cadquery, then CQ-editor is correct ?

adam-urbanczyk, 2020-02-22:
Could you share the output of  `conda info --all`?

lalebarde, 2020-02-22:
Here it is:

```
$ conda info --all

     active environment : base
    active env location : /home/alain/anaconda3
            shell level : 1
       user config file : /home/alain/.condarc
 populated config files : 
          conda version : 4.8.2
    conda-build version : 3.18.9
         python version : 3.7.4.final.0
       virtual packages : __glibc=2.24
       base environment : /home/alain/anaconda3  (writable)
           channel URLs : https://repo.anaconda.com/pkgs/main/linux-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/linux-64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /home/alain/anaconda3/pkgs
                          /home/alain/.conda/pkgs
       envs directories : /home/alain/anaconda3/envs
                          /home/alain/.conda/envs
               platform : linux-64
             user-agent : conda/4.8.2 requests/2.22.0 CPython/3.7.4 Linux/4.9.0-11-amd64 debian/9.11 glibc/2.24
                UID:GID : 1969:1969
             netrc file : None
           offline mode : False

# conda environments:
#
base                  *  /home/alain/anaconda3

sys.version: 3.7.4 (default, Aug 13 2019, 20:35:49) 
...
sys.prefix: /home/alain/anaconda3
sys.executable: /home/alain/anaconda3/bin/python
conda location: /home/alain/anaconda3/lib/python3.7/site-packages/conda
conda-build: /home/alain/anaconda3/bin/conda-build
conda-convert: /home/alain/anaconda3/bin/conda-convert
conda-debug: /home/alain/anaconda3/bin/conda-debug
conda-develop: /home/alain/anaconda3/bin/conda-develop
conda-env: /home/alain/anaconda3/bin/conda-env
conda-index: /home/alain/anaconda3/bin/conda-index
conda-inspect: /home/alain/anaconda3/bin/conda-inspect
conda-metapackage: /home/alain/anaconda3/bin/conda-metapackage
conda-render: /home/alain/anaconda3/bin/conda-render
conda-server: /home/alain/anaconda3/bin/conda-server
conda-skeleton: /home/alain/anaconda3/bin/conda-skeleton
conda-verify: /home/alain/anaconda3/bin/conda-verify
user site dirs: 

CIO_TEST: <not set>
CONDA_DEFAULT_ENV: base
CONDA_EXE: /home/alain/anaconda3/bin/conda
CONDA_PREFIX: /home/alain/anaconda3
CONDA_PROMPT_MODIFIER: (base) 
CONDA_PYTHON_EXE: /home/alain/anaconda3/bin/python
CONDA_ROOT: /home/alain/anaconda3
CONDA_SHLVL: 1
GLADE_CATALOG_PATH: :
GLADE_MODULE_PATH: :
GLADE_PIXMAP_PATH: :
PATH: /home/alain/anaconda3/bin:/home/alain/anaconda3/bin:/home/alain/anaconda3/condabin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games:/opt/AsciidocFX:/usr/local/bin:/usr/local/bin:/snap/bin:/home/alain/.cabal/bin
REQUESTS_CA_BUNDLE: <not set>
SSL_CERT_FILE: <not set>
XDG_SEAT_PATH: /org/freedesktop/DisplayManager/Seat0
XDG_SESSION_PATH: /org/freedesktop/DisplayManager/Session0
```

jmwright, 2020-02-22:
@lalebarde Which Linux distro are you running?

lalebarde, 2020-02-22:
Debian Stretch
```
uname -a
Linux PC-ALAIN 4.9.0-11-amd64 #1 SMP Debian 4.9.189-3+deb9u2 (2019-11-11) x86_64 GNU/Linux

```

lalebarde, 2020-02-22:
```
python --version
Python 3.7.4
```

I have updated the install trace in the former post.

jmwright, 2020-02-22:
@lalebarde Please try the following modified line.
```
conda create -f cqgui_env.yml -n cqgui
```

lalebarde, 2020-02-22:
Here is the result:
```
$ conda create -f cqgui_env.yml -n cqgui
Collecting package metadata (current_repodata.json): done
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed

PackagesNotFoundError: The following packages are not available from current channels:

  - cqgui_env.yml

Current channels:

  - https://repo.anaconda.com/pkgs/main/linux-64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/r/linux-64
  - https://repo.anaconda.com/pkgs/r/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.
```

adam-urbanczyk, 2020-02-22:
I cannot reproduce this issue. The following works for me
```python
conda env create -f cqgui_env.yml -n cqgui-debug
```
but I do run it from the root of the repo. This means that the cqgui_env.yml file is in the current dir. Did you do the same @lalebarde ?

lalebarde, 2020-02-22:
What is the root of the repo, `~/anaconda3/`?

adam-urbanczyk, 2020-02-22:
Sorry, I meant this repository, i.e. `CQ-editor`. So you need to clone `CQ-editor`, `cd` into it and then call:
```
conda env create -f cqgui_env.yml -n cqgui
```
Alternatively just specify the full path to the `cqgui_env.yml`.

lalebarde, 2020-02-22:
Thank you very much Adam, it works! And I was faulty, I realise your instructions are the alternative method in your doc. You could just add to go on inside the repo directory cloned. Just one more question please, the installation ends with:

```
Successfully built cadquery
Installing collected packages: cadquery
Successfully installed cadquery-0.5-SNAPSHOT
```

Does it mean the other installation of cadquery with `conda install -c conda-forge -c cadquery cadquery=2`can be suppressed?

adam-urbanczyk, 2020-02-22:
Glad to hear that it worked. This env installs CadQuery straight from git (the version info is maybe misleading - we need to amend setup.py in the CQ repo). You don't need to install separate CQ package anymore. Just activate the environment and call the run.py file (of course first cd into the cloned CQ-editor repo):
```
conda activate cqgui
python run.py
```

lalebarde, 2020-02-22:
Thanks again Adam. FYI, I have tried the former install with `conda` in a fresh Debian buster VM and it leads to the same conflicts errors.

lalebarde, 2020-04-16:
Hi Adam, if I update with `git pull` in the _CQ-Editor_ root folder, is _cadquery_ also updated?

jmwright, 2020-04-16:
@lalebarde I think it's because I forgot to update the version number in CadQuery's [init.py](https://github.com/CadQuery/cadquery/blob/b631cd17df2b4f265d36227c7d72f6072cd7fa30/cadquery/__init__.py#L65) and so CQ-editor thinks it's not updated. 

If you're running Anaconda you should be able to do the following, but the update message will persist. You should have the latest version though.
```
conda env update -f cqgui_env.yml
```

jmwright, 2020-04-16:
@lalebarde The conda line above assumes that you pulled the latest master.

lalebarde, 2020-04-16:
Thanks Jeremy. Yes, I have already the last version. I am now applying your update process. There are some upgrades actually, but conda looks not optimized because it downloads again packets that don't change in version. But at the end, it works anyway.

closed 2020-04-16 by lalebarde

jmwright, 2020-04-16:
@lalebarde I haven't had time to test it yet, but I pushed an update to CadQuery with the version update. If you update your Conda environment again, it should fix the message on CQ-editor. Conda did a big update the first time for me too, but has only been updating CadQuery since then.

jmwright, 2020-04-16:
Confirmed. The dialog in CQ-editor now says that I'm up to date on CadQuery.