CadQuery/CQ-editor #120: cq-editor has stopped finding python import.

KeithSloan, 2020-02-24:
I have a script that was working just fine, but has now stopped with the error message
ModuleNotFoundError: No module named cqgdml.

The code for the module is in cqgdml.py and is in the directory from where I start cq-editor

Code looks like
`import cadquery as cq

# need for reload
import importlib
#import cqgdml
import cqgdml
importlib.reload(cqgdml)
from cqgdml import *

rotation = 0

v = gVol("world")
s = gBox("box1",10,10,10)
m = gMaterial('G4_AIR0x55d123d17ea0')
o = gObject(s,m,None,rotation)
v.addObject(o)
show_object(v.shape2show())
v.exportVol("/tmp/file.gdml")`

KeithSloan, 2020-02-24:
Okay fixed by adding :. to PYTHONPATH

adam-urbanczyk, 2020-03-26:
Seems to be not related to CQ-editor. Closing.

closed 2020-03-26 by adam-urbanczyk