CadQuery/CQ-editor #121: revolve() causes segmentation fault

DanSGiesbrecht, 2020-03-05:
## Modifying `axisStart`/`axisEnd` in revolve() causes a segmentation fault

### System:
- OS: Ubuntu 19.10 (64-bit)
- CQ-editor SHA: eb730a91f68d448461ad1aef9e67d54b2181c238

---

### Working
The following code runs without a problem:

```
import cadquery as cq

# 1.6mm board thickness
wire_thickness = 0.4
board_thickness = 1.6
coil_height = 8.0 - board_thickness
coil_radius = 2.25
coil_pitch = 1.0
coil_sum_length = 0 # calculate length of wire needed to create coil
connection_angle_from_pcb = 90 - 15

antenna = cq.Workplane("XY").circle(wire_thickness).extrude(10.0)\
.faces(">Z").circle(wire_thickness).revolve(connection_angle_from_pcb, (1,2,10), (-1,2,10))\
.rotate((1,0,0), (-1,0,0), 90 - connection_angle_from_pcb)\
.faces(">Y").workplane().circle(wire_thickness).extrude(10.0)\
.faces(">Y").circle(wire_thickness).revolve(45, (1,2,10), (-1,2,10))

show_object(antenna)
```

resulting in the following object:

[image]

---

### Not Working
Changing `revolve(45, (1,2,10), (-1,2,10))` to `revolve(45, (-1,2,10), (1,2,10))` causes a **segmentation fault**, with the following terminal output:

```
(cqgui) gies@monolith:~/CQ-editor$ python run.py 
Namespace(filename=None)
Layer manager created
Layer dimensions: 1245, 885
Fatal Python error: Segmentation fault

Thread 0x00007f6c296f2700 (most recent call first):
  File "/home/gies/miniconda3/envs/cqgui/lib/python3.7/selectors.py", line 468 in select
  File "/home/gies/miniconda3/envs/cqgui/lib/python3.7/asyncio/base_events.py", line 1746 in _run_once
  File "/home/gies/miniconda3/envs/cqgui/lib/python3.7/asyncio/base_events.py", line 538 in run_forever
  File "/home/gies/miniconda3/envs/cqgui/lib/python3.7/site-packages/tornado/platform/asyncio.py", line 153 in start
  File "/home/gies/miniconda3/envs/cqgui/lib/python3.7/site-packages/ipykernel/iostream.py", line 78 in _thread_main
  File "/home/gies/miniconda3/envs/cqgui/lib/python3.7/threading.py", line 870 in run
  File "/home/gies/miniconda3/envs/cqgui/lib/python3.7/threading.py", line 926 in _bootstrap_inner
  File "/home/gies/miniconda3/envs/cqgui/lib/python3.7/threading.py", line 890 in _bootstrap

Thread 0x00007f6c2af35700 (most recent call first):
  File "/home/gies/miniconda3/envs/cqgui/lib/python3.7/threading.py", line 296 in wait
  File "/home/gies/miniconda3/envs/cqgui/lib/python3.7/threading.py", line 552 in wait
  File "/home/gies/miniconda3/envs/cqgui/lib/python3.7/site-packages/IPython/core/history.py", line 829 in run
  File "/home/gies/miniconda3/envs/cqgui/lib/python3.7/site-packages/IPython/core/history.py", line 58 in needs_sqlite
  File "</home/gies/miniconda3/envs/cqgui/lib/python3.7/site-packages/decorator.py:decorator-gen-24>", line 2 in run
  File "/home/gies/miniconda3/envs/cqgui/lib/python3.7/threading.py", line 926 in _bootstrap_inner
  File "/home/gies/miniconda3/envs/cqgui/lib/python3.7/threading.py", line 890 in _bootstrap

Current thread 0x00007f6c6efa4740 (most recent call first):
  File "/home/gies/miniconda3/envs/cqgui/lib/python3.7/site-packages/cadquery/occ_impl/shapes.py", line 231 in clean
  File "/home/gies/miniconda3/envs/cqgui/lib/python3.7/site-packages/cadquery/cq.py", line 3113 in <listcomp>
  File "/home/gies/miniconda3/envs/cqgui/lib/python3.7/site-packages/cadquery/cq.py", line 3113 in clean
  File "/home/gies/miniconda3/envs/cqgui/lib/python3.7/site-packages/cadquery/cq.py", line 2460 in revolve
  File "<string>", line 17 in <module>
  File "/home/gies/CQ-editor/cq_editor/widgets/debugger.py", line 186 in _exec
  File "/home/gies/CQ-editor/cq_editor/widgets/debugger.py", line 213 in render
  File "/home/gies/CQ-editor/cq_editor/__main__.py", line 27 in main
  File "run.py", line 11 in <module>
Segmentation fault (core dumped)
```

adam-urbanczyk, 2020-03-06:
Confirmed bug. I will close and open this in CQ, this is not GUI related. Probably it will be fixed by moving to the newest OCCT version. Please note that his is the geometry you will get:
[image]
Current workaround is to use `fix=False` in the troublesome revolve step:
```python
import cadquery as cq

# 1.6mm board thickness
wire_thickness = 0.4
board_thickness = 1.6
coil_height = 8.0 - board_thickness
coil_radius = 2.25
coil_pitch = 1.0
coil_sum_length = 0 # calculate length of wire needed to create coil
connection_angle_from_pcb = 90 - 15

antenna = cq.Workplane("XY").circle(wire_thickness).extrude(10.0)\
.faces(">Z").circle(wire_thickness).revolve(connection_angle_from_pcb, (1,2,10), (-1,2,10))\
.rotate((1,0,0), (-1,0,0), 90 - connection_angle_from_pcb)\
.faces(">Y").workplane().circle(wire_thickness).extrude(10.0)\
.faces(">Y").circle(wire_thickness).revolve(45, (-1,2,10), (1,2,10),clean=False)

show_object(antenna)
```
What you are probably interested in is using e.g `sweep` function, e.g.
```python
import cadquery as cq

path = (cq.Workplane("YZ")
        .lineTo(0,8)
        .spline([(1,9)],tangents=[(0,10),(1,.1)],includeCurrent=True)
        .lineTo(6,9.5)
        .spline([(7,10.5)],tangents=[(1,.1),(0,10)],includeCurrent=True)
        )

res = cq.Workplane('XY').rect(1,0.2).sweep(path)
```
resulting in:
[image]

DanSGiesbrecht, 2020-03-08:
Thanks for the helpful tip on the sweep function!

adam-urbanczyk, 2020-03-08:
Moved to CQ issue tracker

closed 2020-03-08 by adam-urbanczyk