CadQuery/CQ-editor #122: ModuleNotFoundError: No module named 'spyder.widgets.sourcecode'

l29ah, 2020-03-12:
```
Traceback (most recent call last):
  File "/usr/lib/python-exec/python3.6/cq-editor", line 11, in <module>
    load_entry_point('CQ-editor==0.1.0.dev0', 'gui_scripts', 'cq-editor')()
  File "/usr/lib64/python3.6/site-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib64/python3.6/site-packages/pkg_resources/__init__.py", line 2852, in load_entry_point
    return ep.load()
  File "/usr/lib64/python3.6/site-packages/pkg_resources/__init__.py", line 2443, in load
    return self.resolve()
  File "/usr/lib64/python3.6/site-packages/pkg_resources/__init__.py", line 2449, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/lib64/python3.6/site-packages/cq_editor/__main__.py", line 12, in <module>
    from .main_window import MainWindow
  File "/usr/lib64/python3.6/site-packages/cq_editor/main_window.py", line 8, in <module>
    from .widgets.editor import Editor
  File "/usr/lib64/python3.6/site-packages/cq_editor/widgets/editor.py", line 1, in <module>
    from spyder.widgets.sourcecode.codeeditor import  CodeEditor
ModuleNotFoundError: No module named 'spyder.widgets.sourcecode'
```

Probably depends on really old spyder? I see Gentoo have purged it since it depended on unmaintained and insecure webkit: https://bugs.gentoo.org/709866

adam-urbanczyk, 2020-03-12:
Please use conda to install (see the `README` file):
```
conda install -c cadquery -c conda-forge cq-editor
```

adam-urbanczyk, 2020-03-26:
any updates on this issue @l29ah ? Does the default installation way work?

adam-urbanczyk, 2020-10-09:
Solved by #195

closed 2020-10-09 by adam-urbanczyk