CadQuery/CQ-editor #123: jupyter_client seems to have broken

r4dr3fr4d, 2020-03-14:
Not sure if anyone else is seeing this; I'm on Arch Linux.  Trying different install options to see if the issue persists.
```
Traceback (most recent call last):
  File "/usr/bin/cq-editor", line 11, in <module>
    load_entry_point('CQ-editor==0.1.0.dev0', 'gui_scripts', 'cq-editor')()
  File "/usr/lib/python3.8/site-packages/cq_editor/__main__.py", line 16, in main
    win = MainWindow()
  File "/usr/lib/python3.8/site-packages/cq_editor/main_window.py", line 42, in __init__
    self.prepare_panes()
  File "/usr/lib/python3.8/site-packages/cq_editor/main_window.py", line 98, in prepare_panes
    ConsoleWidget(self),
  File "/usr/lib/python3.8/site-packages/cq_editor/widgets/console.py", line 24, in __init__
    self.kernel_client = kernel_client = self._kernel_manager.client()
  File "/usr/lib/python3.8/site-packages/qtconsole/base_frontend_mixin.py", line 63, in kernel_client
    if kernel_client.channels_running:
  File "/usr/lib/python3.8/site-packages/jupyter_client/client.py", line 141, in channels_running
    self.control_channel.is_alive())
  File "/usr/lib/python3.8/site-packages/jupyter_client/client.py", line 199, in control_channel
    self._control_channel = self.control_channel_class(
TypeError: ChannelABC() takes no arguments

[1]  + 56397 exit 1     cq-editor
```

r4dr3fr4d, 2020-03-14:
^this was observed from installation from the Arch User Repository.  I've tried several times to install via conda and have run into too many problems going that route, so unable to test that way.

adam-urbanczyk, 2020-03-14:
I will not be able to support different installation routes. If you insist on following this path, check if the package versions match what is specified in `cqgui_env.yml`.

NB: one can do local miniconda install without any modifications to the environment. A directory is all you need if proper installation options are selected.

adam-urbanczyk, 2020-03-14:
BTW: I hear users every now and then complaining about conda. If you have some specific issue feel free to ask here. From my point of view it is a huge time-saver - I have no bandwidth to support 3 operating systems (+multiple distributions) without conda.

r4dr3fr4d, 2020-03-14:
I understand - I just wanted to see if anyone else had seen it and would be able to comment.  I agree with your point about conda, I'm in favor of things like that and virtual environments, but for some reason I haven't gotten it to work yet; I'll take a look at miniconda as well.  Thanks for the reply.

jmwright, 2020-03-15:
@r4dr3fr4d Have you tried the pre-built package? https://github.com/CadQuery/CQ-editor/releases/tag/0.1RC1

jmwright, 2020-03-15:
Sorry, nevermind. Forgot about the Jupyter component. Although I wonder if there's a way to get that working from the pre-built package.

jmwright, 2020-03-15:
That AUR is a out of date compared to the latest commit on master. I don't know if there's anything related to Jupyter though.

r4dr3fr4d, 2020-03-15:
@jmwright thanks for the idea.  The AUR actually has a couple of packages, including a `python-cq-editor-git` which builds from here.

cjgk, 2020-03-15:
I have the same error, ultimately:
```
(cadquery) jonas@pop-os:~$ cq-editor 
Traceback (most recent call last):
  File "/home/jonas/miniconda3/envs/cadquery/bin/cq-editor", line 11, in <module>
    sys.exit(main())
  File "/home/jonas/miniconda3/envs/cadquery/lib/python3.7/site-packages/cq_editor/__main__.py", line 16, in main
    win = MainWindow()
  File "/home/jonas/miniconda3/envs/cadquery/lib/python3.7/site-packages/cq_editor/main_window.py", line 38, in __init__
    self.prepare_panes()
  File "/home/jonas/miniconda3/envs/cadquery/lib/python3.7/site-packages/cq_editor/main_window.py", line 94, in prepare_panes
    ConsoleWidget(self),
  File "/home/jonas/miniconda3/envs/cadquery/lib/python3.7/site-packages/cq_editor/widgets/console.py", line 25, in __init__
    self.kernel_client = kernel_client = self._kernel_manager.client()
  File "/home/jonas/miniconda3/envs/cadquery/lib/python3.7/site-packages/qtconsole/base_frontend_mixin.py", line 63, in kernel_client
    if kernel_client.channels_running:
  File "/home/jonas/miniconda3/envs/cadquery/lib/python3.7/site-packages/jupyter_client/client.py", line 141, in channels_running
    self.control_channel.is_alive())
  File "/home/jonas/miniconda3/envs/cadquery/lib/python3.7/site-packages/jupyter_client/client.py", line 200, in control_channel
    socket, self.session, self.ioloop
TypeError: ChannelABC() takes no arguments
`` 
OS is Pop!_OS 19.10
I did install via conda (miniconda) and if at relevant, Ill put the output of 'conda list` from that environment at the end of this message.
I have no experience with conda, but I'm willing to test whatever or supply any information I can.

```
(cadquery) jonas@pop-os:~$ conda list
# packages in environment at /home/jonas/miniconda3/envs/cadquery:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       0_gnu    conda-forge
alabaster                 0.7.12                     py_0    conda-forge
astroid                   2.3.3                    py37_1    conda-forge
attrs                     19.3.0                     py_0    conda-forge
babel                     2.8.0                      py_0    conda-forge
backcall                  0.1.0                      py_0    conda-forge
bleach                    3.1.1                      py_0    conda-forge
ca-certificates           2019.11.28           hecc5488_0    conda-forge
cadquery                  2.0RC1                    py3.7    cadquery
certifi                   2019.11.28       py37hc8dfbb8_1    conda-forge
cffi                      1.14.0           py37hd463f26_0    conda-forge
chardet                   3.0.4                 py37_1003    conda-forge
cloudpickle               1.3.0                      py_0    conda-forge
cq-editor                 master                    py3.7    cadquery
cryptography              2.8              py37h72c5cf5_1    conda-forge
dbus                      1.13.6               he372182_0    conda-forge
decorator                 4.4.2                      py_0    conda-forge
defusedxml                0.6.0                      py_0    conda-forge
docutils                  0.16             py37hc8dfbb8_1    conda-forge
entrypoints               0.3             py37hc8dfbb8_1001    conda-forge
expat                     2.2.9                he1b5a44_2    conda-forge
fontconfig                2.13.1            h86ecdb6_1001    conda-forge
freeimageplus             3.17.0                        0    cadquery
freetype                  2.10.0               he983fc9_1    conda-forge
gettext                   0.19.8.1          hc5be6a0_1002    conda-forge
glib                      2.58.3          py37he00f558_1003    conda-forge
gst-plugins-base          1.14.5               h0935bb2_2    conda-forge
gstreamer                 1.14.5               h36ae1b5_2    conda-forge
icu                       64.2                 he1b5a44_1    conda-forge
idna                      2.9                        py_1    conda-forge
imagesize                 1.2.0                      py_0    conda-forge
importlib-metadata        1.5.0            py37hc8dfbb8_1    conda-forge
importlib_metadata        1.5.0                         1    conda-forge
ipykernel                 5.1.4            py37h5ca1d4c_0    conda-forge
ipython                   7.13.0           py37h5ca1d4c_0    conda-forge
ipython_genutils          0.2.0                      py_1    conda-forge
isort                     4.3.21           py37hc8dfbb8_1    conda-forge
jedi                      0.16.0           py37hc8dfbb8_1    conda-forge
[... 83 lines cut ...]
sphinxcontrib-jsmath      1.0.1                      py_0    conda-forge
sphinxcontrib-qthelp      1.0.3                      py_0    conda-forge
sphinxcontrib-serializinghtml 1.1.4                      py_0    conda-forge
spyder                    3.3.4                    py37_1    conda-forge
spyder-kernels            0.5.2                    py37_0    conda-forge
sqlite                    3.30.1               hcee41ef_0    conda-forge
tbb                       2020.1               hc9558a2_0    conda-forge
testpath                  0.4.4                      py_0    conda-forge
tk                        8.6.10               hed695b0_0    conda-forge
tornado                   6.0.4            py37h8f50634_1    conda-forge
[... cut ...]
wurlitzer                 2.0.0                    py37_0    conda-forge
xorg-libxau               1.0.9                h14c3975_0    conda-forge
xorg-libxdmcp             1.1.3                h516909a_0    conda-forge
xz                        5.2.4             h14c3975_1001    conda-forge
zeromq                    4.3.2                he1b5a44_2    conda-forge
zipp                      3.1.0                      py_0    conda-forge
zlib                      1.2.11            h516909a_1006    conda-forge
```

adam-urbanczyk, 2020-03-15:
Looks like it is related to https://github.com/jupyter/jupyter_client/issues/523.

Could you try downgrading `jupyter_client` to e.g. 5.3.4?

cjgk, 2020-03-15:
I downgraded `jupyter_client` to 5.3.4 (removed the package, installed the specific version, reran the qc-editor install to get all the packages back) and now it launches!
However, now it dumps core when resizing the window.
Since that's probably not related to this, I'll investigate that a little further and if need be, I'll open another issue.
Thank you for your quick, correct response and your work on CQ-editor.

jmwright, 2020-03-15:
@cjgk This issue is a little old now, but have a look at this comment.

https://github.com/CadQuery/CQ-editor/issues/50#issuecomment-510065403

cjgk, 2020-03-15:
@jmwright Worked! And it was in the readme too, embarrassing...
Thank you!

r4dr3fr4d, 2020-03-18:
downgrading seems to be the appropriate solution.  Will close since conda is the supported installation method.

closed 2020-03-18 by r4dr3fr4d