CadQuery/CQ-editor #126: FR: expose context detection

Andrei-Pozolotin, 2020-03-23:
similar to 

```
if __name__ == "__main__":
   "this is main context"
```

cq-editor should set a global variable or an environment variable
to indicate to the client script that is is being ran inside the cq-editor

that would help client libraries such as cq-parts alter rendering behavior

sure this can be done via stack traversal, but "official" integration would be better

adam-urbanczyk, 2020-03-23:
Sounds reasonable. What do you propose to put `globals`? Now the `__name__` is `temp`

Andrei-Pozolotin, 2020-03-23:
1. currently cq-parts expects environment variable

https://github.com/cqparts/cqparts/blob/master/src/cqparts/display/cqparts_server.py#L22
```
ENVVAR_SERVER = 'CQPARTS_SERVER'
@map_environment(
    name='cqparts_server',
    order=50,
    condition=lambda: ENVVAR_SERVER in os.environ,
)
class CQPartsServerDisplayEnv(DisplayEnvironment):
```

https://github.com/cqparts/cqparts/blob/master/src/cqparts/display/freecad.py#L10
```
@map_environment(
    name="freecad",
    order=10,
    condition=lambda: 'MYSCRIPT_DIR' in os.environ,
)
class FreeCADDisplayEnv(DisplayEnvironment):
```

2. so perhaps both a global variable and an environment variable are needed

3. as for key/value, it is better be long, explicit, and specific, i.e. 
```
CADQUERY_EDITOR_ENGINE
```

adam-urbanczyk, 2020-03-23:
I see that you are referring to some display functionality of CQparts. How do you want to integrate it  with CQ-editor?

Andrei-Pozolotin, 2020-03-23:
I do not know. 
I was expecting to find some analogue to `show_object` but do not see any.
What would you suggest?

jmwright, 2020-03-23:
cqparts has a `display` method that takes the place of `show_object`.

https://github.com/cqparts/cqparts/blob/3302f2ccd56b92df2c368d924e6c843330712d70/tests/t_cqparts/test_display.py#L17

Andrei-Pozolotin, 2020-03-23:
ok, either way :-)
how do you see total integration of `cq` + `cq-editor` +`cq-parts` ?

plus: does `cq-edtor` pan to allow back propagation of user ui assembly edits back to assembly (via import/export)?

jmwright, 2020-03-24:
I personally would like to see the CQ ecosystem standardize on `show_object`.

adam-urbanczyk, 2020-03-26:
@Andrei-Pozolotin  I think the way to go regarding CQParts support is to extend `show_object` (or actually the underlying method `cq_utils.make_AIS` ) to support CQParts objects. Relevant issue to track this is #17 .

Regarding your second point, no specific plans at the moment.

adam-urbanczyk, 2023-04-29:
Solved by #399: `__name__` is set to `__cq_main__`

closed 2023-04-29 by adam-urbanczyk