CadQuery/CQ-editor #128: Export as STL / STEP not working on macOS 10.15.4

jwhevans, 2020-03-28:
[image]

I am starting cq-editor from within an Anaconda environment
  * Python 3.7.1
  * CADQuery 2 https://github.com/CadQuery/cadquery

I get the following warning when starting cq-editor

`Warning: QApplication was created before pyqtgraph was imported; there may be problems (to avoid bugs, call QApplication.setGraphicsSystem("raster") before the QApplication is created).
Namespace(filename=None)
Layer manager created
Layer dimensions: 800, 668`

adam-urbanczyk, 2020-03-28:
You need to select an object in the object tree widget for this action to be enabled.

jwhevans, 2020-03-28:
Thanks for the quick response.  It's taking me a little while to get used to the layout and interactions.  It works just like you mentioned.

closed 2020-03-28 by jwhevans