CadQuery/CQ-editor #134: Error from cq_editor/widgets/viewer.py", line 162, in display_many

dpellegr, 2020-04-11:
Testing on the latest releases of ipykernel (5.2.0) and jupyter_client (6.1.2).

Issue https://github.com/CadQuery/CQ-editor/issues/123 is gone, but now I get:

```
$ cq-editor 
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/cq_editor/widgets/viewer.py", line 162, in display_many
    context.Display(ais)
RuntimeError: Units_NoSuchType
BAD LEXICON descriptor
Aborted (core dumped)
```

Is this still upstream related?

adam-urbanczyk, 2020-04-11:
Sounds similar to #9 . Do you execute via `run.py`? Is `CASROOT variable set in your env?

dpellegr, 2020-04-11:
Yes, you are right. Unsetting `CASROOT` helps. Unfortunately the GUI does not show up. I get no output, but 100% CPU usage for >10 minutes, until I close the terminal.
What is weird is that pytest gets executed very quickly, and in the process I see the GUI appearing with some content in it.
BTW running from the package is ok for me, but I wanted to try the bleeding edge to experiment with https://github.com/CadQuery/cadquery/issues/242#issuecomment-569074833 ...and I am on ArchLinux :)

adam-urbanczyk, 2020-04-11:
How are you installing the dependencies (OCE, PythonOCC)? I remember seeing similar issue already (see last comment in #9). I never understood the root cause (OCE was stuck on some data file parsing AFAIR) but chaniging the compilation flags helped (hence I'm asking about the dependencies - the recommended way is to install everything form the cadquery conda channel). If you want to dig into it:

1. Run CQ-editor with strace and paste the result
2. Run with gdb and paste the backtrace at the moment it gets stuck

BTW: if bleeding edge is what you want just install the cadquery conda package versioned as master.

dpellegr, 2020-04-12:
Since I had to dig, I decided to do it properly. 
I got rid of `OCE` and `cadquery/pythonocc-core` replacing them with of `opencascade` and `tpaviot/pythonocc-core v7.4.0`.
I applied a couple of trivial fixes to the imports: https://github.com/CadQuery/cadquery/compare/master...dpellegr:master

...and this test script run just fine:

```
import cadquery as cq
from cadquery.occ_impl.exporters import exportShape

result = cq.Workplane("XY").box(3, 2, 1)

f = open("test.stl", "w")
exportShape(result, "STL", f, tolerance=0.1)

```

I still have to check about CQ-editor, but at this point I feel like I have satisfying working environment already.

PS. I never liked over-complicated managers hiding the working flow. I believe that if I have the source code and the appropriate building tools, everything should come together effortlessly in my terminal.

adam-urbanczyk, 2020-04-13:
Current CQ-editor master will not work with this version of OCCT/PythonOCC I'm afraid.

dpellegr, 2020-04-13:
Yes, I saw. I am looking into merging PythonOCC-core 7.4 into your fork,
although the best course of action would probably be getting rid of it,
running CQ-editor straight from the upstream PythonOCC.
Would you mind commenting a bit on the rationale of that fork?
Also I guess we can close this issue and migrate the discussion into a new
one about having CQ-editor on OCCT/PythonOCC. Are you with me?

adam-urbanczyk, 2020-04-13:
Actually we are most likely going to move to a custom OCCT wrapper for both CQ-editor and CQ. Take a look the the OCP repo, and some related issues in the issue tracker of CQ. There is a OCP branch of CQ and I'm also working on a OCP branch of CQ-editor. The rationale was changes related to visualization in the  + ease of debugging.

dpellegr, 2020-04-13:
Wow, you are taking it very widely, writing your own C++ wrapping tool!
I also dislike swig for anything that doesn't need to provide bindings to a
thousand languages, but...
I really hope that this will pay, although I am worried about the resources
needed to step off the shoulders of the giants.
Anyway if I can be of any help, my C++ background is way stronger than my
Python one :)

adam-urbanczyk, 2020-04-22:
Actually it is mostly working, you can download OCP via conda from the cadquery channel:
```
conda install -c cadquery ocp
```

winksaville, 2020-06-23:
I've ran into this error too on an uptodate Arch Linux sytem:
```
$ uname -a
Linux 3900x 5.7.4-arch1-1 #1 SMP PREEMPT Thu, 18 Jun 2020 16:01:07 +0000 x86_64 GNU/Linux
```
 With conda and running python:
```
$ python --version
Python 3.7.6
```
Unsetting CASROOT resolves the problem as it did for #9 : 
```
$ echo $CASROOT
/usr
(cadquery) wink@3900x:~
$ cq-editor 
Traceback (most recent call last):
  File "/opt/anaconda/envs/cadquery/lib/python3.7/site-packages/cq_editor/widgets/viewer.py", line 162, in display_many
    context.Display(ais)
RuntimeError: Units_NoSuchType
BAD LEXICON descriptor
Aborted (core dumped)
(cadquery) wink@3900x:~
$ export CASROOT=
(cadquery) wink@3900x:~
$ echo $CASROOT

(cadquery) wink@3900x:~
$ cq-editor
Namespace(filename=None)
TKOpenGl | Type: Error | ID: 0 | Severity: High | Message:
  Failed to compile shader object. Compilation log:
0(239) : error C1503: undefined variable "occFragColor"

TKOpenGl | Type: Error | ID: 0 | Severity: High | Message:
  Error! FBO blitting has failed
Layer manager created
Layer dimensions: 1646, 357
TKOpenGl | Type: Error | ID: 0 | Severity: High | Message:
  Font 'Courier' is not found in the system!
```
So there are still a few errors, but for the moment adding `export CASROOT=` to my .bashrc gets me going.

I did try
```
$ conda install -c cadquery -c conda-forge cq-editor=master
...
Proceed ([y]/n)? y

Downloading and Extracting Packages
_openmp_mutex-4.5    | 5 KB      | ###################################################################################################################################### | 100% 
libblas-3.8.0        | 10 KB     | ###################################################################################################################################### | 100% 
libopenblas-0.3.9    | 7.8 MB    | ###################################################################################################################################### | 100% 
llvm-openmp-10.0.0   | 2.8 MB    | ###################################################################################################################################### | 100% 
cq-editor-master     | 111 KB    | ###################################################################################################################################### | 100% 
libcblas-3.8.0       | 10 KB     | ###################################################################################################################################### | 100% 
liblapack-3.8.0      | 10 KB     | ###################################################################################################################################### | 100% 
Preparing transaction: done
Verifying transaction: done
Executing transaction: done

$ cq-editor 
Traceback (most recent call last):
  File "/opt/anaconda/envs/py3.7-3dstuff/bin/cq-editor", line 7, in <module>
    from cq_editor.__main__ import main
  File "/opt/anaconda/envs/py3.7-3dstuff/lib/python3.7/site-packages/cq_editor/__main__.py", line 12, in <module>
    from .main_window import MainWindow
  File "/opt/anaconda/envs/py3.7-3dstuff/lib/python3.7/site-packages/cq_editor/main_window.py", line 8, in <module>
    from .widgets.viewer import OCCViewer
  File "/opt/anaconda/envs/py3.7-3dstuff/lib/python3.7/site-packages/cq_editor/widgets/viewer.py", line 11, in <module>
    from OCP.AIS import AIS_Shaded,AIS_WireFrame, AIS_ColoredShape, \
ModuleNotFoundError: No module named 'OCP'
```
So I tried to install `ocp`, but that failed:
```
$ conda install -c cadquery ocp
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: - 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed                                                                                                                                                                           

UnsatisfiableError: The following specifications were found to be incompatible with each other:

Output in format: Requested package -> Available versionsThe following specifications were found to be incompatible with your CUDA driver:

  - feature:/linux-64::__cuda==10.2=0

Your installed CUDA driver is: 10.2
```
Seems like my 10.2 CUDA driver  for my Nvidia 1660 GTX video card isn't compatible :(

So for now I'll stick with just clearing CASROOT, but would like to know what I need to do to run `master`. Any other recommendations?

adam-urbanczyk, 2020-06-23:
OK, I think that there are multiple things going on here.

1) `CASROOT` is set by upstream, if you run via `python run.py` then it should be automatically deleted. I might need to check why running via `cq-editor` does to do that.
2) Conda package spec needs to be updated to reflect the OCP dependency
3) Could try doing that in a new conda env? We don't depend on CUDA nor do anything with it.

adam-urbanczyk, 2020-06-23:
At least (2) is resolved now. Feel free to open a new issue to discuss (3) @winksaville .

winksaville, 2020-06-23:
Status.

(1) IS resolved on master:
```
(cqgui-master) wink@3900x:~
$ echo $CASROOT
/usr
(cqgui-master) wink@3900x:~
$ cq-editor 
Namespace(filename=None)
TKOpenGl | Type: Other | ID: 0 | Severity: Medium | Message:
  OpenGl_Window::CreateWindow: window Visual is incomplete: no stencil buffer
Font_FontMgr, warning: unable to find font 'Courier' [regular]; 'Cantarell' [aspects: regular,bold] [paths: /usr/share/fonts/cantarell/Cantarell-Regular.otf;/usr/share/fonts/cantarell/Cantarell-Bold.otf] is used instead
```
But still FAILS with the release version. I retested this AM and with a new env:
```
$ echo $CASROOT
/usr
(cq-editor) wink@3900x:~
$ cq-editor
Traceback (most recent call last):
  File "/opt/anaconda/envs/cq-editor/lib/python3.7/site-packages/cq_editor/widgets/viewer.py", line 162, in display_many
    context.Display(ais)
RuntimeError: Units_NoSuchType
```

(2) IS resolved, installs fine now.
(3) IS resolved, no issues with CUDA.