CadQuery/CQ-editor #140: Updated modules not taken into account

lalebarde, 2020-05-03:
I don't know if it is a _CadQuery_ or a _CQ-editor_ issue: when you import a module, if it is updated, updates are not taken into account in _CadQuery/CQ-editor_.

Steps to reproduce:

- Open CadQuery with CQ-editor

- Import a python module and use a function in the code

- Run the CQ script, it works

- Add a function in the python module

- Change the CQ code to use this new function

- Run again, it fails

- Quit CadQuery and open it again

- Run again, it works

jmwright, 2020-05-04:
Depending on how CQ-editor handles the execution environment this may not apply, but in the CQ 1.x FreeCAD workbench, we had to unload modules after each execution to prevent this sort of thing.

https://github.com/jmwright/cadquery-freecad-module/pull/103/files

adam-urbanczyk, 2020-05-04:
So far you need to manually reload your modules. This is a duplicate of #99 .

closed 2020-05-04 by adam-urbanczyk