CadQuery/CQ-editor #146: Errors downstream when calling show_object on lists

maderero, 2020-05-22:
Traceback 1:
```
[2020-05-22 15:44:15.317691] ERROR: CQ GUI: Uncaught exception occurred
Traceback (most recent call last):
  File "/home/dsw/Desktop/CQ-editor-master/cq_editor/widgets/cq_object_inspector.py", line 138, in setObject
    while cq_obj.parent:
AttributeError: 'list' object has no attribute 'parent'
```

Traceback 2:
```
[2020-05-22 16:41:18.376750] ERROR: CQ GUI: Uncaught exception occurred
Traceback (most recent call last):
  File "/home/dsw/Desktop/CQ-editor-master/cq_editor/widgets/object_tree.py", line 154, in <lambda>
    self.export('step'))
  File "/home/dsw/Desktop/CQ-editor-master/cq_editor/widgets/object_tree.py", line 342, in export
    if fname is not '':
  File "/home/dsw/Desktop/CQ-editor-master/cq_editor/cq_utils.py", line 59, in export
    comp = to_compound(obj)
  File "/home/dsw/Desktop/CQ-editor-master/cq_editor/cq_utils.py", line 26, in to_compound
    for o in obj: vals.extend(o.vals())
AttributeError: 'list' object has no attribute 'vals'
```

Notes:
This occurs when using show_object on a list of cadquery objects and clicking on any part of any part of the shown objects in the main window.

From what I can gather, show_object works fine on lists of cadquery objects since the cq_utils to_compound method can take an iterable as its first argument when making a shape. However, addObject then goes on to create a new ObjectTreeItem using the list as its 'shape.' Later down the line, when interacting with the main window, this causes issues. Clicking the object in the main window yields the first error, and trying to export the object from the object inspector causes the second error.

adam-urbanczyk, 2020-05-22:
Interesting, could you post a simple example?

maderero, 2020-05-22:
Away from my desk atm. I'll get back to you in a bit.

maderero, 2020-05-22:
Run this in the editor:
```
sp = cq.Workplane('XY').sphere(1)
sp2 = sp.translate((5, 0, 0))

spheres = [sp, sp2]
```

Then in console call:
`show_object(spheres)`

Afterwards click on model in the main window and you should see the first error. 
Try right clicking the model in the object inspector and try to export to get the second error.
I added the line from my pr and it seems to fix that.

However, if you run this in the editor:
```
sp = cq.Workplane('XY').sphere(1)
sp2 = sp.translate((5, 0, 0))

spheres = [sp, sp2]
show_object(spheres)
```
I end up getting the same issues as before. Is there some difference with calling show_object from the console vs the editor?

adam-urbanczyk, 2020-05-23:
Clear, why are you trying to show a list? Is it suggested somewhere in the docs? As far as I remember the intent was to support `cq.Workplane` or `cq.Shape`.

maderero, 2020-05-23:
Hmm I see what you're saying. I may have just tried it on my own and taken the lack of an immediate error as evidence that it should work.

adam-urbanczyk, 2020-05-23:
Well, CQ-editor (and Python in general) does not check the types too strictly. Do you still want to merge the PR?

maderero, 2020-05-23:
I think that if the intention was for the function to only support those two types then the changes are unnecessary. If you don't think there's much need for it either then let's just close it.

closed 2020-05-26 by maderero