CadQuery/CQ-editor #153: cq-editor master ERROR: CQ GUI: Uncaught exception occured

winksaville, 2020-06-23:
On my Arch Linux system:
```
(base) wink@3900x:~/aur/inxi (master)
$ inxi -Fxz
System:    Kernel: 5.7.4-arch1-1 x86_64 bits: 64 compiler: gcc v: 10.1.0 Desktop: Cinnamon 4.6.4 Distro: Arch Linux 
Machine:   Type: Desktop Mobo: ASRock model: X570 Creator serial: <filter> UEFI: American Megatrends v: P2.10 
           date: 12/16/2019 
CPU:       Topology: 12-Core model: AMD Ryzen 9 3900X bits: 64 type: MT MCP arch: Zen 2 L2 cache: 6144 KiB 
           flags: avx avx2 lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm bogomips: 182491 
           Speed: 4071 MHz min/max: 2200/3800 MHz Core speeds (MHz): 1: 4077 2: 4076 3: 4075 4: 4075 5: 4076 6: 4076 7: 4076 
           8: 4077 9: 4076 10: 4076 11: 4074 12: 4074 13: 4076 14: 4088 15: 4075 16: 4075 17: 4075 18: 4077 19: 4076 20: 4077 
           21: 4074 22: 4074 23: 4074 24: 4075 
Graphics:  Device-1: NVIDIA TU116 [GeForce GTX 1660] vendor: Micro-Star MSI driver: nvidia v: 440.82 bus ID: 33:00.0 
           Display: x11 server: X.org 1.20.8 driver: nvidia unloaded: modesetting resolution: <xdpyinfo missing> 
           Message: Unable to show advanced data. Required tool glxinfo missing. 
Audio:     Device-1: NVIDIA TU116 High Definition Audio vendor: Micro-Star MSI driver: snd_hda_intel v: kernel 
           bus ID: 33:00.1 
           Device-2: Advanced Micro Devices [AMD] Starship/Matisse HD Audio vendor: ASRock driver: snd_hda_intel v: kernel 
           bus ID: 35:00.4 
           Sound Server: ALSA v: k5.7.4-arch1-1 
Network:   Device-1: Intel I211 Gigabit Network vendor: ASRock driver: igb v: 5.6.0-k port: b000 bus ID: 28:00.0 
           IF: enp40s0 state: down mac: <filter> 
           Device-2: Intel Wi-Fi 6 AX200 driver: iwlwifi v: kernel port: b000 bus ID: 29:00.0 
           IF: wlp41s0 state: down mac: <filter> 
           Device-3: Aquantia AQC107 NBase-T/IEEE 802.3bz Ethernet [AQtion] vendor: ASRock driver: atlantic v: kernel 
           port: a000 bus ID: 2e:00.0 
           IF: enp46s0 state: up speed: 1000 Mbps duplex: full mac: <filter> 
Drives:    Local Storage: total: 465.76 GiB used: 90.61 GiB (19.5%) 
           ID-1: /dev/nvme0n1 vendor: Sabrent model: Rocket 4.0 500GB size: 465.76 GiB 
           ID-2: /dev/nvme1n1 vendor: Sabrent model: Rocket 4.0 500GB size: 465.76 GiB 
Partition: ID-1: / size: 491.15 GiB used: 90.54 GiB (18.4%) fs: ext4 dev: /dev/dm-2 
           ID-2: /boot size: 1.86 GiB used: 71.2 MiB (3.7%) fs: vfat dev: /dev/nvme0n1p1 
Swap:      Alert: No Swap data was found. 
Sensors:   System Temperatures: cpu: 80.1 C mobo: N/A 
           Fan Speeds (RPM): N/A 
Info:      Processes: 490 Uptime: 23h 01m Memory: 31.29 GiB used: 6.47 GiB (20.7%) Init: systemd Compilers: gcc: 10.1.0 
           clang: 10.0.0 Shell: bash v: 5.0.17 inxi: 3.1.03
```
I installed `cq-editor=master` in a new env:
```
(base) wink@3900x:~
$ conda create --name cqgui-master
Collecting package metadata (current_repodata.json): done
Solving environment: done

## Package Plan ##

  environment location: /opt/anaconda/envs/cqgui-master

Proceed ([y]/n)? 

Preparing transaction: done
Verifying transaction: done
Executing transaction: done
#
# To activate this environment, use
#
#     $ conda activate cqgui-master
#
# To deactivate an active environment, use
#
#     $ conda deactivate

(base) wink@3900x:~
$ conda activate cqgui-master
(cqgui-master) wink@3900x:~
$ conda install -c cadquery -c conda-forge cq-editor=master
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
[... 323 lines cut ...]
Namespace(filename=None)
TKOpenGl | Type: Other | ID: 0 | Severity: Medium | Message:
  OpenGl_Window::CreateWindow: window Visual is incomplete: no stencil buffer
Font_FontMgr, warning: unable to find font 'Courier' [regular]; 'Cantarell' [aspects: regular,bold] [paths: /usr/share/fonts/cantarell/Cantarell-Regular.otf;/usr/share/fonts/cantarell/Cantarell-Bold.otf] is used instead
```
But then when I tried to `run` the default script:
```
import cadquery as cq
result = cq.Workplane("XY" ).box(3, 3, 0.5).edges("|Z").fillet(0.125)
```
It failed with the `Uncaught exception occurred`:
```
[2020-06-23 20:20:13.201055] WARNING: Code Editor: could not open box.py
[2020-06-23 20:20:16.548558] ERROR: CQ GUI: Uncaught exception occurred
Traceback (most recent call last):
  File "/opt/anaconda/envs/cqgui-master/lib/python3.7/site-packages/cq_editor/widgets/object_tree.py", line 262, in addObjects
    ais,shape_display = make_AIS(obj.shape,obj.options)
  File "/opt/anaconda/envs/cqgui-master/lib/python3.7/site-packages/cq_editor/cq_utils.py", line 49, in make_AIS
    ais = AIS_ColoredShape(shape.wrapped)
TypeError: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. OCP.AIS.AIS_ColoredShape(theShape: OCP.TopoDS.TopoDS_Shape)
    2. OCP.AIS.AIS_ColoredShape(theShape: OCP.AIS.AIS_Shape)

Invoked with: class<'TopoDS_Compound'; id:551605425>
```

adam-urbanczyk, 2020-06-24:
Thanks for the very detailed bug report! The root cause is the CQ version (2.0 uses PythonOCC). Could you install the master version of cadquery as well (`conda install -c cadquery -c conda-forge cadquery=master`).

winksaville, 2020-06-24:
Sure.

But, I'm wondering if you should change the dependencies in
cq-editor=master to cadquery=master?

adam-urbanczyk, 2020-06-24:
You are right. I'd like to keep it to a release version, but the current released version is not compatible. I'll amend the conda recipe.

winksaville, 2020-06-24:
SG, I'll retest when you're done.

adam-urbanczyk, 2020-06-24:
#154 is merged, let me know if the current conda package works out of the box.

winksaville, 2020-06-24:
Rested and worked, closing

closed 2020-06-24 by winksaville