CadQuery/CQ-editor #156: ModuleNotFoundError for local files

winksaville, 2020-06-26:
I was testing the following file from a clone of [this repo](https://github.com/winksaville/cadquery-wing1) and all was working fine for a few hours. And then "suddenly" cq-editor reported that the ModuleNotFoundError:

```
from mh49 import mh49
from scale import scaleListOfTuple
from pprint import pprint
import cadquery as cq

chord: float = 50
span: float = 50

# Normalize mh49 to 1. Assumes first point is
# the length of the chord and will scale all
# of the points to 1/mh49[0][0]
scaleFactor = 1/mh49[0][0]
print(f'scaleFactor={scaleFactor}')
nMh49 = scaleListOfTuple(mh49, scaleFactor)

# Scale the normalized mh49 to the size of the chord
# and turn it into a list
sMh49 = scaleListOfTuple(nMh49, chord)
#print(f'sMh49={sMh49}')

result = cq.Workplane("XY").polyline(sMh49).close().extrude(span)
pprint(vars(result))
#show_object(result)
```
Here is a screenshot of the ModuleNotFoundError:

[image]

But if I execute `python wing1.py` there is no error:
```
(cqgui-master) wink@3900x:~/prgs/CadQuery/projects/wing1 (master)
$ python wing1.py 
scaleFactor=0.001
{'_tag': None,
 'ctx': <cadquery.cq.CQContext object at 0x7f44aa009190>,
 'obj': None,
 'objects': [<cadquery.occ_impl.shapes.Compound object at 0x7f44a9918f50>],
 'parent': <cadquery.cq.Workplane object at 0x7f44a9908fd0>,
 'plane': <cadquery.occ_impl.geom.Plane object at 0x7f44a9918d50>}
```

I then created a new env, cqgui-master2, and that didn't help:

[image]

But `python wing1.py` still works:
```
(cqgui-master2) wink@3900x:~/prgs/CadQuery/projects/wing1 (master)
$ python wing1.py 
scaleFactor=0.001
{'_tag': None,
 'ctx': <cadquery.cq.CQContext object at 0x7f3e78f386d0>,
 'obj': None,
 'objects': [<cadquery.occ_impl.shapes.Compound object at 0x7f3e78c08350>],
 'parent': <cadquery.cq.Workplane object at 0x7f3e78bfad10>,
 'plane': <cadquery.occ_impl.geom.Plane object at 0x7f3e78c081d0>}
```

I assume I've got some enviorment problem,  again :(

If you can give me some hints on the best way to debug this,
like maybe there is a "verbose" debug mode for cq-editor.

I'll be glad to enable that or anything else you'd like me to try
as I'm guessing you'll not be able to reproduce the problem.

adam-urbanczyk, 2020-06-26:
Your custom module has to be on the PYTHONPATH. The current directory is probably not added to the PYTHONPATH when you start using the `cq-editor` command. You can debug this kind of issues in the IPython console embedded in the CQ-editor.

I see that you are using `polyline`, you might be also interested in `spline`.

winksaville, 2020-06-26:
When I get up I'll try setting PYTHONPATH, but why does `python wing1.py`
work? Also, the two files, mh49.py and scale.py, are in the same directory
as wing1.py.

I'll take a look at spline, txs.

winksaville, 2020-06-26:
So, as you expected, setting PYTHONPATH worked. Can you explain why `python
wing1.py` works but `cq-editor wing1.py` fails?

It seems there should be project level preferences via a "config file" for
things like PYTHONPATH?

adam-urbanczyk, 2020-06-26:
Python, when called directly, adds the current folder to PYTHONPATH. If all the files are in the same dir there is an option to add the dir to PYTHONPATH when exectuing.
[image]

winksaville, 2020-06-26:
Hmm, AFAICT I have the latest:
```
(cqgui-master) wink@3900x:~/prgs/CadQuery/cq-editor (master)
$ git log -1
commit f6fafbe275590465582aaee1e88882d2a8552ab3 (HEAD -> master,
origin/master, origin/HEAD)
Author: Adam Urbańczyk <adam-urbanczyk@users.noreply.github.com>
Date:   Thu Jun 25 21:32:48 2020 +0200

    Mention master version in the README
```
But my Preferences doesn't have `Add current dir to path`. And another
difference is I don't have any of the icon's on my Preferences window:

[image: cq-editor-gf6fafbe.preferences.png]

adam-urbanczyk, 2020-06-27:
Maybe it is an issue with configuration persistance. Could you remove `~/.config/CadQuery`? BTW: I don't see the pics.

winksaville, 2020-06-27:
> Maybe it is an issue with configuration persistance. Could you remove `~/.config/CadQuery`?

I deleted it and ran `python run.py` and still no "Add current dir to path":

[image]

My inxi output:
```
(cqgui-master2) wink@3900x:~/prgs/CadQuery/cq-editor (master)
$ inxi -Fxz
System:    Kernel: 5.7.5-arch1-1 x86_64 bits: 64 compiler: gcc v: 10.1.0 Desktop: Cinnamon 4.6.5 Distro: Arch Linux 
Machine:   Type: Desktop Mobo: ASRock model: X570 Creator serial: <filter> UEFI: American Megatrends v: P2.10 date: 12/16/2019 
CPU:       Topology: 12-Core model: AMD Ryzen 9 3900X bits: 64 type: MT MCP arch: Zen 2 L2 cache: 6144 KiB 
           flags: avx avx2 lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm bogomips: 182489 
           Speed: 4052 MHz min/max: 2200/3800 MHz Core speeds (MHz): 1: 4052 2: 4071 3: 4046 4: 4046 5: 4052 6: 4053 7: 4052 
           8: 4051 9: 4051 10: 4052 11: 4051 12: 4051 13: 4052 14: 4051 15: 4047 16: 4050 17: 4047 18: 4051 19: 4051 20: 4051 
           21: 4051 22: 4051 23: 4051 24: 4046 
Graphics:  Device-1: NVIDIA TU116 [GeForce GTX 1660] vendor: Micro-Star MSI driver: nvidia v: 440.82 bus ID: 33:00.0 
           Display: x11 server: X.org 1.20.8 driver: nvidia unloaded: modesetting resolution: <xdpyinfo missing> 
           Message: Unable to show advanced data. Required tool glxinfo missing. 
Audio:     Device-1: NVIDIA TU116 High Definition Audio vendor: Micro-Star MSI driver: snd_hda_intel v: kernel bus ID: 33:00.1 
           Device-2: Advanced Micro Devices [AMD] Starship/Matisse HD Audio vendor: ASRock driver: snd_hda_intel v: kernel 
           bus ID: 35:00.4 
           Sound Server: ALSA v: k5.7.5-arch1-1 
Network:   Device-1: Intel I211 Gigabit Network vendor: ASRock driver: igb v: 5.6.0-k port: b000 bus ID: 28:00.0 
           IF: enp40s0 state: down mac: <filter> 
           Device-2: Intel Wi-Fi 6 AX200 driver: iwlwifi v: kernel port: b000 bus ID: 29:00.0 
           IF: wlp41s0 state: down mac: <filter> 
           Device-3: Aquantia AQC107 NBase-T/IEEE 802.3bz Ethernet [AQtion] vendor: ASRock driver: atlantic v: kernel 
           port: a000 bus ID: 2e:00.0 
           IF: enp46s0 state: up speed: 1000 Mbps duplex: full mac: <filter> 
Drives:    Local Storage: total: 465.76 GiB used: 90.48 GiB (19.4%) 
           ID-1: /dev/nvme0n1 vendor: Sabrent model: Rocket 4.0 500GB size: 465.76 GiB 
           ID-2: /dev/nvme1n1 vendor: Sabrent model: Rocket 4.0 500GB size: 465.76 GiB 
Partition: ID-1: / size: 491.15 GiB used: 90.41 GiB (18.4%) fs: ext4 dev: /dev/dm-2 
           ID-2: /boot size: 1.86 GiB used: 71.2 MiB (3.7%) fs: vfat dev: /dev/nvme0n1p1 
Swap:      ID-1: swap-1 type: file size: 512.0 MiB used: 0 KiB (0.0%) file: /swapfile 
Sensors:   System Temperatures: cpu: 81.4 C mobo: N/A 
           Fan Speeds (RPM): N/A 
Info:      Processes: 477 Uptime: 1d 17h 14m Memory: 31.29 GiB used: 6.48 GiB (20.7%) Init: systemd Compilers: gcc: 10.1.0 
           clang: 10.0.0 Shell: bash v: 5.0.17 inxi: 3.1.03
```
Here is my conda environment:
```
(cqgui-master2) wink@3900x:~/prgs/CadQuery/cq-editor (master)
$ conda info

     active environment : cqgui-master2
    active env location : /opt/anaconda/envs/cqgui-master2
            shell level : 2
       user config file : /home/wink/.condarc
 populated config files : 
          conda version : 4.8.3
    conda-build version : 3.18.11
         python version : 3.7.6.final.0
       virtual packages : __cuda=10.2
                          __glibc=2.31
       base environment : /opt/anaconda  (writable)
           channel URLs : https://repo.anaconda.com/pkgs/main/linux-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/linux-64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /opt/anaconda/pkgs
                          /home/wink/.conda/pkgs
       envs directories : /opt/anaconda/envs
                          /home/wink/.conda/envs
               platform : linux-64
             user-agent : conda/4.8.3 requests/2.24.0 CPython/3.7.6 Linux/5.7.5-arch1-1 arch/ glibc/2.31
                UID:GID : 1000:985
             netrc file : None
[... 20 lines cut ...]
```
Here is the contents of 'CQ GUI.conf':
```
(base) wink@3900x:~/.config/CadQuery
$ cat CQ\ GUI.conf 
[General]
3D%20Viewer=@Variant(\0\0\0\b\0\0\0\r\0\0\0\xe\0v\0i\0s\0i\0\x62\0l\0\x65\0\0\0\x1\x1\0\0\0\n\0v\0\x61\0l\0u\0\x65\0\0\0\0\xff\xff\xff\xff\0\0\0\b\0t\0y\0p\0\x65\0\0\0\0\xff\xff\xff\xff\0\0\0\n\0t\0i\0t\0l\0\x65\0\0\0\0\xff\xff\xff\xff\0\0\0\x18\0s\0t\0r\0i\0\x63\0t\0N\0\x61\0m\0i\0n\0g\0\0\0\x1\0\0\0\0\x12\0r\0\x65\0n\0\x61\0m\0\x61\0\x62\0l\0\x65\0\0\0\x1\0\0\0\0\x12\0r\0\x65\0m\0o\0v\0\x61\0\x62\0l\0\x65\0\0\0\x1\0\0\0\0\x10\0r\0\x65\0\x61\0\x64\0o\0n\0l\0y\0\0\0\x1\0\0\0\0\b\0n\0\x61\0m\0\x65\0\0\0\n\0\0\0\b\0P\0r\0\x65\0\x66\0\0\0\x10\0\x65\0x\0p\0\x61\0n\0\x64\0\x65\0\x64\0\0\0\x1\x1\0\0\0\xe\0\x65\0n\0\x61\0\x62\0l\0\x65\0\x64\0\0\0\x1\x1\0\0\0\xe\0\x64\0\x65\0\x66\0\x61\0u\0l\0t\0\0\0\0\xff\xff\xff\xff\0\0\0\x10\0\x63\0h\0i\0l\0\x64\0r\0\x65\0n\0\0\0\x7f\0\0\0\xePyQt_PyObject\0\0\0\x2S\x80\x3\x63\x63ollections\nOrderedDict\nq\0)Rq\x1(X\x11\0\0\0\x46it automaticallyq\x2}q\x3(X\x4\0\0\0typeq\x4X\x4\0\0\0\x62oolq\x5X\b\0\0\0readonlyq\x6\x89X\a\0\0\0visibleq\a\x88X\a\0\0\0\x65nabledq\b\x88X\t\0\0\0renamableq\t\x89X\t\0\0\0removableq\n\x89X\f\0\0\0strictNamingq\v\x89X\b\0\0\0\x65xpandedq\f\x88X\x5\0\0\0titleq\rNX\x4\0\0\0nameq\x [... cut ...] x4\0\0\0\b\0\0\0\b\xfc\0\0\0\x1\0\0\0\x2\0\0\0\x1\0\0\0\x18\0M\0\x61\0i\0n\0 \0t\0o\0o\0l\0\x62\0\x61\0r\x1\0\0\0\0\xff\xff\xff\xff\0\0\0\0\0\0\0\0)
```
In the `Debugger=` I only see:

  - \0Reload CQq\x2}
  - \x64 script dir to pathq\x11}
  - \x43hange working dir to script dirq\x13}

To double check, I looked for 'urrent' (leaving off the 'c' incase it was converted to hex), nothing:
```
(base) wink@3900x:~/.config/CadQuery
$ grep urrent 'CQ GUI.conf' 
(base) wink@3900x:~/.config/CadQuery
````

adam-urbanczyk, 2020-06-27:
Sorry for the noise - my options were obsolete, your are correct:
[image]

With that being said, adding script dir to path should be the solution. I cloned your repo and it does work for me with the settings you can see above:
[image]

winksaville, 2020-06-27:
I was starting to think my system was totally messed up, so I'm glad you're now seeing something similar to me :)

But lets see if we can figure out why your's works and my doesn't. In the console window of cq-editor I executed `import sys; print(sys.path)`:
```
import sys; print(sys.path)
['/opt/anaconda/envs/cqgui-master/bin', '/opt/anaconda/envs/cqgui-master/lib/python37.zip', '/opt/anaconda/envs/cqgui-master/lib/python3.7', '/opt/anaconda/envs/cqgui-master/lib/python3.7/lib-dynload', '/opt/anaconda/envs/cqgui-master/lib/python3.7/site-packages', '/opt/anaconda/envs/cqgui-master/lib/python3.7/site-packages/IPython/extensions']
```
So I'm betting if you execute `import sys; print(sys.path)` your sys.path contains the path to `cadquery-wing1`.

For example if I execute cq-editor with "PYTHONPATH=.":
```
(cqgui-master) wink@3900x:~/prgs/CadQuery/projects/wing1 (master)
$ PYTHONPATH=. cq-editor wing1.py 
Namespace(filename='wing1.py')
TKOpenGl | Type: Other | ID: 0 | Severity: Medium | Message:
  OpenGl_Window::CreateWindow: window Visual is incomplete: no stencil buffer
Font_FontMgr, warning: unable to find font 'Courier' [regular]; 'Cantarell' [aspects: regular,bold] [paths: /usr/share/fonts/cantarell/Cantarell-Regular.otf;/usr/share/fonts/cantarell/Cantarell-Bold.otf] is used instead
```
Now notice the second entry is **'/home/wink/prgs/CadQuery/projects/wing1'**, and all is well:
```
import sys; print(sys.path)
['/opt/anaconda/envs/cqgui-master/bin', '/home/wink/prgs/CadQuery/projects/wing1', '/opt/anaconda/envs/cqgui-master/lib/python37.zip', '/opt/anaconda/envs/cqgui-master/lib/python3.7', '/opt/anaconda/envs/cqgui-master/lib/python3.7/lib-dynload', '/opt/anaconda/envs/cqgui-master/lib/python3.7/site-packages', '/opt/anaconda/envs/cqgui-master/lib/python3.7/site-packages/IPython/extensions']
```

There is still one other difference, notice that you have the "icons" and I don't, my guess is that problem has to do with the version of QT or maybe the version of OS?

 I'm using, Qt 5.9.7, on my Arch Linux system, what do you have?
```
$ qtdiag
Qt 5.9.7 (x86_64-little_endian-lp64 shared (dynamic) release build; by GCC 7.3.0) on "xcb" 
OS: Arch Linux [linux version 5.7.5-arch1-1]

Architecture: x86_64; features: SSE2 SSE3 SSSE3 SSE4.1 SSE4.2 AVX AVX2

Environment:

Features: QT_NO_EXCEPTIONS

Library info:
  PrefixPath: /opt/anaconda/envs/cqgui-master
  DocumentationPath: /opt/anaconda/envs/cqgui-master/doc
  HeadersPath: /opt/anaconda/envs/cqgui-master/include/qt
  LibrariesPath: /opt/anaconda/envs/cqgui-master/lib
  LibraryExecutablesPath: /opt/anaconda/envs/cqgui-master/libexec
  BinariesPath: /opt/anaconda/envs/cqgui-master/bin
  PluginsPath: /opt/anaconda/envs/cqgui-master/plugins
  ImportsPath: /opt/anaconda/envs/cqgui-master/imports
  Qml2ImportsPath: /opt/anaconda/envs/cqgui-master/qml
  ArchDataPath: /opt/anaconda/envs/cqgui-master
  DataPath: /opt/anaconda/envs/cqgui-master
  TranslationsPath: /opt/anaconda/envs/cqgui-master/translations
  ExamplesPath: /opt/anaconda/envs/cqgui-master/examples
  TestsPath: /opt/anaconda/envs/cqgui-master/tests
  SettingsPath: /opt/anaconda/envs/cqgui-master

Standard paths [*...* denote writable entry]:
  DesktopLocation: "Desktop" */home/wink/Desktop*
  DocumentsLocation: "Documents" */home/wink/Documents*
  FontsLocation: "Fonts" */home/wink/.local/share/fonts* /home/wink/.fonts
  ApplicationsLocation: "Applications" */home/wink/.local/share/applications* /usr/local/share/applications /usr/share/applications
  MusicLocation: "Music" */home/wink/Music*
  MoviesLocation: "Movies" */home/wink/Videos*
  PicturesLocation: "Pictures" */home/wink/Pictures*
  TempLocation: "Temporary Directory" */tmp*
  HomeLocation: "Home" */home/wink*
  AppLocalDataLocation: "Application Data" */home/wink/.local/share/QtProject/qtdiag* /usr/local/share/QtProject/qtdiag /usr/share/QtProject/qtdiag
  CacheLocation: "Cache" */home/wink/.cache/QtProject/qtdiag*
  GenericDataLocation: "Shared Data" */home/wink/.local/share* /usr/local/share /usr/share
  RuntimeLocation: "Runtime" */run/user/1000*
  ConfigLocation: "Configuration" */home/wink/.config* /etc/xdg
[... 78 lines cut ...]
  Virtual geometry: 6000x3840+0+0 Available: 6000x3049+0+0
  2 virtual siblings
  Physical size: 600x340 mm  Refresh: 59 Hz Power state: 0
  Physical DPI: 162.56,161.365 Logical DPI: 96,96 Subpixel_None
  DevicePixelRatio: 1 Pixel density: 2
  Primary orientation: 2 Orientation: 2 Native orientation: 0 OrientationUpdateMask: 0

# 1 "DP-0" Depth: 24 Primary: no
  Manufacturer: 
  Model: 
  Serial number: 
  Geometry: 2160x3840+3840+0 Available: 2160x3049+3840+0
  Virtual geometry: 6000x3840+0+0 Available: 6000x3049+0+0
  2 virtual siblings
  Physical size: 340x600 mm  Refresh: 59 Hz Power state: 0
  Physical DPI: 161.365,162.56 Logical DPI: 96,96 Subpixel_None
  DevicePixelRatio: 1 Pixel density: 2
  Primary orientation: 1 Orientation: 4 Native orientation: 0 OrientationUpdateMask: 0

LibGL Vendor: NVIDIA Corporation
Renderer: GeForce GTX 1660/PCIe/SSE2
Version: 4.6.0 NVIDIA 440.82
Shading language: 4.60 NVIDIA
Format: Version: 4.6 Profile: 2 Swap behavior: 0 Buffer size (RGB): 8,8,8 Depth buffer: 24
```

adam-urbanczyk, 2020-06-28:
What shows up in the console is not really relevant, the path is modified for the script execution. Could prepend the following lines to your code and tell me what you see in the logging window when you execute?
```python
import sys
log(sys.path)
```

winksaville, 2020-06-28:
Executing `cq-editor wing1.py` has the ModuleNotFoundError and the "." in the path is missing:

[2020-06-28 13:50:56.967162] INFO: Generic: ['/opt/anaconda/envs/cqgui-master/bin', '/opt/anaconda/envs/cqgui-master/lib/python37.zip', '/opt/anaconda/envs/cqgui-master/lib/python3.7', '/opt/anaconda/envs/cqgui-master/lib/python3.7/lib-dynload', '/opt/anaconda/envs/cqgui-master/lib/python3.7/site-packages', '/opt/anaconda/envs/cqgui-master/lib/python3.7/site-packages/IPython/extensions']

Executing `PYTHONPATH=. cq-editor wing1.py`, no errors and we see the added "." path:

[2020-06-28 13:54:51.552553] INFO: Generic: ['/opt/anaconda/envs/cqgui-master/bin', **'/home/wink/prgs/CadQuery/projects/wing1'**, '/opt/anaconda/envs/cqgui-master/lib/python37.zip', '/opt/anaconda/envs/cqgui-master/lib/python3.7', '/opt/anaconda/envs/cqgui-master/lib/python3.7/lib-dynload', '/opt/anaconda/envs/cqgui-master/lib/python3.7/site-packages', '/opt/anaconda/envs/cqgui-master/lib/python3.7/site-packages/IPython/extensions']

adam-urbanczyk, 2020-06-28:
Thanks, could you check if #158 solves the issue?

winksaville, 2020-06-28:
Yes, but why the "Path(xxx)" and what's different between your enviroment and mine?

[2020-06-28 15:05:14.375411] INFO: Generic: ['/home/wink/prgs/CadQuery/cq-editor', '/opt/anaconda/envs/cqgui-master2/lib/python37.zip', '/opt/anaconda/envs/cqgui-master2/lib/python3.7', '/opt/anaconda/envs/cqgui-master2/lib/python3.7/lib-dynload', '/opt/anaconda/envs/cqgui-master2/lib/python3.7/site-packages', '/opt/anaconda/envs/cqgui-master2/lib/python3.7/site-packages/IPython/extensions', **Path('/home/wink/prgs/CadQuery/projects/wing1')**]

winksaville, 2020-06-28:
Incase there are modules in the script dir that replace/enhance/override functionality in other modules in the sys.path, maybe it should be at the beginning?

adam-urbanczyk, 2020-06-28:
You are probably right with the prepending instead of appending. Could you confirm that the PR does solve the issue? From what you pasted I conclude that it is the case, but reading the reply I'm not sure. BTW: Path is a class from the path.py module I use for doing anything path related.

winksaville, 2020-06-28:
Yes it worked fine, sorry I wasn't clear.

As for Path, I was referring to seeing
Path('/home/wink/prgs/CadQuery/projects/wing1') in the log output, which
seemed odd as the Path seems unnecessary and isn't present in any of the
other entries of sys.path.

adam-urbanczyk, 2020-06-28:
> As for Path, I was referring to seeing Path('/home/wink/prgs/CadQuery/projects/wing1') in the log output, which seemed odd as the Path seems unnecessary and isn't present in any of the other entries of sys.path.

See the answer above, it is just a class I'm using.

winksaville, 2020-06-28:
I'm no expert in Python, so bear with me, please.

I added some debug to debugger.py _exec:

    def _exec(self, code, locals_dict, globals_dict):

        with ExitStack() as stack:
            p =
Path(self.parent().components['editor'].filename).abspath().dirname()
            print(f'_exec: p={p}')
            if self.preferences['Add script dir to path'] and p:
                print(f'_exec: sys.path={sys.path}')
                sys.path.append(p)
                print(f'_exec: sys.path={sys.path}')
                stack.callback(sys.path.remove, p)
            if self.preferences['Change working dir to script dir'] and p:
                stack.enter_context(p)
            exec(code, locals_dict, globals_dict)

And the output is:

(cqgui-master2) wink@3900x:~/prgs/CadQuery/cq-editor (pr158)
$ python run.py
Namespace(filename=None)
TKOpenGl | Type: Other | ID: 0 | Severity: Medium | Message:
  OpenGl_Window::CreateWindow: window Visual is incomplete: no stencil
buffer
Font_FontMgr, warning: unable to find font 'Courier' [regular]; 'Cantarell'
[aspects: regular,bold] [paths:
/usr/share/fonts/cantarell/Cantarell-Regular.otf;/usr/share/fonts/cantarell/Cantarell-Bold.otf]
is used instead
_exec: p=/home/wink/prgs/CadQuery/projects/wing1
_exec: sys.path=['/home/wink/prgs/CadQuery/cq-editor',
'/opt/anaconda/envs/cqgui-master2/lib/python37.zip',
'/opt/anaconda/envs/cqgui-master2/lib/python3.7',
'/opt/anaconda/envs/cqgui-master2/lib/python3.7/lib-dynload',
'/opt/anaconda/envs/cqgui-master2/lib/python3.7/site-packages',
'/opt/anaconda/envs/cqgui-master2/lib/python3.7/site-packages/IPython/extensions']
_exec: sys.path=['/home/wink/prgs/CadQuery/cq-editor',
'/opt/anaconda/envs/cqgui-master2/lib/python37.zip',
'/opt/anaconda/envs/cqgui-master2/lib/python3.7',
'/opt/anaconda/envs/cqgui-master2/lib/python3.7/lib-dynload',
'/opt/anaconda/envs/cqgui-master2/lib/python3.7/site-packages',
'/opt/anaconda/envs/cqgui-master2/lib/python3.7/site-packages/IPython/extensions',
Path('/home/wink/prgs/CadQuery/projects/wing1')]

So "p=/home/wink/prgs/CadQuery/projects/wing1", yet after appending "p" to
sys.path we have Path(xxx) as the last entry:

Path('/home/wink/prgs/CadQuery/projects/wing1')

Where as I would have expected sys.path to just the actual path as the last
entry:

'/home/wink/prgs/CadQuery/projects/wing1'

It is "working" and I shouldn't worry about it, but it just seems
surprising, can you explain?

adam-urbanczyk, 2020-06-29:
I don't see how this is relevant. If you want to inspect the type of an object stop in pdb (or use `print(type(o))`) and don't infer it from the string representation it is providing.

winksaville, 2020-06-29:
> I don't see how this is relevant. If you want to inspect the type of an object stop in pdb (or use `print(type(o))`) and don't infer it from the string representation it is providing.

I'm not understanding something about Python, but you've fixed the main issue so it's not relavent, txs.

winksaville, 2020-06-29:
There is one last tweak to the code, the new sys.path element should be prepended, correct?

adam-urbanczyk, 2020-06-29:
That's already in.

closed 2020-06-29 by adam-urbanczyk

winksaville, 2020-06-29:
I did make some progress on increasing my understanding of Python. I used your suggestion of printing type(p) and saw clearly that "p = Path(self.parent().components['editor'].filename).abspath().dirname()" returned a <class 'path.Path'> not as I'd originally assumed, a string. Not being strictly typed is one of the things I don't like about Python, but it's nice there is `mypy` and type annotations, which at least slaps your wrist when you do stupid stuff.

Again, thanks for being so helpful.

PR #158 (2020-06-29): Fix pythonpath update for relative file paths
Fixes #156