CadQuery/CQ-editor #157: Show Active File Name and Edit Status

dominic-valentino, 2020-06-27:
It would a nice to have the opened file name and path displayed somewhere obvious like the top of the main window.
It could also show an asterisk if it has been edited and not saved.

P.S.
I love this project, and CadQuery.  I was trying to use Autodesk Fusion's Python API and it was ugly!
This is just what I was looking for, a simple Python based GUI and API.

-Dominic

roipoussiere, 2021-03-02:
The file name could prepended to the window title, like in Firefox of many other software, like this: `box_with_holes.py - CQ-Editor`.

roipoussiere, 2021-03-02:
Oh, this will be solved when #228 will be merged.

jmwright, 2025-02-11:
#481 added the last part of this which is edit status.

closed 2025-02-11 by jmwright