CadQuery/CQ-editor #159: ModuleNotFoundError: No module named 'OCP'

lalebarde, 2020-07-05:
Hi,

I have just updated my CQ-editor with the following process:

```
git pull
conda env update -f cqgui_env.yml
```

Then when I start it with `python run.py`, I get the following error:

```
Traceback (most recent call last):
  File "run.py", line 9, in <module>
    from cq_editor.__main__ import main
  File "/opt/CQ-editor/cq_editor/__main__.py", line 12, in <module>
    from .main_window import MainWindow
  File "/opt/CQ-editor/cq_editor/main_window.py", line 8, in <module>
    from .widgets.viewer import OCCViewer
  File "/opt/CQ-editor/cq_editor/widgets/viewer.py", line 11, in <module>
    from OCP.AIS import AIS_Shaded,AIS_WireFrame, AIS_ColoredShape, \
ModuleNotFoundError: No module named 'OCP'
```

This looks relative to [this](https://github.com/CadQuery/CQ-editor/pull/76). How to install OCP please? Is it related to OpenShift?

lalebarde, 2020-07-06:
As a temporary workaround, I have checked out to version tagged 0.1.1 : `git checkout tags/0.1.1`

lalebarde, 2020-07-06:
What if I perform a `conda install -c conda-forge -c cadquery ocp` in my CQ-Editor setup built with:

```
cd /opt
git clone https://github.com/CadQuery/CQ-editor.git
cd CQ-editor/
conda env create -f cqgui_env.yml -n cqgui
```
?

lalebarde, 2020-07-06:
It works, so now to install {CadQuery + CQ-Editor} with OCP dependency, one has to do (get the latest anaconda or update it):

```
wget https://repo.anaconda.com/archive/Anaconda3-2019.10-Linux-x86_64.sh
bash Anaconda3-2019.10-Linux-x86_64.sh
source ~/.bashrc
cd /opt
git clone https://github.com/CadQuery/CQ-editor.git
cd CQ-editor/
conda env create -f cqgui_env.yml -n cqgui
conda activate cqgui
conda install -c conda-forge -c cadquery ocp 
```

and run with:

```
(base) ~$ conda activate cqgui
(cqgui) ~$ cd /opt/CQ-editor/
(cqgui) /opt/CQ-editor$ python run.py 
```

update with:

```
(cqgui) /opt/CQ-editor$ git pull
(cqgui) /opt/CQ-editor$ conda env update -f cqgui_env.yml
```

adam-urbanczyk, 2020-07-06:
Thanks for the investigation. I don't have experience with `conda env update` it is better to create the env from scratch. Note that the current env file already contains the OCP dependancy.

lalebarde, 2020-07-06:
The update procedure was provided to me in another ticket by Jeremy. Unfortunately, I tested only CQ-Editor launch, not model building. Actually, my model does not build and no error is showing.

lalebarde, 2020-07-06:
Just for the case, back to version tagged 0.1.1 works - my moel builds.

lalebarde, 2020-07-09:
> I don't have experience with `conda env update` it is better to create the env from scratch. Note that the current env file already contains the OCP dependency.
You were right Adam, rebuilding from scratch make it works and no needs to explicitely install ocp, it is included as a dependency.

closed 2020-07-09 by lalebarde